UltimateHackingKeyboard/firmware #170: MacOS - Cmd key sticking down when multiple keys pressed on mod layer

dkilmer-mfj, 2018-09-06:
My use case is that the Mod layer is for text editing/manipulation. So I have arrows, backspace, home, end and so forth mapped there, as well as things like Cmd-C, Cmd-V, Cmd-A, etc. Any time I use a key that has Cmd as a modifier, it carries over to my next press of an arrow key, taking me to the top or bottom of a document.

I used the Key Codes app to log what it's doing:

## Mod down, key mapped to Cmd-C down/up, key mapped to up-arrow down/up, Mod up

```
Modifier Change
	Keys:		⌘
	Key Code:	65535 / 0xffff
	Modifiers:	1048840 / 0x100108 ⓘ

Key Down
	Characters:	c
	Unicode:		99 / 0x63
	Keys:		⌘C
	Key Code:	8 / 0x8
	Modifiers:	1048840 / 0x100108 ⓘ

Key Up
	Characters:	c
	Unicode:		99 / 0x63
	Keys:		⌘C
	Key Code:	8 / 0x8
	Modifiers:	1048840 / 0x100108 ⓘ

Key Down
	Characters:	
	Unicode:		63232 / 0xf700
	Keys:		⌘↑
	Key Code:	126 / 0x7e
	Modifiers:	11534600 / 0xb00108 ⓘ

Modifier Change
	Keys:		
	Key Code:	65535 / 0xffff
	Modifiers:	256 / 0x100 ⓘ

Key Up
	Characters:	
	Unicode:		63232 / 0xf700
	Keys:		↑
	Key Code:	126 / 0x7e
	Modifiers:	10486016 / 0xa00100 ⓘ
```

## Mod down, key mapped to Cmd-C down/up, Mod up, Mod down, key mapped to up-arrow down/up, Mod up

```
Modifier Change
	Keys:		⌘
	Key Code:	65535 / 0xffff
	Modifiers:	1048840 / 0x100108 ⓘ

Key Down
	Characters:	c
	Unicode:		99 / 0x63
	Keys:		⌘C
	Key Code:	8 / 0x8
	Modifiers:	1048840 / 0x100108 ⓘ

Key Up
	Characters:	c
	Unicode:		99 / 0x63
	Keys:		⌘C
	Key Code:	8 / 0x8
	Modifiers:	1048840 / 0x100108 ⓘ

[mod up here issues the modifier change]

Modifier Change
	Keys:		
	Key Code:	65535 / 0xffff
	Modifiers:	256 / 0x100 ⓘ

Key Down
	Characters:	
	Unicode:		63232 / 0xf700
	Keys:		↑
	Key Code:	126 / 0x7e
	Modifiers:	10486016 / 0xa00100 ⓘ

Key Up
	Characters:	
	Unicode:		63232 / 0xf700
	Keys:		↑
	Key Code:	126 / 0x7e
	Modifiers:	10486016 / 0xa00100 ⓘ
```

I tested a similar use case on Linux. Ctl-C and Home both mapped to keys on mod layer. Pressing those in sequence with mod held down does not go to top of the document and goes to the beginning of the line as expected.

mondalaci, 2018-09-09:
I cannot reproduce this issue with 8.4.5. Please share your config, and elaborate in detail about which keys of which layer of which keymap you use.

dkilmer-mfj, 2018-09-10:
I'm using the "QWERTY for mac" keymap in the attached configuration. To reproduce the behavior in the first example with that keymap activated on the keyboard, you can open any multi-line file in Text Edit, move to the middle of the document, and press the following sequence of keys:

* `Space` key on left side of keyboard down (enables the `mod` layer)
* `D` key down and then up (issues Cmd-C)
* `I` key down and then up (issues up-arrow)
* `Space` key on left side of keyboard up (returns to `base` layer)

The behavior I see is that when the `I` key goes down, the cursor moves to the top of the document rather than to the next line up in the document.

[UhkConfigIssue170.json.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2367927/UhkConfigIssue170.json.zip)

mondalaci, 2018-09-16:
I still cannot reproduce this issue on my 2012 Macbook Pro.

Any luck, @eltang?

dkilmer-mfj, 2018-09-25:
Ah. I'd forgotten that I had Karabiner Elements installed on the Mac with the problem, and although it wasn't set up to do anything, when I exited it, the behavior disappeared.

closed 2018-09-25 by dkilmer-mfj