UltimateHackingKeyboard/firmware #173: Center mouse in current window

wesdean, 2018-09-14:
I would love a way to center the mouse in my currently focused window. I have 3 screens and often switch windows without using the mouse. When I do need my mouse in a window that I have switched to the pointer is often at opposite end of my displays. It can take a while to move the mouse to my current window. This feature would be very helpful, thank you.

mondalaci, 2018-09-14:
I'm sorry but only relative pointer movement is possible with the UHK. Centering the pointer in the current window is not possible because the origin pointer coordinates and window coordinates are unknown. This feature is best suited for window managers.

closed 2018-09-14 by mondalaci

luteijn, 2018-09-14:
I'd set up couple of similar macros to move the mouse an extreme amount to the left and up, to make sure the pointer is homed at (0,0) and then follow that up with a mouse move to the coordinates centered for screen 1, 2 and 3 respectively. Then you can bind them to three keys on e.g. mouse or fn layer to at least quickly center the mouse on a particular screen.

mondalaci, 2018-09-14:
This should work for centering the pointer on screens, but not in windows.

mondalaci, 2018-09-14:
Sorry, but this is not possible.