UltimateHackingKeyboard/firmware #177: Win10 - Boot hanging until USB cable pulled out

Timoshun, 2018-09-20:
It's happening frequently. I've tried with various firmware 2-3 versions back. This is an issue since I got my UHK. I'm happy to assist the investigation with any data necessary.
Motherboard: ASRock X99M Extreme4

Lotharyx, 2018-09-20:
My Win7 machine hangs during boot 100% of the time if the UHK is plugged 
in, continuing to boot once the UHK is unplugged.  Glad to hear at least 
one other person is seeing a boot hang on a supported platform.  I hope 
if this is fixed against Win 10, it'll also fix Win 7...

-B

Timoshun, 2018-09-20:
> My Win7 machine hangs during boot 100% of the time if the UHK is plugged in, continuing to boot once the UHK is unplugged.  Glad to hear at least one other person is seeing a boot hang on a supported platform.  I hope if this is fixed against Win 10, it'll also fix Win 7...
> […](#)
> -B

What is strange for me is that sometimes I just let my PC there to boot when I'm get home and when I'm back it's loaded, so I cannot say it for sure that it's 100%. I should spend more time to test this issue out, changing USB ports maybe.. Let's see what the guys saying about it.

Thanks for your feedback, I'm also "glad" that I'm not the only one with this problem.

Timos

mondalaci, 2018-09-20:
Please update the [latest firmware](https://github.com/UltimateHackingKeyboard/firmware/releases/download/v8.4.5/uhk-firmware-8.4.5.tar.bz2) in Agent, and let me know if it resolves the issue, guys.

Lotharyx, 2018-09-20:
Since firmware update isn't supported on Windows 7, I'll wait to hear 
what Gergo finds to determine if the update is worth the effort and risk.

Cheers,
-B

Timoshun, 2018-09-20:
Szia Laci,
I've missed the new patch! After an update on the agent and firmware to 8.2.5:
I've done some testing cycle
Shutdown and a restart > fine
Restart from windows > fine
Shutdown

Then I changed the USB port, plugged the UHK into a different port, the LCD powered up and stayed that way, PC was powered off.
Then I started a boot, it got stuck. PC restarted itself, after a restart during the Win logo booting the UHK turned off and on and OS boot completed successfully. Then I hit a restart and this time everything went fine.

Major issue solved. Thanks.

mondalaci, 2018-09-21:
Szia @Timoshun!

I'm glad that the major issue is resolved. Would you please elaborate about the remaining issue and how to reproduce it? I'm not sure based on the above.

Lotharyx, 2018-09-21:
Sorry for what feels like a stupid question, but how do I get Agent to 
use something other than the firmware image that came with it? The 
current release of Agent comes with 8./2/.5, and you've suggested trying 
out 8./4/.5.  I wasn't able to find instructions.

Thanks,
-Brian

kareltucek, 2018-09-21:
1) Download the tar.bz2 file at https://github.com/UltimateHackingKeyboard/firmware/releases
2) Click ''Chose firmware file and flash it" in agent and select the downloaded file...

Lotharyx, 2018-09-21:
*feels dumb*

Yep.  I hadn't had my coffee yet.  Thank you!

-B

Lotharyx, 2018-09-21:
Hooray!  I can confirm the boot-hang issue is resolved with 8.4.5 
against Windows 7.

Cheers,
-B

Timoshun, 2018-09-25:
How to reproduce the issue:
- upgrade firmware to the latest.
- shut down the PC 
- unplug the keyboard, plug in to a different USB port, UHK powers up (LCD will show the default mode)
- start the PC, then it will be holding up the windows boot.

I also noticed yesterday, that -after a few days of use- the original problem occurred again. UHK stopes the boot on Windows OS.

mondalaci, 2018-09-25:
@Timoshun I cannot reproduce this issue on my Thinkpad T520 with Windows 10.

1. Does it matter which USB port you plug your UHK into?
2. Does this issue always happen?