UltimateHackingKeyboard/firmware #180: Remapping layer switcher on the target layer doesn't work.

kareltucek, 2018-09-27:
This might be by design (I hope it is not.), but still - remapping layer switcher on the target layer doesn't work.

For example, import the attached config 
[layer_switchers_UserConfig.json.txt](https://github.com/UltimateHackingKeyboard/firmware/files/2423380/layer_switchers_UserConfig.json.txt)

- Check that abbrev reads TST
- press and hold left case switcher
- tap right case switcher
  - expected output: "a"
  - actual output: nothing

My use case: I have a few copies of the same keymap (the "Mirroring QWERTY") with different mappings of the right and left half of keyboard, which mimic ability to lock a nondefault layer on just right keyboard half. I lock them by "mouse switcher + \<layer switcher of the target layer\>", which is very intuitive but cannot be used for the "mouse" layer (actually mod layer). I was using this scheme with one of the layers and it worked fine. Just today I had hit this restriction. (Hit mouse + "+" if you wish to see it in action, then mouse+enter switches a custom mouse/browsing mode on right half of keyboard)

Of course, I am going to solve this differently for the time being.

kareltucek, 2018-09-27:
(Btw., I have a mild suspiction that something weird goes on in serialization of target switchers.)

luteijn, 2018-09-27:
mapping a key to a layer switcher also overwrites that key's position on the switched layer as well, although you can then map it back. Also using secondary functions might help here as there might be checks not done when using them...

kareltucek, 2018-09-27:
Nope, no difference with secondary roles.

mondalaci, 2018-09-28:
> Check that abbrev reads TST

I assume you mean that I should switch to the Test keymap which I did.

> press and hold left case switcher

Ok, I held the left case button which is mapped to Mod.

> tap right case switcher
> expected output: "a"
> actual output: nothing

Given that the right case button of the Mod layer maps to Mod, nothing is the expected behavior.

kareltucek, 2018-09-28:
Well, that means that you have now encountered another bug (in import/export/serialization/i-dont-know-where).

So now please go on and:
- open the TST keymap in agent
- map "a" key to the right case switcher on the Mod layer
- save the settings
- repeat the procedure described above and notice that the "a" on the mod layer does indeed not work
- Now restart the agent and check if the mapping survived the restart. (It didn't.)
- Map it again. Export config and reimport it. Check the mapping - it didn't survive this procedure either.
- Now, if my motivation is clear, please try to dry some conclusions on your own. (I.e., decide to yourself which ones of these are features and which are bugs.)

mondalaci, 2018-09-28:
I'm just realizing that Agent shouldn't even allow this mapping to begin with.

Given a layer switcher action on the base layer, the same layer switcher action should be featured on the target layer of the key.

Given this concrete example, you want to map Mod on the base layer and "a" on the Mod layer of the key. This doesn't make sense to me because when you press the Mod key, you're suddently on the Mod layer, so the Mod layer switcher action should be mapped on the Mod layer, too.

What do you expect to happen when you press a key that is mapped in such a way?

kareltucek, 2018-09-28:
Yes, exactly as luteijn says. I think it is an expectation consistent with the switching logic which we discussed some time ago... Anyway, I assumed this was intended since the layer switching tab is programatically disabled in this case but the rest of the dialog is not. 

I should probably enumerate all the problems I see here:
- GUI inconsistency (obviously GUI should not allow something that makes the agent misbehave so badly)
- The question whether it should be allowed or not. (My opinion is as always - the user should be given as much freedom as possible.)
- The actual implementational issue with serialization/deserialization*, since correct behaviour of this case may be vital for further generalizations of layer switching logic. 
- The conceptual problem of serialization/deserialization* - I find it a little odd that serialization is not handled so that result of serialization and deserialization is guaranteed to be identical.

*I mean generally the problem of the mapping being lost in this case irrespectively of where exactly things go wrong.

kareltucek, 2018-09-28:
Anyway, I think that secondary roles also allow this kind of behaviour, do they not?

mondalaci, 2018-09-28:
Users are prone to lock themselves out, and mess up stuff in a million different ways. Then they write to support, asking why things don't work the way they expect to.

This is a quite special case. The simplest behavior, and the behavior I think is expected by most is for layer switcher keys is to behave consistently regardless of which one of them was pressed.

We'll make Agent disallow this case. When the key action popover pops up for a key on a non-base layer which is a layer switch action of the base layer which targets the aforementioned layer, then:
* The Layer tab will be active, and it won't be possible to switch to another tab.
* The "Remap key" button will be inactive.
* The content under the Layer tab will be:
> You can't change this mapping because on the base layer a layer switcher key targets this key.

luteijn, 2018-09-29:
With secondary roles, at least as I remember it, it was possible to have a key mapped to say 'a' or even 'shift' on both mod and base layer, but have a secondary role to be a 'mod-switch', and it would work as expected (although maybe not survive an export and re-import config cycle).

Although I understand that issues around having to deal with customers (including Karel and myself ;) ) that get themselves into a hole, it's a bit saddening that the 'hacking' keyboard ends up being more of a fancy normal keyboard each time a 'hacker' asks a question about a feature not quite working the way we expect it, or would think it to be more generally useful. The whole feature is declared a bug, instead of a buggy feature, and we end up with even less in most of these cases.

Unfortunately there is no real way around it, even if you had an "I know what I'm doing" tick mark to unlock expert features, or choose how a feature behaves, due to well-known effects, the wrong kind of people would just tick it and then flood support with questions and demands... 

As for the 'simplest behavior' being A or B, it highly depends on where you are coming from, and what your ideas of how the keyboard should work are. From what I've seen, there are a few people like Karel and me that would prefer a very general approach that makes as little difference between things like for example switching layers and switching keymaps, or pressing 'super'/'ctrl'/'alt'/'shift' , 'mod'/'mouse'/'fn', 'menu'/'caps-lock', 'normal keys' and even 'hyper'/some-other-traditional-but-not-common-anymore-modifier-key. I certainly prefer a much more decoupled and generalized model, where the key, chord or even sequence of keys pressed has a much more flexible impact on what code-sequence is actually sent to the host. Basically, I'd expect to be able to do what a program like 'autohotkey' did on windows, only OS-agnositic, because it would completely run on the keyboard.

kareltucek, 2018-09-29:
> What you're trying to achieve is not possible via secondary roles.

Actually, it is perfectly well possible. The only catches are:

- That leds become pretty useless as an indicator of anything.
- That you have disallowed mapping secondary role without a primary role some time ago, so now I have to have shift mapped as a primary on my layer switchers for no sensible reason at all.

> This is an Agent issue.

Could you please say as much as one sentence regarding the "serialization" issue?

I ask because this exact problem is what have kept me from diving into any hacking of UHK *firmware* so far - mainly the knowledge that something related to this is wrong and that it can be anywhere in agent's code or in the serializer, which are indeed very complicated parts of code.

I know you have warned me that you do not have resources to support my meddling with your code, but still, this is like having a car whose engine falls apart each time its user turns on radio. 

> (although maybe not survive an export and re-import config cycle)

Actually I think that this works as expected with respect to all format transforms.

mondalaci, 2018-09-30:
@luteijn I can see where you're coming from, and I know you're writing these comments because you genuinely care which I appreciate, even if I disagree from time to time.

> That you have disallowed mapping secondary role without a primary role some time ago, so now I have to have shift mapped as a primary on my layer switchers for no sensible reason at all.

The reason is that it doesn't make practical sense to set up a secondary role without a primary role because one can just use a primary role. It was confusing for some users.

> Could you please say as much as one sentence regarding the "serialization" issue?

Chances are it's either due to the configuration serializer of Agent or the configuration parser of the firmware. In any case, it'll be a non-issue as soon as this option will be disallowed in Agent.

I think the radio metaphor was quite undue. The UHK offers a lot, but you're after much more. I have a feeling that you guys would only be happy if the UHK ran QMK or Kaleidoscope. But then most users would be unhappy about how hard it is to customize their UHKs. There's no way to make everyone happy, even if I write such comments at Sunday midnight, which I'm just doing.

luteijn, 2018-10-01:
@mondalaci Know that I do appreciate all your effort, and of course this medium combined with using a second language does pose some limits on expression, so everyone might come over a bit more harsh than they mean.  Also fully understand the way you have to keep balance between many different stakeholders, including yourself ... 

For me the 'arguments' given here are only meant in the rhetorical way, to try and nudge disagreements a bit more to the 'power user' side, not to pick any fights.

Although confusing, there is/was a point or two to using secondary modifiers with no primary function. Granted, the main reason was a bit of a hack to work around the way layer switching works bur there is also the fact that using a modifier as a secondary action means that nothing will get sent to the OS until one actually pushes that second key to go with it or releases the key. This helps when you change your mind or stop to think halfway through typing with a key held down, or have some secondary automation running to trigger on particular chords.

I actually prefer keys to trigger only on key-up in most cases, so was thinking of putting a dummy secondary function on each key, but that would break a lot of things with the proposed updates..

kareltucek, 2018-10-01:
> The reason is that it doesn't make practical sense to set up a secondary role without a primary role because one can just use a primary role. It was confusing for some users.

I do see that, still it's a good example of how expecting a user to use a feature in some manner can make life difficult for those who decide to use it in a bit different manner.

> Chances are it's either due to the configuration serializer of Agent or the configuration parser of the firmware. In any case, it'll be a non-issue as soon as this option will be disallowed in Agent.

Thank you.

> I think the radio metaphor was quite undue. The UHK offers a lot, but you're after much more. 

This was not about features. It was about the fixing serializer via disabling the offending combination in agent (which I consider to be two different modules which should each work flawlessly on its own in the first place), and about you quietly skipping the point. It was definitely not meant in an offensive manner, but you are right that the choice of words was mean. I am sorry about that. 

> But then most users would be unhappy about how hard it is to customize their UHKs.

It's just that seeing the potential (of what actually is already implemented in UHK) being held back by your policy not to give a u user enough rope to hang himself feels super-irritating. All the necessary features (the small generic building blocks the power users strive after) are already present in UHK. Unfortunately, many of them seems to be just half implemented or to impose some expectation of how they are to be used, which means that many sensible combinations cannot be achieved even though they seem to be perfectly possible within the set of features you *do* provide... 

> even if I write such comments at Sunday midnight, which I'm just doing

Well, if you are implying that you working at such hours should make me happy, let me assure you it will not. It is absolutely understandable that you need rest and if I may have a suggestion you should give it enough high priority not to end up writing such comments at Sunday midnight - even if it means that we get reply a few days later and that production and development will go slower. 

> For me the 'arguments' given here are only meant in the rhetorical way, to try and nudge disagreements a bit more to the 'power user' side, not to pick any fights.

+1 

Have a nice day!

kareltucek, 2018-10-01:
uhm...

> the configuration parser of the firmware

Configuration parser of the firmware would not propagate the problem back to the Agent, would it?

mondalaci, 2018-10-01:
@kareltucek No, it wouldn't.