UltimateHackingKeyboard/firmware #181: 's' char typed with delay during macro playback

p4elkin, 2018-10-04:
I've been trying to create a macro action that would open spotify via spotlight search and stumbled upon a weird issue: the spotlight is opened just fine via key press/release macro instructions, but the 'type text' command yields 'potifys' instead of 'spotify'. Oddly enough - it looks like the issue is only with 's' character. I do not have any additional magic attached to the 's' key if it matters.

here's the macro config (i intentionally added a 2s delay before typing to eliminate possible timing issues):
[image]

 here's what spotlight shows after all:
[image]

mondalaci, 2018-10-04:
Thanks for your report and the nice words! @eltang will look into it.

eltang, 2018-10-06:
I cannot reproduce this issue on macOS 10.14.

@p4elkin By the way, a tap action consists of a press and a release, so the release action in your macro is redundant and can be removed.

mondalaci, 2018-10-06:
Not only the release action is redundant, but the proper shortcut should be Cmd+Option+Space, not Ctrl+Option+Space. Works perfectly on my end. Closing this issue.

mondalaci, 2018-10-06:
By the way, I tested with the most recent 8.5.1 firmware.

closed 2018-10-06 by mondalaci

p4elkin, 2018-10-06:
> but the proper shortcut should be Cmd+Option+Space
@mondalaci the spotlight shortcut is modifiable in macos and it opens just fine with my macro, with the only issue that 's' chars are messed up. E.g. 'foosbar' is also typed as 'foobars' in my case.

@eltang thanks for heads up (though I was just playing around with tap/press/release variants to see if it helps).

I am ok with closing this if you can't reproduce it - maybe the issue is with my particular keyboard. 
BTW, another observation - if I assign the secondary role to 's', it primary role starts to be faulty ('s' keystroke misses and such). if I assign the same secondary role to e.g. 'a' - the experience with typing 'a' is still normal.

p4elkin, 2018-10-06:
@mondalaci I installed the latest firmware and everything's looking in order now. Thanks for fast answers and cooperation.