UltimateHackingKeyboard/firmware #182: Long Macro Playback Issue

WhereIsHopper, 2018-10-04:
Sometimes with longer macros, characters will randomly get skipped. It is not consistent. I suspect that this is not an issue the UHK, but with the application not being able to accept keystrokes as fast as the macro is playing them back. Perhaps we can have an option to control the macro playback speed. Then we could slow down the macro playback to a rate that is accepted by the application.

Look at the two macro playback examples below of the Gettysburg Address. Different letters are missing from each example. These examples were originally entered into a Thunderbird email program on Ubuntu and copied here. Ignore the lack of commas, that is a separate issue.

***** Macro Playback Example #1 *******
Four score and seven years ago our fathers brought forthn this continent a new nation coneived in Liberty and dedicated to the proposition that all men re created equal.

Now we are engaged in a great civil war testingwhether that nation or any nation so conceived and so dedicated can log endure. We are met on a great battle-field of that war. We have come to dedicae a portion of that field as a finl resting placefor those who here gave their lives that that nation might live. It is altogether fitting and proper thatwe should dothis.

But in a larger sense we can not dedicate we can not consecrate we can not hallow this ground. The brave men living ad dead who struggled here have consecrated it far above our poor power to addor detract. The world will little note nor long remember what we say here but it can never forget what they di here. It is for us the living rather to be dedicated here to the unfinished work which tey who fought here have thus far so nobly advanced. It is rather for us to be hre dedicated to the great task remaining before usthat from these honored dead we take increased devotion to that cause for which they gave the last full measure of devotionthat we here highly resolve that these dead shll not have died in vainthat thisnation uder God shall have a new birth of reedomand that government of the people by the people for the people shall not perish from the earth.

*****Macro Playback Example #2 *******
Four score and seven years ago ourfathers brought forth on this continent a new nation cnceived in Liberty and dedicated to the proposition that all men re cated equal.

Now we are engaged in a great civil war testing whether that nation or any nation soconceived and so dedicated can long endure. We are met on a great battle-field of that war. We hav come to dedicate a portion of that field as a final resting place for those who here gave their lives tha that nation might live. It is altogether fitting and proper tha we should do this.

But in a larger sense we can not dedicate we can not consecrate we can not hallow this ground. The brave men living and dead who struggled here have consecratd it far above ourpoor pwer to add or detrac. The world will little note nor long remember what we say her but it can never forget what they did here. It is for us the living rather to beddicated her to he unfinished work which they who fught here have thus far so nobly advanced. It is rather for us to be here dedicated tohe great task remaining before usthat from these honred dead wetake increased dvotion to that cause forwhich they gave the last full measure of devotionthat we herighly resolve that thesedead shll not have died in vainthat this nation under God shall have a nw birth of freedomand that government of the people by the people for the people shall not perish fom the earth.

mondalaci, 2018-10-04:
Thanks for your report! @eltang will look into it.

eltang, 2018-10-06:
I cannot reproduce this issue on macOS 10.14.

mondalaci, 2018-10-06:
I cannot reproduce this issue either. I consider this an application-specific issue which should be solved on the application-level instead of being worked around in the UHK firmware / Agent. I'm closing this issue.

closed 2018-10-06 by mondalaci