UltimateHackingKeyboard/firmware #183: Secondary role not activating when combined with mouse

igoralmeida, 2018-10-05:
Hi folks,

Got my UHK, loving it. I've configured the Mouse key (above left Shift) to be an Escape when tapped, and LCtrl as secondary role. It works fine, but I've noticed that it really only is secondary if something else in the keyboard happens, so I can't for example hold the key and scroll with the mouse (to zoom-in) because it still registers as an Escape when I let it go.

I understand it's probably not possible to do this outside of the OS, but I'm wondering if the secondary role could also be activated when long-pressed, that might help...

mondalaci, 2018-10-05:
Hi Igor,

I'm glad you love your UHK!

Unfortunately, what you're looking for is not possible. The activation of secondary roles is momentary by nature. I can't think of a way around it.

closed 2018-10-05 by mondalaci

kareltucek, 2018-10-05:
Timeouting primary role is already planned (agent #792), so switching to the secondary role after the timeout seems to make sense. 

I am not sure if it should be the default behaviour (I guess not) and I do not think that it should be a high priority issue, but I don't see any fundamental problem here. Am I missing something?

mondalaci, 2018-10-05:
This should be possible in the future indeed, just not now until the timeout is implemented.

igoralmeida, 2018-10-05:
Is there a "timeout implementation on firmware" issue I should be following as well? I might tackle this myself to get my feet wet, if I can find enough information around. Pointers welcome.

mondalaci, 2018-10-05:
I'm not sure, but in any case, this issue depends on Agent. First, we'll have to extend Agent to expose the timeout option, then modify the binary configuration format to include this setting. We'll implement this afterwards.