UltimateHackingKeyboard/firmware #184: Unable to Switch Out of Test Keymap

WhereIsHopper, 2018-10-05:
Once in the Test Keymap, I cannot get back into the QWERTY, Dvorak, or Colemak keymaps by pressing FN+1, FN+2, or FN+3. Keyboard power must be cycled to return to QWERTY mode.

According to the UHK Agent, the Test Keymap should allow one to return to the other Keymaps.

Running Firmware 8.4.5 on an Ubuntu system.

mondalaci, 2018-10-05:
Hi there!

The test keymap will be removed from future versions. Feel free to remove
it in Agent.

Have a great weekend!

closed 2018-10-05 by WhereIsHopper

dharkness, 2021-12-18:
How do I exit test mode? This doesn't help me because `Enter` is mapped to `+` in test mode, and I can't log in to get to the agent software. :(

mondalaci, 2021-12-18:
@dharkness Please see https://ultimatehackingkeyboard.com/knowledgebase?q=test

dharkness, 2021-12-18:
That page only tells me that I _entered_ test mode, not how to _exit_ it. Since I cannot log in, I can't delete that keymap. Is there no way to exit test mode without deleting the keymap?

pcooke9, 2021-12-19:
@dharkness  Physically unplugging/reconnecting the UHK returns it to the default keymap on my system.

mondalaci, 2021-12-19:
@pcooke9 Thank you for your answer! I amended the knowledgebase article to include this information.