UltimateHackingKeyboard/firmware #185: Ctrl key key downs, but doesn't key up in a certain configuration

owlowlgo, 2018-10-05:
Copied over from https://github.com/UltimateHackingKeyboard/agent/issues/801
--
Apologies in advance if this is not a bug, but I would like to describe some unexpected behavior with the following settings:

[image]

I have remapped the Ctrl key to "Ctrl + c", as a shortcut for Copy. Same with the Super key, but for Paste.

This works. However, it appears that the Ctrl key keys down, but a key up does not occur. This causes all consecutive keys to be pressed with the Ctrl key down. Hitting backspace somehow causes the Ctrl key to be released.

Oddly, this does not occur with another shortcut I assigned for "Ctrl + Space". (Switch between English and CJK language input)

[image]

My OS is Ubuntu, if that matters.

Thank you.

eltang, 2018-10-06:
I cannot reproduce this issue on macOS 10.14.

@owlowlgo Are you running the latest firmware?

mondalaci, 2018-10-06:
I could only reproduce this with Linux so far. Firmware 8.5.1.

mondalaci, 2018-10-07:
I can actually reproduce this on both Linux and Windows, but not on Mac.

eltang, 2018-10-07:
I'm not sure why Linux and Windows would be affected, but not Mac. This could be a side effect of the sticky modifier behavior that's gone unnoticed until now.

owlowlgo, 2018-10-07:
@eltang Yes, 8.4.5.

coxley, 2018-10-18:
I've discovered similar behavior but it's not consistent. This is with secondary role: `ESC` on tap, `CTRL` on hold.

Sometimes `CTRL` will become stuck and hitting another key (I usually do `SPACE`) releases it.

Firmware: 8.5.2

[image]

mondalaci, 2018-10-18:
@coxley Which OS?

owlstead, 2018-10-18:
I've possibly got similar behavior. When I program my Y key to be Windows-Tab (for an overview of the open windows) then it presses the Windows button again once I move my mouse in mouse mode (the Windows menu gets opened). This could also be a different bug, but I imagine it could well be related.

mondalaci, 2018-10-18:
@owlowlgo Thanks for your report and for the nice words! We'll look into all of these cases upon resolving this bug.

coxley, 2018-10-19:
@mondalaci Linux

```
$ uname -a
Linux stretch 4.18.6-arch1-1-ARCH #1 SMP PREEMPT Wed Sep 5 11:54:09 UTC 2018 x86_64 GNU/Linux

```

demonsweatlive, 2019-03-08:
I've also run into this issue in Windows 10.  I created a new keymap to use in Visual Studio.  I assigned the '4' key to be 'ctrl + F4' in the base layer, in order to easily close windows.  After hitting it, ctrl is stuck.  I have to...do something else to un-stick it (the only thing I've discovered/used to unstick it thus far is to hit the '3' key, which I've also re-mapped in that layer to 'F3').

BTW, I just received the keyboard, and I'm absolutely loving it after just 4 days.  I've been looking forward to getting it for months, and I haven't been disappointed.

kgnn-io, 2019-03-09:
@demonsweatlive
I was able to work around this issue by implementing the same key combinations as macros.

I haven't encountered this issue ever since.

owlowlgo, 2019-03-09:
@demonsweatlive
I was able to work around this issue by implementing the same key combinations as macros.

I haven't encountered this issue ever since.

demonsweatlive, 2019-03-09:
Good tip, thank you.  That should do until this gets fixed.

kareltucek, 2019-03-09:
Btw, has anyone tried to reproduce this with 8.5.4. firmware? (@demonsweatlive's report sounds simply as the behaviour before 8.5.4., i.e., quite different problem. Regarding the op's issue, it sounds weird, so i think that retesting with 8.5.4. might be worthwhile.)

I didn't manage to reproduce this under linux (Xorg (1.20.4) without display manager + i3wm + modified UHK firmware (but set to configuration which should yield identical output)).

demonsweatlive, 2019-03-11:
I'm not sure how my problem was any different than the original, but I did update firmware to 8.5.4, and it does now seem to be fixed.

kareltucek, 2019-03-11:
> I'm not sure how my problem was any different than the original

> However, it appears that the Ctrl key keys down, but a key up does not occur. This causes all consecutive keys to be pressed with the Ctrl key down. Hitting backspace somehow causes the Ctrl key to be released.

If I understand both reports correctly, the original implies that Ctrl is becoming stuck and affecting several next  keystrokes of the UHK, while in your case, the Ctrl is being pressed just until the next action (i.e., can affect keystrokes sent from a different keyboard or mouse, but not the next action of UHK). The former behaviour is indeed wrong if it is real. The latter behaviour is a feature called "sticky modifiers" and was introduced so that shortcuts like `alt+tab` still work when bound to a single key. The 8.5.4. refactored the behaviour so that the "sticky modifiers" function is active only within "held layers" (i.e., not in the base layer and not in toggled layers), which should effectively prevent unwanted side effects.

demonsweatlive, 2019-03-11:
Okay, I guess I was causing confusion by trying to be more precise.  I took my behavior to match the OP exactly: Ctrl seemed to be stuck, affecting most subsequent UHK and mouse activity.  I was simply making a distinction between the different affected keys (Ctrl vs 4), Keypress settings (Ctrl + C vs Ctrl + F4), and keypress (scancode?) to resolve the situation (Backspace vs 3/F3), in order to more easily isolate the problem.  

Although, in re-reading, I neglected the fact that the Ctrl key was the one that the OP was re-mapping, which could cause some ambiguity in the actual problem (Ctrl key vs. Keypress w/LCtrl modifier).

At any rate, resolved for my case.

kareltucek, 2019-12-26:
@mondalaci I think that based on age of this ticket and demonsweatlive's feedback, this can be closed.

closed 2019-12-26 by mondalaci