UltimateHackingKeyboard/firmware #187: Alphabetic

p4elkin, 2018-10-08:


luteijn, 2018-10-08:
Using Space with shift as secondary function and will occasionally get something run togeter likeThis when typing quickly. Perhaps it would be better if there was some sort of setting to delay going to 2nd role for x ms. Of course, ideal settings would differ for different people.

p4elkin, 2018-10-08:
Another aspect of this is the following: in my IDE I use alt + Enter shortcut extensively. On my UHK alt is facilitated via secondary-role, at the same time Enter key has a secondary role assigned (mouse layer). **So, when I hold ALT, tapping ENTER triggers its secondary role". (Should this be extracted into a separate issue?).

@luteijn I agree that the simplest solution to this (in ideal world) would be a threshold delay before the secondary role kicks in.

Maybe the maintainers could give some initial pointer at the code base part to look at for investigation and experiments?

p4elkin, 2018-10-10:
> A workaround I'd consider is to use a least frequently used key.

FWIW - it is just so much nicer to be able to trigger MOD and MOUSE layers without having to leave the home row compared to e.g. having to hold the Mouse key with the pinky all the time (mine gets quite sore). And it somehow did work quite non intrusively in Karabiner (not the Elements version ofc). 

I personally think that the 'secondary role' feature loses a lot with this limitation, but I'm no PO to push it into backlog =)

> feel free to search for "secondaryRole", perferably in a case insensitive fashion. 

thanks, I've done some git log grepping and checked out this awesome blogpost of yours (https://www.toptal.com/embedded/from-the-ground-up-how-i-built-the-developers-dream-keybooard) and things are much clearer now.

kareltucek, 2018-10-22:
I am also interested in this. I find the idea of using home row keys as modifiers quite interesting since it solves multiple ergonomy problems I've been struggling with. However, I am skeptical about predictability of behaviour of such delay mechanism too...

Regarding the PR (it seems I cannot comment on the PR itself)... Could you publish a firmware release package for testing please? (Run /firmware/scripts/make-release.js package via node, then upload the uhk-firmware...tar.bz2 from the same directory.)

p4elkin, 2018-11-01:
I've created a pre-release of the modified algorithm in my fork - feel free to try it out (more details in description there):

https://github.com/p4elkin/firmware/releases/tag/8.5.3.187-1

p4elkin, 2018-11-02:
thanks for the elaborate report! Most of the points do ring the bell and I was planning to invetigate them as next steps. With these logs that should be easier to test/reproduce/debug

kareltucek, 2018-11-03:
- writing two characters which both have some secondary role produces twice the second letter. E.g., with secondaries on "s" and "d" keys, writing "speeds" results in "speess"

p4elkin, 2018-11-03:
> writing two characters which both have some secondary role produces twice the second letter. 

yeah, I mentioned that as a known issue - need to see how to mitigate that. Might be a tad tricky, but should be doable.

p4elkin, 2018-11-12:
Just released yet another iteration which contains a rather refactored version of the previous release plus a couple of fixes for annoying issues. 

If nothing major pops-up, the next thing would be to come up with some good strategy for key suppression.

kareltucek, 2018-11-12:
Seems good. Actually you seem to have solved some problems I've had with the firmware since last release (probably lowered some hard coded timeouts?).

One new (?) problem I've noticed - writing "user" results in "usre" when wrote fast, while having secondary on "s". While the action key is being postponed for a brief moment, the "r" is not.

p4elkin, 2018-11-12:
> One new (?) problem I've noticed - writing "user" results in "usre" when wrote fast

yeah, indeed can confirm - there is some quirky logic related to enforcing the order of the sent keys (so that the deferred are sent first) and I kinda cut one corner in there yesterday thinking that it will not affect the other functionality. Will look into it asap.

> Actually you seem to have solved some problems I've had with the firmware since last release (probably lowered some hard coded timeouts?)

Nope, those weren't changed - just did some refactoring and tried to take care of the state mgmt a bit more carefully (though obviously there's still stuff to do).

p4elkin, 2018-11-12:
uploaded a new artifact that has a tentative fix for the ordering issue. I will take a closer look when I've more time tho.

https://github.com/p4elkin/firmware/releases/edit/8.5.3.187-3