UltimateHackingKeyboard/firmware #188: Secondary roles buggy in conjunction with mouse clicks

ajh3, 2018-10-09:
I have my Shift keys set to send parentheses, with Shift as a secondary role, as described in UltimateHackingKeyboard/agent#804. The 8.5.1 firmware solved the earlier problem I was having with that, but I noticed a new issue. 

(I am not certain whether this is the _technically_ the fault of the UHK or my OS, but regardless, it may be something you want to address either way.)

**When the Shift keys are set this way, shift-clicking (for example to select multiple items in a file browser), does not work, and behavior becomes very erratic, usually requiring me to close the window and start over.** I cannot tell exactly what is happening but it appears that shift is being stuck in a pressed state, among other possible issues. 

I do not have another machine/OS readily available to test further, unfortunately. I am using Xubuntu 18.04 with Thunar file manager, although I suspect this problem will occur on other machines.

mondalaci, 2018-10-09:
Closing this issue because you did not update to the latest firmware as instructed upon opening this issue.

closed 2018-10-09 by mondalaci

ajh3, 2018-10-09:
Ok. Sorry. I updated.

The issue still exists on 8.5.2 and I figured out the cause, which is that the primary scancode is sent when the key is released, regardless of how long the key is held or what happens in between keydown and keyup. Meaning, I hold shift, click to select files, and then "(" is typed when I release shift even if it's several seconds later.

Since this probably breaks shift-clicking in many applications, and possibly other use cases as well, I propose some kind of timeout. I believe this is how QMK handles it; the primary scancode is only sent if the key is released within 0.5 or 1 second or so.

mondalaci, 2018-10-09:
This is essentially the duplicate of https://github.com/UltimateHackingKeyboard/agent/issues/679, so I'm keeping this closed.

The dual role feature works as expected. The secondary role should only kick in while another key of the UHK is pressed while holding the original key.

As for the stuck issue, it shouldn't ever happen. Please let me know if you can reproduce it consistently.

ajh3, 2018-10-09:
Ok, sorry, I tried searching for an existing issue before opening this one but I was searching for the wrong thing since I did not realize the root cause.

Regarding the stuck issue: to clarify - it looks to just be a consequence of the "(" being sent, it's causing Thunar to freak out. It does not look like there is any further bug. Once the timeout is implemented, my issue should be fully resolved. Thanks.