UltimateHackingKeyboard/firmware #189: Intermittent USB stop working

irwand, 2018-10-09:
I've had this intermittent problem for a while. I think I confused the problem with intermittent "Freeze Bug" https://github.com/UltimateHackingKeyboard/firmware/issues/172 .
I'm on Firmware 8.5.1, that has the Freeze Bug fix. Just now my keyboard no longer types. Agent can't communicate with it (It says "Can't find UHK, please plug it in" or something like that). At this point, I can still  change the keymap and the LED display changed. All the mod lights are still working when I click on different mods.
It's as if the firmware is still chugging along, but the USB connection to the host machine is dead.
Has anyone else encountered this before?

irwand, 2018-10-09:
I understand the spirit of that rule. But by definition, nobody can submit an intermittent bug that's hard to reproduce just because there's always newer firmware? I was trying out 8.5.1 for a while, trying to see if it could reproduce this problem I'm seeing, since firmware 8.2.5. I don't think it's fair to close this issue just because there's 1 more new firmware version, that has a commit that isn't related to the problem in question here.

mondalaci, 2018-10-09:
Sorry, I mistakenly assumed that your current firmware version doesn't have the freeze fix, but it clearly does. Did you only experience this issue only on 8.5.1? Which OS?

irwand, 2018-10-09:
No, I think I've encountered this problem on earlier firmware too, since 8.2.5. This happens on Windows. I never shut off my machine, I always only put it to sleep, in which case the USB port keeps being powered. I haven't discern any pattern yet to reproduce the problem. It's rare that it happens, but I've seen it a few times. Average maybe about once a month? maybe less.
I'm curious if it's just me or if there are others who have seen this problem.

mondalaci, 2018-10-09:
1. Did this also happen on 8.2.5?
2. When you switch layers, does the LED display get updated accordingly?

irwand, 2018-10-09:
@mondalaci 
1. I believe so, yes
2. Yes, the keyboard still acting normal as far as LED display getting updated accordingly and keymap changes are reflected on the 3 letter LED segment.
Unplugging and re-plugging USB fixes the prob.

mondalaci, 2018-10-09:
According to both of your answers, the bug you're encountering with is not (the variant) of the freeze bug that I fixed.

1. What's your hardware spec?
2. Can you reproduce this with other computers?
3. Do you use a USB hub / docking station / any device sitting in between?

irwand, 2018-10-09:
@mondalaci 
1. Dell Precision Tower 5810, Intel Xeon E5-1620, 16GB RAM, 64-bit OS. Has USB3, host controller: https://pci-ids.ucw.cz/read/PC/8086/8d31
2. I don't know. It's always plugged into this machine
3. No USB hub. It's plugged straight to the port on my tower.

mondalaci, 2018-10-09:
Which OS exactly?

irwand, 2018-10-09:
Sorry Windows 10 Enterprise. Version 1803 (OS Build 17134.285)

mondalaci, 2018-10-09:
Is there a way to query some kind of USB related system log when this happens to see what went wrong from the perspective of the OS?

irwand, 2018-10-09:
Sorry, I'm not aware of any log like that, specially on Windows

mondalaci, 2018-10-10:
Well, the next time it happens, please at least take a look at the device manager. It'd be useful to get more info on this bug, like the state of the UHK USB interfaces. These should be visible in the device manager. Screenshots would be useful.

mondalaci, 2018-10-25:
Just a heads-up: firmware 8.5.3 may improve things for this issue, so it's worth upgrading.

ManuelLevi, 2018-10-27:
Hi @mondalaci , it seems to be working fine now. 
Because I use two different OSX machines, I can't tell for sure if it was the upgrade or the restart, in any case it seems to be working!

mondalaci, 2018-10-28:
@irwand How is firmware 8.5.3 working out for you?

irwand, 2018-10-28:
I haven't seen it frozen yet, so that's great. But this problem is rare happening to begin with. I don't know if it's "fixed". You can close this issue for now I suppose. If it happens again, I'll post here. Thanks man!

mondalaci, 2018-10-28:
Awesome! Feel free to follow up if anything goes wrong.

closed 2018-10-28 by mondalaci

irwand, 2019-01-24:
OK, I just saw it hanging again just now. I'm on 8.5.3. Keyboard stopped working, but mod, fn, and mouse layer led did light up when I press them.
It hasn't happened for the past 3 months, so that's pretty good.. :-)
It seems to me that there's a bug lurking down there with the USB communication part maybe.. given that the keyboard itself seems to behave normally (the led still works).
Just FYI for additional data here. We don't need to re-open or anything.. It's damn hard to reproduce.

mondalaci, 2019-01-25:
Thanks for the heads up!

I think chances are good that this is a hardware issue of some sort, because we haven't received such a report in months. Please feel free to shoot us an email to support at ultimatehackingkeyboard dot com if you want your keyboard to be replaced. Every UHK should be rock stable.

irwand, 2019-01-29:
I was thinking about this on and off for the past couple of days. What happened when it started happening again for me was that I noticed my usb cable was tangled up. So I unplugged it and plugged it back in immediately. Not long after that, I encountered this issue. So I, again, unplugged it and plugged it back in immediately. Again, Not long after that, my UHK was misbehaving. It stopped working and spamming 'h' across.

I thought that maybe there's something that's not discharged from UHK when I unplugged and plugged it back in immediately. This time I unplugged it, and let it rest for a bit. I also pressed fn, mod, mouse a couple of times to hopefully get whatever leftover charge to discharge, since those 3 keys turn on the LED (they didn't turn on, as expected). Then I plugged it back in. I haven't encountered this problem again for the past 5 days.

So I'm going to keep monitoring this issue.
@mondalaci , thanks for the offer, but I think I'm going to keep my uhk and monitor this issue further.