UltimateHackingKeyboard/firmware #190: Firmware update failed

reardencode, 2018-10-13:
Start flashing firmware
[16:08:43.706] Agent version: 1.2.11
[16:08:43.713] [UhkHidDevice] Available devices unchanged
[16:08:43.716] [UhkHidDevice] Used device: {"vendorId":7504,"productId":24866,"path":"0001:0039:00","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":0}
[16:08:43.717] [UhkHidDevice] USB[W]: 09
[16:08:43.735] [UhkHidDevice] USB[R]: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:08:43.736] [DeviceOperation] USB[T]: Read left module version information
[16:08:43.737] [UhkHidDevice] USB[W]: 0e 01 00
[16:08:43.747] [UhkHidDevice] USB[R]: 00 01 04 00 00 00 00 00 08 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:08:43.748] [DeviceOperation] USB[T]: Read right module version information
[16:08:43.749] [UhkHidDevice] USB[W]: 00 01
[16:08:43.757] [UhkHidDevice] USB[R]: 00 08 00 02 00 05 00 04 00 03 00 01 00 04 00 00 00 00 00 04 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:08:43.758] Device right firmware version: 8.2.5
[16:08:43.759] Device left firmware version: 8.2.5
[16:08:44.184] New firmware version: 8.5.2
[16:08:44.187] [UhkOperations] Operating system: Linux 4.14.74 x64
[16:08:44.188] [UhkOperations] Start flashing right firmware
[16:08:44.189] [UhkHidDevice] Start reenumeration, mode: Bootloader
[16:08:44.309] [UhkHidDevice] USB[T]: Enumerate device. Mode: Bootloader
[16:08:44.310] [UhkHidDevice] USB[W]: 01 00 88 13 00 00
[16:09:04.207] [UhkHidDevice] Could not find reenumerated device: Bootloader. Timeout
[16:09:05.064] [DeviceService] updateFirmware error { message: 'Could not reenumerate as Bootloader', stack: 'Error: Could not reenumerate as Bootloader\n at UhkHidDevice.<anonymous> (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:194:1)\n at Generator.next (<anonymous>)\n at fulfilled (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:4:42)\n at <anonymous>' }
[16:09:05.068] [DeviceService] Read hardware modules information failed Error: Unable to read from a closed HID device at HID.read (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/node_modules/node-hid/nodehid.js:65:11) at Promise (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:94:1) at Promise (<anonymous>) at UhkHidDevice.<anonymous> (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:89:1) at Generator.next (<anonymous>) at /home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:7:1 at Promise (<anonymous>) at __awaiter (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:3:1) at UhkHidDevice.write (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:88:1) at UhkHidDevice.<anonymous> (/home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:140:1)
[16:09:05.608] [DeviceService] Device connection state changed to: { bootloaderActive: false, connected: false, hasPermission: true }
Could not reenumerate as Bootloader

mondalaci, 2018-10-13:
Please include the output of `cat /etc/udev/rules.d/50-uhk60.rules` here.

reardencode, 2018-10-13:
I run NixOS, so here's what I put in my udev config:
```
# Ultimate Hacking Keyboard rules
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", OWNER="brandon"
```

reardencode, 2018-10-13:
Hmm, I thought I did use the exact settings that agent recommended, but regardless, I've updated to those found here:
https://github.com/UltimateHackingKeyboard/agent/blob/166834e46cd0c4d8160b494c8c8c8cf06f6ee991/rules/50-uhk60.rules

Except that I used the `input` group instead of `plugdev` on the later 2 lines because I don't have a `plugdev` group.

Output looks the same as before, last two lines:
```
16:39:24.482] [DeviceService] Device connection state changed to: { bootloaderActive: false, connected: false, hasPermission: true }
Could not reenumerate as Bootloader
```

mondalaci, 2018-10-13:
What's your hardware spec?

mondalaci, 2018-10-13:
Also, please give a try to every one of your USB ports without any other hubs and USB devices connected.

reardencode, 2018-10-13:
Thanks for the quick responses!

I'm on a ThinkPad X1 Yoga Gen 3.

I'll do some more experimenting with different ports and the udev setup later.

reardencode, 2018-10-13:
I forgot to mention: I think that after changing the udev settings the keyboard behaved slightly differently: before changing I got just a flicker on the mode LEDs, but now it flashes FTX (or similar) on the alphanumeric for a few milliseconds.

mondalaci, 2018-10-14:
The display flashes FTY for a moment which stands for factory keymap before loading the user keymap from the EEPROM. This is normal.

Please keep me in the loop, and possibly also look into your BIOS settings. This might be a hardware-specific issue.

reardencode, 2018-10-14:
Gotcha, thanks!

reardencode, 2018-10-14:
When the firmware write fails in this way, agent spams its logs with:
```
[06:48:02.286] [error] [ { TypeError: Cannot read property 'leftModuleInfo' of undefined
      at Object.eval [as updateRenderer] (ng:///SharedModule/DeviceFirmwareComponent.ngfactory.js:151:44)
      at Object.debugUpdateRenderer [as updateRenderer] (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64895:21)
      at checkAndUpdateView (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64009:14)
      at callViewAction (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64355:21)
      at execComponentViewsAction (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64287:13)
      at checkAndUpdateView (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64010:5)
      at callViewAction (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64355:21)
      at execEmbeddedViewsAction (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64313:17)
      at checkAndUpdateView (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64005:5)
      at callViewAction (file:///home/brandon/.cache/appimage-run/e06eef7718c3107aa8ecf5001f96e86582140fbf3d16f9b370f42266cb0c95b6/squashfs-root/app/resources/app.asar/renderer/vendor.bundle.js:64355:21)
    ngDebugContext: 
     DebugContext_ {
       view: [Object],
       nodeIndex: 15,
       nodeDef: [Object],
       elDef: [Object],
       elView: [Object] },
    ngErrorLogger: [Function: bound ] } ]
```
(actually filled 250 gigs of it overnight)

reardencode, 2018-10-14:
Problem seems to be with the agent detecting the UHK when it's in Bootloader mode, because it definitely does enter Bootloader mode per my dmesg:
```
[  833.464888] usb 1-3.2: USB disconnect, device number 18
[  833.807470] usb 1-3.2: new full-speed USB device number 19 using xhci_hcd
[  833.941458] usb 1-3.2: New USB device found, idVendor=1d50, idProduct=6120
[  833.941459] usb 1-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  833.941460] usb 1-3.2: Product: UHK Bootloader
[  833.941461] usb 1-3.2: Manufacturer: Ultimate Gadget Laboratories
[  833.943886] hid-generic 0003:1D50:6120.001F: hiddev98,hidraw3: USB HID v1.00 Device [Ultimate Gadget Laboratories UHK Bootloader] on usb-0000:00:14.0-3.2/input0
```

mondalaci, 2018-10-14:
Did you build Agent from source?

reardencode, 2018-10-14:
I did not. Haven't been able to build it successfully yet due to nixos oddness about pulling certain packages from github.

I have been reading the code, and it looks like the `node-hid.devices` call must not be getting the updated device list for some reason. I'll try running agent as root when I get back to the computer in an hour or two.

reardencode, 2018-10-14:
BTW, I really appreciate the care you put into making sure the UHK doesn't get stuck in bootloader mode.

reardencode, 2018-10-14:
Able to update firmware when running agent as root. I wonder what permission it needs to detect the changing devices that it doesn't need to do everything else.

Thanks!

closed 2018-10-14 by reardencode

mondalaci, 2018-10-14:
Glad it finally works on your side!

Agent has worked out of the box for everyone else who used the provided udev file so far. Sorry, but it's all I know, and I don't have a better idea.