UltimateHackingKeyboard/firmware #191: Make volume handling slower / more precise

owlstead, 2018-10-18:
Before submitting a new issue, make sure to do the following:

1. If you're using Karabiner Elements on your Mac, close it!
2. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
3. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
4. Try to reproduce the issue, and only report it if it still persists.

mondalaci, 2018-10-18:
There is nothing to do about this as far as the firmware is concerned. The OS handles this. This is purely an OS-specific setting.

closed 2018-10-18 by mondalaci

owlstead, 2018-10-18:
On my UHK when I use Fn + Up or Down the volume bar moves up way too fast. Even if I very shortly hit the volume up key it will jump up about 25 "units" or whatever they are. I'm on Windows 10 Home (yeah, sorry, I need the drivers). They should move up only 1 unit if possible, with a high repetition rate. Extra points for allowing the mouse speed changes to also change the speed of the volume button.

owlstead, 2018-10-18:
Possibly, but I never had this issue with any other keyboard, including a mechanical one. So there is something different about the UHK that makes the volume go up or down faster than normal. Possibly you could emulate a wheel instead of a pair of keys?

mondalaci, 2018-10-18:
Ah now, I remember! This was caused by #72 which was fixed a long time ago. Please update to the latest UHK firmware via Agent.

owlstead, 2018-10-18:
Oh, I downloaded the latest stable version I thought. This thing is brand new :) OK, I'll try. Thanks, and my apologies. I'll report only if this won't solve the issue.

mondalaci, 2018-10-18:
This is actually our fault. Agent should always auto-update to the latest firmware. It will behave this way, but we're not there yet. :)

owlstead, 2018-10-18:
Sorry, no, I should be on the latest firmware. I can flash it but the agent reports 8.2.5 which should be the latest, right?

mondalaci, 2018-10-18:
No. 8.5.2 is the latest. See https://github.com/UltimateHackingKeyboard/firmware/releases

owlstead, 2018-10-18:
Ah, those pesky digits at the wrong place. Thanks, ilt is working now. Would it be hard to make modifier keys work to adjust the volume up / down speed? I could create a feature request, but I won't say that it is high on my prio list ...

mondalaci, 2018-10-18:
The volume up and down keys don't have a speed. When you keep them pressed, the OS simply increases / decreases the volume at a given rate. Maybe you should create a macro that contains multiple volume up / volume down actions in sequence. (Macros work with the latest firmware, even though Agent displays that they aren't supported.)

owlstead, 2018-10-22:
That's an interesting idea! I'll check if I can do something like that to make the volume go up faster. Maybe it would be an idea to make the mouse speed buttons more generic "speed" buttons though for future functionality. Yeah, I found out that Macro's work fine now and I've already implemented one macro: Mod + Backspace now deletes the last word I typed! Thanks :)