UltimateHackingKeyboard/firmware #192: Debounce timestamps stored as uint8_t instead of uint32_t

p4elkin, 2018-10-21:
While fiddling with the firmware I've noticed that the ```CurrentTime``` is stored as uint32_t, which seems to be a correct type for milliseconds. However, the ```key_state_t#timestamp``` is declared as uint8_t which is just one byte and won't be able to accomodate anything larger than 255. 

Or am I missing smth here?

mondalaci, 2018-10-21:
According to my memory, our code is actually fine in this respect, but don't remember why.

@eltang Would you explain why?

p4elkin, 2018-10-21:
If I understand it correctly subtracting such a timestamp from the CurrentTime and casting back to uint8 will produce quite random results, making the debounce algorithm also quite random.

p4elkin, 2018-10-21:
UPD: looking further into it - as long as the debounce timeout stays within 0..255, the timeout tracking should be indeed calculated correctly.

closed 2018-10-21 by p4elkin