UltimateHackingKeyboard/firmware #194: Impossible to switch off Fn mode after switching key maps

owlstead, 2018-10-22:
Before submitting a new issue, make sure to do the following:

1. If you're using Karabiner Elements on your Mac, close it!
2. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
3. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
4. Try to reproduce the issue, and only report it if it still persists.

owlstead, 2018-10-22:
Reproduce:

 - create a button that *toggles* the Fn layer;
 - use this mode to switch to a map that only allows you to *activate* Fn layer;
 - now you can click Fn buttons all you want, but Fn mode won't switch off, making it impossible to ever get out of the Fn layer.

Simple resolution: allow Fn toggle to switch off if Fn active button is pressed. Furthermore, switch to the base layer once the map is changed.

Better resolution: have one *current* layer which is the base layer by default. This current layer can be changed by *toggling* a layer. The activate buttons temporarily switch away from this current layer. This allows you to switch to one layer to another and temporarily use functions from e.g. the Fn layer (volume up / down from the mouse layer).

mondalaci, 2018-10-22:
Thank you for reporting this issue! I understand that your suggested "better resolution" is more general, but I much prefer your "simple resolution" because it's simpler from a user standpoint, and sufficiently flexible for most, so we'll implement it.

owlstead, 2018-11-21:
This can be closed in favor of [issue #199](https://github.com/UltimateHackingKeyboard/firmware/issues/199) I suppose which adds way more detail.

kareltucek, 2018-11-21:
Oh, sorry for the duplicate, I missed the toggle/activate difference when reading this issue ;-)

owlstead, 2018-11-22:
Meh, I'm all in favor of more generalized fixes and sometimes dupes are hard to find. I'm closing this issue with extreme prejudice and I'll thumb up #199.

closed 2018-11-22 by owlstead