UltimateHackingKeyboard/firmware #196: LED display sometimes turns off / doesn't indicate anything anymore

kbdh, 2018-10-25:
I am using firmware 8.5.3, the OS is Debian 9.

Sometimes the LED display turns off while using the UHK. The LEDs don't lit up any more  to show mod / fn / mouse / selected keymap.

[Agent 1.2.11 from the releases page](https://github.com/UltimateHackingKeyboard/agent/releases) is still able to find the UHK when this happens, but usually the Agent built from source [(tree a4e3696)](https://github.com/UltimateHackingKeyboard/agent/tree/a4e3696078216bc787b682e3323d89a35a517a81) is not. Usually it is able to find the UHK only after reconnecting USB (reconnecting the halves is not sufficient). 

Today was the first time that the Agent build from source (still tree a4e3696) was able to find the UHK without reconnecting. I am not sure if running get-debug-info.js is useful for this issue, but just in case:

```
$ packages/usb/get-debug-info.js
dev 1d50 / 6122 : usage_page:6572, usage:6320
dev 1d50 / 6122 : usage_page:5978, usage:178
dev 1d50 / 6122 : usage_page:282b, usage:3a3f
dev 1d50 / 6122 : usage_page:6920, usage:2073
dev 1d50 / 6122 : usage_page:746e, usage:3233
I2cWatchdog:142583611 | I2cSlave:46439422 | I2cWatch:149147 | I2cRecovery:1847 | KeyMatrix:72674003 | UsbReport:70662675 | Time:16717450 | UsbGeneric:146 | UsbBasic:3608 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142593478 | I2cSlave:46442625 | I2cWatch:149157 | I2cRecovery:1847 | KeyMatrix:72678938 | UsbReport:70667610 | Time:16718603 | UsbGeneric:147 | UsbBasic:3608 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142603276 | I2cSlave:46445848 | I2cWatch:149167 | I2cRecovery:1847 | KeyMatrix:72683842 | UsbReport:70672514 | Time:16719747 | UsbGeneric:148 | UsbBasic:3608 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142613074 | I2cSlave:46449071 | I2cWatch:149177 | I2cRecovery:1847 | KeyMatrix:72688742 | UsbReport:70677414 | Time:16720875 | UsbGeneric:149 | UsbBasic:3608 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142622872 | I2cSlave:46452294 | I2cWatch:149187 | I2cRecovery:1847 | KeyMatrix:72693714 | UsbReport:70682253 | Time:16721991 | UsbGeneric:150 | UsbBasic:3613 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142632709 | I2cSlave:46455468 | I2cWatch:149197 | I2cRecovery:1847 | KeyMatrix:72698692 | UsbReport:70687129 | Time:16723131 | UsbGeneric:151 | UsbBasic:3618 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142642567 | I2cSlave:46458627 | I2cWatch:149207 | I2cRecovery:1847 | KeyMatrix:72703667 | UsbReport:70692024 | Time:16724260 | UsbGeneric:152 | UsbBasic:3622 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142652404 | I2cSlave:46461800 | I2cWatch:149217 | I2cRecovery:1847 | KeyMatrix:72708603 | UsbReport:70696938 | Time:16725402 | UsbGeneric:153 | UsbBasic:3624 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142662269 | I2cSlave:46464942 | I2cWatch:149228 | I2cRecovery:1847 | KeyMatrix:72713538 | UsbReport:70701873 | Time:16726528 | UsbGeneric:154 | UsbBasic:3624 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142672037 | I2cSlave:46468198 | I2cWatch:149238 | I2cRecovery:1847 | KeyMatrix:72718426 | UsbReport:70706760 | Time:16727655 | UsbGeneric:155 | UsbBasic:3625 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142681834 | I2cSlave:46471424 | I2cWatch:149248 | I2cRecovery:1847 | KeyMatrix:72723432 | UsbReport:70711570 | Time:16728795 | UsbGeneric:156 | UsbBasic:3632 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142691633 | I2cSlave:46474647 | I2cWatch:149258 | I2cRecovery:1847 | KeyMatrix:72728393 | UsbReport:70716418 | Time:16729944 | UsbGeneric:157 | UsbBasic:3637 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142701470 | I2cSlave:46477820 | I2cWatch:149268 | I2cRecovery:1847 | KeyMatrix:72733327 | UsbReport:70721325 | Time:16731050 | UsbGeneric:158 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142709838 | I2cSlave:46480448 | I2cWatch:149278 | I2cRecovery:1848 | KeyMatrix:72737669 | UsbReport:70725667 | Time:16732158 | UsbGeneric:159 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49646
I2cWatchdog:142717855 | I2cSlave:46482924 | I2cWatch:149288 | I2cRecovery:1849 | KeyMatrix:72743669 | UsbReport:70728327 | Time:16733231 | UsbGeneric:160 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49729
I2cWatchdog:142727770 | I2cSlave:46486059 | I2cWatch:149298 | I2cRecovery:1849 | KeyMatrix:72753061 | UsbReport:70729555 | Time:16734283 | UsbGeneric:161 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:49929
I2cWatchdog:142737539 | I2cSlave:46489314 | I2cWatch:149308 | I2cRecovery:1849 | KeyMatrix:72763389 | UsbReport:70729824 | Time:16735324 | UsbGeneric:162 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50181
I2cWatchdog:142745866 | I2cSlave:46491943 | I2cWatch:149318 | I2cRecovery:1850 | KeyMatrix:72772208 | UsbReport:70730220 | Time:16736491 | UsbGeneric:163 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50426
I2cWatchdog:142755704 | I2cSlave:46495115 | I2cWatch:149328 | I2cRecovery:1850 | KeyMatrix:72777271 | UsbReport:70735018 | Time:16737613 | UsbGeneric:164 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142765474 | I2cSlave:46498372 | I2cWatch:149338 | I2cRecovery:1850 | KeyMatrix:72782158 | UsbReport:70739905 | Time:16738736 | UsbGeneric:165 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142775272 | I2cSlave:46501593 | I2cWatch:149349 | I2cRecovery:1850 | KeyMatrix:72787059 | UsbReport:70744806 | Time:16739885 | UsbGeneric:166 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142785041 | I2cSlave:46504851 | I2cWatch:149359 | I2cRecovery:1850 | KeyMatrix:72791947 | UsbReport:70749694 | Time:16741027 | UsbGeneric:167 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142794878 | I2cSlave:46508024 | I2cWatch:149369 | I2cRecovery:1850 | KeyMatrix:72796868 | UsbReport:70754615 | Time:16742161 | UsbGeneric:168 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142804676 | I2cSlave:46511247 | I2cWatch:149379 | I2cRecovery:1850 | KeyMatrix:72801768 | UsbReport:70759515 | Time:16743288 | UsbGeneric:169 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142814482 | I2cSlave:46514457 | I2cWatch:149389 | I2cRecovery:1850 | KeyMatrix:72806673 | UsbReport:70764420 | Time:16744408 | UsbGeneric:170 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142822799 | I2cSlave:46517087 | I2cWatch:149399 | I2cRecovery:1851 | KeyMatrix:72810996 | UsbReport:70768743 | Time:16745537 | UsbGeneric:171 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142832702 | I2cSlave:46520159 | I2cWatch:149409 | I2cRecovery:1851 | KeyMatrix:72815947 | UsbReport:70773694 | Time:16746663 | UsbGeneric:172 | UsbBasic:3638 | UsbMedia:26 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142842532 | I2cSlave:46523348 | I2cWatch:149419 | I2cRecovery:1851 | KeyMatrix:72820908 | UsbReport:70778571 | Time:16747791 | UsbGeneric:173 | UsbBasic:3638 | UsbMedia:30 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142852376 | I2cSlave:46526505 | I2cWatch:149429 | I2cRecovery:1851 | KeyMatrix:72825873 | UsbReport:70783459 | Time:16748927 | UsbGeneric:174 | UsbBasic:3638 | UsbMedia:34 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142862205 | I2cSlave:46529696 | I2cWatch:149439 | I2cRecovery:1851 | KeyMatrix:72830845 | UsbReport:70788324 | Time:16750074 | UsbGeneric:175 | UsbBasic:3638 | UsbMedia:40 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142868822 | I2cSlave:46531691 | I2cWatch:149449 | I2cRecovery:1853 | KeyMatrix:72834541 | UsbReport:70791933 | Time:16751177 | UsbGeneric:176 | UsbBasic:3638 | UsbMedia:46 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142878621 | I2cSlave:46534914 | I2cWatch:149459 | I2cRecovery:1853 | KeyMatrix:72839442 | UsbReport:70796834 | Time:16752315 | UsbGeneric:177 | UsbBasic:3638 | UsbMedia:46 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142888438 | I2cSlave:46538100 | I2cWatch:149470 | I2cRecovery:1853 | KeyMatrix:72844352 | UsbReport:70801744 | Time:16753477 | UsbGeneric:178 | UsbBasic:3638 | UsbMedia:46 | UsbSystem:2 | UsbMouse:50433
I2cWatchdog:142898297 | I2cSlave:46541258 | I2cWatch:149480 | I2cRecovery:1853 | KeyMatrix:72849282 | UsbReport:70806674 | Time:16754613 | UsbGeneric:179 | UsbBasic:3638 | UsbMedia:46 | UsbSystem:2 | UsbMouse:50433
[... cut ...]
I2cWatchdog:143098743 | I2cSlave:46606361 | I2cWatch:149686 | I2cRecovery:1854 | KeyMatrix:72950243 | UsbReport:70906543 | Time:16777856 | UsbGeneric:185 | UsbBasic:3642 | UsbMedia:46 | UsbSystem:4 | UsbMouse:50433
I2cWatchdog:143108511 | I2cSlave:46609619 | I2cWatch:149697 | I2cRecovery:1854 | KeyMatrix:72955129 | UsbReport:70911429 | Time:16778986 | UsbGeneric:186 | UsbBasic:3642 | UsbMedia:46 | UsbSystem:4 | UsbMouse:50433
I2cWatchdog:143118309 | I2cSlave:46612845 | I2cWatch:149707 | I2cRecovery:1854 | KeyMatrix:72960035 | UsbReport:70916335 | Time:16780112 | UsbGeneric:187 | UsbBasic:3642 | UsbMedia:46 | UsbSystem:4 | UsbMouse:50433
```

mondalaci, 2018-10-25:
Did you upgrade your system to the [latest udev file](https://github.com/UltimateHackingKeyboard/agent/blob/ca74b0a76b1bb976eedef9ed475960b67354cbf4/rules/50-uhk60.rules) (located in /etc/udev/rules.d)? It's necessary for the yet to be released Agent version to work properly, and once you use the unreleased Agent, don't fall back to the old, released one because you'll have to reconnect your UHK (just as you experienced) due to a system bug.

1. When you display turns off, does it recover upon reconnecting the halves?
2. Did you capture the above output after the issue hit?
3. Does the I2cRecovery variable always keeps incrementing, even when this bug hasn't hit? It's a sign of an unreliable communication between the keyboard halves.
4. Does the pace of the I2cRecovery incrementing varies when you totally disconnect the bridge cable and merge the halves?

kbdh, 2018-10-25:
I used both the old and the unreleased Agent, so this may explain why the unreleased Agent wasn't able to connect.

1. Yes, it recovers upon reconnecting the halves.
2. Yes, I captured the output after the issue hit.
3. Yes, the I2cRecovery variable keeps incrementing when the halves are connected with the bridge cable (in a similar pace as the output above).
4. The i2cRecovery variable doesn't seem to increment when the halves are merged (I waited a few minutes, but no increment so far).

mondalaci, 2018-10-25:
According to 4. your bridge cable may be the culprit. Please give your UHK another try, but this time use the bridge cable of your other UHK.

kbdh, 2018-10-25:
I had the same idea and just tried that, but the result is similar - the pace of the I2cRecovery incrementing seems to be slower, though.

mondalaci, 2018-10-25:
This may be a hardware issue. How your other keyboard behaves in this respect?

kbdh, 2018-10-25:
The behavior is the same for both UHKs.

mondalaci, 2018-10-25:
Statistically speaking, this is very unusual. What's your hardware spec?

kbdh, 2018-10-25:
It is a [cirrus7 nimbus](https://www.cirrus7.com/blog/cirrus7-nimbus-release-plan-en) from 2013, Intel Core i5-3470T.

mondalaci, 2018-10-25:
I'm not exactly sure about the exact cause of the issue, but I think it your power supply may be noisy, and/or the USB ports may be underpowered. Have you ever experienced issues with USB peripherals? Are there such known issues of this product?

kbdh, 2018-10-25:
I haven't experienced issues with USB peripherals before. I will do some more testing tomorrow to see if all 8 USB ports behave in the same way. Anything else I could try?

mondalaci, 2018-10-25:
Good idea! There isn't anything else on my mind right now.

kbdh, 2018-10-26:
I tested both UHKs on all 8 USB ports (4 x USB2 and 4 x USB3) and they all behave in the same way:

The i2cRecovery variable doesn't increment when the halves are merged, but keeps incrementing when the halves are connected with the bridge cable.

I tested also another computer: the I2cRecovery variable incremented once from 0 to 1 on one of the UHKs (the one I tested first yesterday), but except of that it didn't increment within several minutes.

So the problem seems to depend on the computer. Unfortunately it is my main computer which has these problems, so I wonder if anyone has an idea what I could try to solve these problems?

mondalaci, 2018-10-26:
I have two ideas:
1. Can you try to plug your main computer into another power outlet? The power source itself might be the source of the problem.
2. Using a powered USB hub may help.

kbdh, 2018-10-27:
It seems I'll have to buy a new power adapter ...

I found one with the same technical data for testing and so far the I2cRecovery variable isn't incrementing any more. 

I will wait a few days to see if the problem with the LED display is gone now, and keep you informed about the result.

Thanks for your support!

kbdh, 2018-11-03:
The problem seems to be solved, it didn't occur again during the last 7 days.

closed 2018-11-03 by kbdh