UltimateHackingKeyboard/firmware #197: Double-sending of keystrokes with delay

Ploogle, 2018-11-07:
I've installed the latest Agent and firmware, still have this issue:

Recently, my keyboard has started to send certain keystrokes twice occasionally, with a delay. The most common results of this: (In order of frequency)

- "h" "g" keys sent twice. (With delay. If I type the next key fast enough, it will be sent before the repeated key. The most common word that triggers this: "thihs" instead of "this")
- Mod layer activated with a single keypress 
- "i" and "j" keys sent twice on the Mod layer.

Thhis (see!) has become more frequent in thhe (again!!) last week and didn't use to hhappen (!!!) at all. I don't think it's my typing as I have other keyboards I use without issue.

Ploogle, 2018-11-07:
I should note that sometimes I've seen the "h" key especially be sent with a very noticeable delay. Not just "seems like", but I have taken my ffinger off the key and saw another "h" show up. Sometimes it's instant, sometimes it's a noticeable beat before sending again.

Ploogle, 2018-11-07:
Upon further inspection it's almost definitely the cable. It got roughened up quite a bit with the first-run shell and has an exposed wire. I've ordered a new one and will close this issue.

closed 2018-11-07 by Ploogle

mondalaci, 2018-11-07:
Thanks for your report, @Ploogle! Any further issues, feel free to let us know.

orn1983, 2019-07-24:
@Ploogle Did replacing the cable fix your issue? Our of curiosity, which switch type did you order?

crvs, 2019-07-26:
I have had this issue for the past couple of months actually, and I am actually using the cable from my cooler master rapidfire (because it's longer). The cable works well with the original keyboard so that seems to not be the problem. I tried everything I could think of to get the problem to go away, to no avail.

crvs, 2019-07-26:
red switches also.

orn1983, 2019-07-26:
I have the same issue. Red switches too. My friend has brown and has no issues. I'm thinking it's a defective set of switches in the red batch. Upon returning from holiday the issue was much worse. I was able to fix it to a degree by blowing into the switches (for most of them) and pressing down on them really hard. But I'm thinking it's likely an issue with the red switches.

crvs, 2019-07-26:
Yeah, I also blew on all the switches just before going on vacation, but it didn't do any good. Still double pressing, and some flickering on some select keys (I notice mostly on Mod & Ctrl). I'm gonna try taking the keyboard home (I have it at work) and try it with the original cable to see if there may be something with the cable, but I don't expect it to work

mondalaci, 2019-07-26:
The cable doesn't affect key chatter. Contact cleaner spray should fix this issue. Please see [this article](https://ultimatehackingkeyboard.com/modding-and-repair/fixing-misbehaving-keys) and feel free to shoot an email to support@ultimatehackingkeyboard.com for the quickest follow-up.

orn1983, 2019-07-26:
Yeah I took it home and tried that as well but the same problem persisted. I am hesitant to send it in for service because it seems to be getting progressively worse (i.e. more and more keys affected as time goes on) and I don't want to lose it for weeks only to have it appear on new buttons.

Once it is a majority of switches I'll probably either change the switches myself or send it in for service. I'm considering switching to brown switches anyhow. My friend's UHK has brown and the Kailh brown are in our opinion nicer than the Cherry brown.

mondalaci, 2019-07-26:
@orn1983 Make sure to give a try to contact cleaner spray. It works surprisingly well according to our experience. Statistically speaking, chatter is a rare issue, but it does happen for some, unfortunately.

crvs, 2019-07-26:
@mondalaci can you specify what you mean by contact cleaner spray? I have tried with pressured air to no avail.

orn1983, 2019-07-26:
Thanks for the tip. I will try contact cleaner. Will report back on this thread for documentation purposes.

crvs, 2019-08-03:
@mondalaci just to set the record straight, I tried using contact cleaner and the problem seems to be solved.

mondalaci, 2019-08-03:
@crvs So glad to hear!

orn1983, 2019-08-07:
It seems that my problem has been resolved as well after using contact cleaner! Thanks for the tip :)

orn1983, 2019-12-16:
Just a slight update for anyone else experiencing this; these switches (Kailh red) seem to be very vulnerable to corrosion. They start getting out of whack about 2 months after a thorough spray treatment, and I have to redo the procedure to keep them behaving.  They work perfectly after each cleaning procedure.

I had another mechanical keyboard with cherry switches in the same environment for years without issues. Seeing as how these switches are very commonly used, this is probably just a problem with a particular production batch. At least, a friend of mine has had his UHK with Kailh brown switches since February with no issues whatsoever.

crvs, 2019-12-17:
@orn1983 Yes, I have experienced the same, I had to do a second cleaning a couple of months ago and it's starting to happen again.

orn1983, 2019-12-17:
@crvs Sounds like the exact same experience I have. Out of curiosity, what was your order number? Mine was #904 according to my invoice. I'm thinking this has to be a production fault in the switches.

mondalaci, 2019-12-17:
@orn1983 @crvs Very sorry to hear about your experiences! According to the feedback we've received so far, this issue affects 1.5% of all UHKs, and the statistical distribution of the affected switches reflects the UHK sales. In other words, this issue is not specific to Cherry or Kalih switches. We haven't yet to figured out the root cause of this issue because we haven't been able to reproduce it yet. Feel free to shoot us a mail to support@UltimateHackingKeyboard.com if you want to get your UHKs replaced.

crvs, 2019-12-17:
@mondalaci Right now it's not double-sending very often but if experience means anything it should be difficult to ignore in a month or so, next time it reaches the stage where the problem is clearly reproducible I'll send it back in and tell you which keys are the problem so it will hopefully be some help then :)
@orn1983 seems that it's order #23744 according to the crowdsupply invoice it was shipped on June 20th 2018.

orn1983, 2020-03-24:
FYI, I haven't sent mine in for replacement yet as I haven't found a suitable time to be without it.
Some of the keys had started misbehaving quite often (which I didn't use contact cleaner to fix this time because I was going to send it in for replacement).
However, I just updated the agent and firmware to version 8.8.1 (not because I was trying to solve my problems), and it seems that this new firmware has significant debounce improvements. I confirmed by looking at GitHub issues that some improvements have been made recently to this functionality.

Anyway, I haven't had a double input issue all day today after upgrading. Might be worth upgrading for others experiencing this problem. This might at the very least provide a longer buffer between having to use contact cleaner, for those solving the issue that way.