UltimateHackingKeyboard/firmware #198: Can't use secondary key role as modifier on a different key that has a secondary role

daenney, 2018-11-14:
I'm trying to configure the UHK as follows:

- When I tap on a key, it should emit ( (so 9 with shift). I've also enabled Shift as a secondary role
- When I tap on the key it should emit [ (so configured for [{). LCtrl is configured as a secondary role

However, when I then hold the key to toggle the secondary Shift role and then press the key that emits [{, I don't get anything. I would expect that to emit {.

So this works:
[image]

Then I can get the { when I hold the shift and hit the other key.

But configured this way, it fails:
[image]

If I now hold shift and then tap the key above it, nothing happens.

I've uploaded my config here: https://gist.github.com/daenney/b73b1d61a18d168efdd9f36fda477802

mondalaci, 2018-11-14:
@daenney Thanks for reporting! I could reproduce this issue, and we'll look into it.

daenney, 2018-11-14:
I'm glad to hear that. Well, sorry it's causing you work but glad it might get fixed. If there's anything I can do to help test/debug this I'm happy to.

mondalaci, 2018-11-14:
We're dedicated to fixing issues, so please keep reporting anything you might find. We have everything we need regarding this one.

daenney, 2019-01-04:
I'm curious if this one is affected by #203?

mondalaci, 2019-01-04:
@daenney. No. #203 did not fix this issue.

daenney, 2019-01-05:
Ah, that's too bad, they felt related somehow.

Do you happen to have some pointers as to where I could look to fixing this one myself? It's probably not a super common problem for people but in my case it means I can't program my UHK the way I want to in order to replace my Ergodox as my daily driver. It's literally the only feature I miss, so if I can help out and speed up the process I'd be happy to invest some time into it. Just not really sure where to start.

mondalaci, 2019-01-05:
@daenney Thank you for considering to helping us! Look for `secondaryRole` in `right/src/usb_report_updater.c`. This code is not easy to deal with, and it will be refactored by @eltang.

p4elkin, 2019-03-19:
@daenney in the meantime I can suggest a firmware fork of mine that probably could help with your issue (if I got it correctly from the description): https://github.com/p4elkin/firmware/releases/tag/8.5.3.187-4

See the probably relevant readme part:
```
the algorithm tracks secondary role keys separately from each other, making it possible to e.g. have both ALT and CMD configured as secondary roles for some keys and have them activated at the same time (not possible in stock firmware);
```

Feel free to give it a shot, cheers.

daenney, 2019-03-26:
@p4elkin Would you not be willing to contribute these changes back? I don't see the algorithm changes you've made to cause any obvious regressions in current functionality.

p4elkin, 2019-03-26:
@daenney some discussion about that hapened in this issue: #207. In general - I'd be both hands up for that and planning to invest some time on making the changes as transparent and understandable as possible (hopefully those are not hard to comperehd apart from couple of gotchas/tricks).

However, it is eventually @mondalaci's and @eltang's call and I am pretty sure that they're quite busy atm esp with the add-on design going on.

jcowgar, 2020-01-03:
Good evening, I just took delivery of my UHK yesterday and this bug is affecting me quite a bit. I switched over to p4elkin's branch so this will work properly but would really like to be on the latest from UHK.

Any thoughts on when this might get fixed?

kareltucek, 2020-01-04:
Good evening! 

I expect this to be fixed within a few weeks if all goes well. Currently, I have quite a good idea on what to do with secondary roles, but need to look into refactoring options for layer switching first.

kareltucek, 2020-04-11:
Fixed in #265.

closed 2020-04-11 by kareltucek