UltimateHackingKeyboard/firmware #199: Layer deactivation

kareltucek, 2018-11-15:
Let's assume following mapping:

- left case button is mapped to "activate mod"
- right case button is mapped to "toggle mod"

Scenario:
- Single tap right button
- Single tap left button

Expected outcome:
- First tap toggles the layer
- Second tap untoggles the layer

Actual outcome:
- The first tap toggles the layer
- Second tap does nothing

Scenario no. 2:
- Doubletap left button
- Tap right button
- Tap right button again

Expected outcome:
- layer toggles
- layer untoggles
- layer toggles again

Actual outcome:
- layer toggles
- layer untoggles
- nothing

What is the reason for checking for `isLayerDoubleTapToggled` on line 214 in usb_report_updater.c? I see no good reason for that and I assume that it is the culprit.

owlstead, 2018-11-22:
Note that my bug report #194 contains possible ideas for fixes. This is a more generalized and better documented bug report so I closed mine.

piratk, 2018-12-12:
This annoys me as well. I would like to have these options:
* Tap to toggle layer
* Hold to enable layer
* Tap + 2nd button to toggle layer, for instance "Left space" for Mod layer and "Right case button" to toggle, then "Right case button" could deactivate Mod Layer.
* Using  same button as tap to toggle, holding and pressing any other button on layer would cause the layer to deactivate when tap to toggle button is released.

owlstead, 2018-12-12:
@piratk Having different ways of switching between layers should be posted as a separate feature request. And yes, for that feature request the switching back and forth between layers should be correct as well. You may want to provide a more detailed description or plan if you do. Let's first fix this issue though.

owlstead, 2019-02-11:
Note that I have to actively add an "Activate Fn" key to key maps with just toggle to make it even *possible* to switch off Fn, after switching to another key map using toggle.

Now I do understand that devs can be busy, but it has been easily over 2 months, quickly nearing 3 and multiple firmware releases and this question (including my previous one filed October 22th, but this *feels* like not being taken seriously, for something that seems a relatively easy to fix bug (if Fn is active and any Fn key is hit - be it toggle or activate, switch it off!).

kareltucek, 2020-04-24:
@mondalaci what is the intended behaviour of layer *deactivations*? 

- Should "standard toggle" action unlock "doubletap toggled" layer?
- Should (simple press of) "doubletaptoggle" action unlock "standard toggled" layer?
- Should "standard hold" action unlock "doubletap toggled" layer?
- Should "standard hold" action unlock "standard toggled" layer?

Current behaviour is`(sometimes, no, no, no)`.

This ticket assumes it should be at least `(yes, no, no, no)`. I am unsure if it shouldn't be `(yes, yes, no, no)` or even `(yes, yes, yes, yes)`.

mondalaci, 2020-04-24:
What does "standard hold" action mean?

kareltucek, 2020-04-24:
Agent-wise "Activate layer" without checked "Lock layer when doubletapping this key". Firmware-wise `SwitchLayerMode_Hold`.

mondalaci, 2020-04-24:
I see. All yes then.

kareltucek, 2020-04-24:
Alright!

closed 2020-04-29 by mondalaci