UltimateHackingKeyboard/firmware #201: Reconnected UHK doesn't send input until computer reboot

CalmDownMonkey, 2018-11-19:
If I unplug the UHK and plug it back in again, I can no longer type anything. The keyboard screen still lights up, and responds to changing keymaps, layer toggles, etc, but the computer never seems to receive anything (except from other keyboards).

The UHK Agent seems to know that the keyboard is plugged in successfully and it can communicate with it just fine.

The only way to resolve this issue seems to be restarting the computer with the UHK still plugged in. 
Strangely, if I take the UHK to another computer whilst it's in this bugged state, it doesn't type anything on those machines either.

I'm using Agent version 1.2.12 and firmware 8.5.3 on Windows 7, on a Dell Precision M4800.

It's not an issue someone is going to encounter frequently as most will leave it plugged in - but I encounter it frequently as I have to dock/undock my laptop a lot.

I'm not sure how to debug whether it is the computer that is causing this, or the keyboard/firmware itself. I'd gladly follow some debugging steps if anyone could help though.

Thanks!

mondalaci, 2018-11-19:
This is a known issue which we don't know how to fix. Windows 7 seems to be broken in this respect. Please see #73.

Any plans to update to a newer version of Windows? Windws 7 mainstream support has already ended.

I'm closing this issue, but feel free to follow up.

closed 2018-11-19 by mondalaci

CalmDownMonkey, 2018-11-19:
Ah I see, no problem. Yeah the Windows 7 thing is out of my control until my work decides to upgrade =D

owlstead, 2018-11-19:
You can see at #73 that simply closing the agent before reconnecting it should work around the issue. As all the state is kept in the keyboard / config files (at least, if you click "save to keyboard" of course) there is nothing that prevents you from closing the agent and restarting it when necessary.