UltimateHackingKeyboard/firmware #202: Caps Lock as primary role not working when secondary role is set to a layer

bhaeussermann, 2018-12-19:
When I set a key's scancode to "Caps Lock" and the secondary role to "Mouse", the mouse layer activates when I hold the key down, but tapping the key does not toggle caps lock.

[image]

Agent: 1.2.12
Firmware: 8.5.3

bhaeussermann, 2018-12-19:
An observation: when the key is set to "Caps Lock" with no secondary role, it toggles caps lock, but only when I press the key down long enough. Tapping it quickly yields no effect. Perhaps the delay required to toggle caps lock overlaps with the delay where the secondary role takes over.

I can't tell why caps lock should have this minimum delay to activate. Perhaps that is the actual bug.

mondalaci, 2018-12-19:
This feature works flawlessly on my side. Are you using the latest 8.5.3 firmware? If not, please update your UHK via Agent.

bhaeussermann, 2018-12-20:
I have the latest firmware (8.5.3) installed. It seems this does work as intended on Windows; the issue reported occurs on Mac OS.

bhaeussermann, 2018-12-20:
No :\

mondalaci, 2018-12-20:
Then I don't have an idea, but this should be an installation-specific issue on your side.

eltang, 2018-12-20:
As a way of filtering out accidental keypresses, macOS only activates Caps Lock if the length of the keypress is above a certain threshold. The firmware currently sends the shortest possible keypress, so a delay must be added to fix the problem.

mondalaci, 2018-12-20:
@eltang Haven't ever heard about this (mis)feature of OSX. Does it only pertain to Caps Lock?

mondalaci, 2018-12-21:
Ok, this was eye-opening. Thanks for the sharp insight, @eltang, and the further reference, @jfieber. I'm closing this, given that this is an OSX-specific behavior that I don't believe should be addressed on the firmware level. This can be worked around with a UHK macro.

mondalaci, 2018-12-21:
Slight correction: This cannot be worked around for dual-role keys, and this feature should be disabled in OSX.

closed 2018-12-21 by mondalaci

bhaeussermann, 2018-12-25:
Thanks! I found a way to disable the feature in Mac OS: [Karabiner Elements issues: "any reliable way to remove capslock delay on macos?"](https://github.com/tekezo/Karabiner-Elements/issues/1423). It's not ideal, but I couldn't find any other way to do it.

Now I have Karabiner handling the Caps Lock, but the Caps Lock LED is not lighting up, even though the "Manipulate LED" option in Karabiner is checked. However, this LED function does not work with my Microsoft Comfort Curve Keyboard either, so I guess I will give the UHK the benefit of the doubt on that one and not log it as an issue. At least my Caps Lock key is working now :).