UltimateHackingKeyboard/firmware #205: Left keyboard half sprays keys and display goes off

picaschaf, 2019-01-05:
A weird problem started to occour without any change before.

After 10-15 minutes the LEDs on the left half go out and the keyboard starts spraying keypresses every 1-2 minutes (mostly MOD-F, switch virtual desktop, it seems).

This can temporarily be fixed be reconnecting the keyboard, but comes again.

How can I provide additional info? System is macOS.

picaschaf, 2019-01-05:
Had only reconnect the keyboard half (not the entire keyboard) and hour ago, now I had suddenly mod fn and mouse LED light up for around 10 seconds (but not pressed a key, and the according function was also not active). Also the LED display jumped from QWM to QWW and back.

mondalaci, 2019-01-05:
1. Do you use the bridge cable that we provided, or a different one?
2. Does this issue happen if the halves are merged, and the bridge cable is disconnected?

picaschaf, 2019-01-05:
1. Yes, the provided bridge cable.
2. Haven't tested that yet. Doing it now.

ad 2) I tried to flash the keyboard again with the current firmware (same version as before, to exclude that maybe some bit flipped in the eeprom). That failed for the left half on macOS with a I2C error. I did also try flashing with the keyboard halfs merged, but that failed also. Flashing on Linux did work without any problem. The behavior on macOS is still the same after the flashing.

mondalaci, 2019-01-05:
This is not an EEPROM error because Agent can read the configuration of your UHK. (Firmware fashing won't affect the EEPROM, but the flash of the MCU.)

We're aware that firmware flashing fails on some hosts until https://github.com/UltimateHackingKeyboard/agent/issues/755 gets implemented.

Please follow up with 2.

picaschaf, 2019-01-05:
No weird behavior so far - merged everything is fine. Did plug the cable a few minutes ago in again and after typing some characters all LEDs flashed up for half a seconds. But typing function was normal. Now, same behavior as before.

While typing this message suddenly mod fn and mouse LED went on (typing worked still normal) and LED display showed QO. When I double pressed MOD everyhing went back to normal. I made a video of this behavior if you want it to see, just let me know.

Sadly I don't have a 4pin RJ cable around atm, but is there maybe something I can check on the cable? (Lab, equipment and some electronics knowlege available) But on the other side, as I've said, I havent't touched the cable much at all since I have the UHK.

Now, after typing this paragraph the left half stopped working while the LED display was still active. After reconnecting the left half, some keys get eaten away while typing.

mondalaci, 2019-01-05:
The root cause of this issue is probably a bad bridge cable. Can you possibly make or get made a bridge cable based on [these instructions](https://ultimatehackingkeyboard.com/knowledgebase/can-i-have-a-longer-bridge-cable)? It'd be much faster than us sending you a cable. We'd be glad to reimburse the cost of the replacement cable, or send you a cable if needed.

picaschaf, 2019-01-06:
Thank you for the offer! I'll try with shielding the cable first (have much electronics in dev lying around, maybe they spray EMF) and checking the cable itself for any defects more detailed. When I get to a store next week I'll try a different cable.

But I think after the investigation with your help we can more or less an issue with the firmware. Should I then close the issue?

PS. While writing this comment, no problem with the cable occoured so far. Currently any electronic dev on my desk is turned off. Maybe this is really the cause. Is it possible that the bridging cable is unshielded and the communication quite sensitive to EMF?

mondalaci, 2019-01-06:
Has your UHK been working well for a while, and then this issue suddenly happened in the same environment? If so, it's hardly an EMC issue, but a cable issue.

picaschaf, 2019-01-06:
> Has your UHK been working well for a while, and then this issue suddenly happened in the same environment? If so, it's hardly an EMC issue, but a cable issue.

More or less, yes - but a second dev board came onto my desk.
Strangely today I had no issue with the cable since my last comment so far. But nothing changed in comparison to yesterday. Nothing is not 100% correct, I have my MacBook Pro upside down in its vertical stand next to the display and keyboard. But I hardly believe that this could make any difference. :D

mondalaci, 2019-01-06:
Nothing is not 100% correct, indeed. :D

Please try to get a new bridge cable, and let me know how it works. Or alternatively, we can provide you one if needed, but again, it'll be slower than getting it locally.

mondalaci, 2019-01-28:
Closing this due to the lack of replies. Feel free to follow up, and I'll answer.

closed 2019-01-28 by mondalaci

picaschaf, 2019-02-05:
> Closing this due to the lack of replies. Feel free to follow up, and I'll answer.

Haven't had any issue since my last reply, not with the original cable and not with my custom made. Probably there were gremlins, or it really was some kind of stray EMF from one of our dev boards.
Thank you very much for investing time in that issue! 👍