UltimateHackingKeyboard/firmware #206: [question] flash write endurance / durability

orthoceros, 2019-01-10:
Hi, this is just a question, rather than an issue: How often is it safe to use "save to keyboard" in Agent? I mean, how often can the firmware flash be overwritten? Does Agent always overwrite the same flash cells when saving/updating my UHK's config or does it add some offset and rotate over a larger flash area to distribute the write load (like UEFI BIOSes do when saving)?

mondalaci, 2019-01-10:
Hey there! I originally planned to implement wear leveling, but concluded that it doesn't worth the effort given the 1,000,000 program/erase cycles of the [CAT24C256 EEPROM IC](https://www.onsemi.com/pub/Collateral/CAT24C256-D.PDF).

closed 2019-01-10 by mondalaci

owlstead, 2019-01-11:
It is not an issue for writing EEPROM firmware, but bad programming could still lead to issues of course. One forgotten loop that writes a variable to EEPROM and 1 M is small change. I'm a smartcard developer by profession, where these kind of issues are common.