UltimateHackingKeyboard/firmware #209: Start scroll timer on keydown

robobenklein, 2019-02-01:
Currently it's possible to quickly tap a scroll key and not have any scroll event registered, which I think is caused by the repeating scroll timer running independently of the keypress. (Or the trigger just not happening fast enough.)

Could it be changed so that pressing the key down will start the scroll repeat timer (and thus immediately relay a scroll event) instead of having to hold down the scroll key long enough for a scroll timer tick to occur?

Hopefully that's a clear enough explanation, but I can draw a diagram or something to help explain what it feels like is happening.

FW 8.5.3

[image]
For example, adding an option here to ensure that the scroll event happens at least once immediately on keydown.

mondalaci, 2019-02-05:
@robobenklein I agree that what you're suggesting is the correct behavior, and we'll fix the firmware to behave this way. Good catch!

robobenklein, 2019-11-05:
Just a bump that this is still the case in 8.6.0

Is this something that's relatively easy to fix or should I potentially dig into the code myself during my next break?

mondalaci, 2019-11-05:
The relevant code has not been changed in recent firmware releases.

See https://github.com/UltimateHackingKeyboard/firmware/blob/master/right/src/usb_report_updater.c#L60

There's a lot on our plate, and I can't help much. I'll be able to fix this pretty quickly once I get there.

mondalaci, 2019-12-24:
Fixed by #249.

closed 2019-12-24 by mondalaci