UltimateHackingKeyboard/firmware #214: Unable to enter capslock if layer-switch secondary role assigned

pilotclass, 2019-03-22:
I am trying to set up the `Mouse` key to work as Caps Lock _plus_ switch to the Mod layer when held down, but the Caps Lock isn't working.

Steps to reproduce:
- Agent version: 1.2.12
- Firmware version: 8.5.4
- Assign Scancode `Caps Lock` to `Mouse` key
- Try to toggle caps lock using the `Mouse` key - it works. 
- Assign Secondary role `Mod` to `Mouse` key
- Try to toggle caps lock using the `Mouse` key - it doesn't work any more.

mondalaci, 2019-03-22:
I've just tried your use case, and it works flawlessly on my side.

Do you happen to use a Mac with Karabiner Elements running? If so, please close Karabiner Elements and retry. Also, please try this on another computer.

I'm closing this issue, but feel free to follow up.

closed 2019-03-22 by mondalaci

pilotclass, 2019-03-28:
I am using a Mac, but I'm not using Karabiner Elements.  I don't have access to a non-mac right now as my keyboard is used at work and everyone uses a mac.

I've tested further and I can get similar behaviour to work with almost any key _other than capslock_.  So I can assign `NumLock` to the `Mouse` key with a secondary modifier and it works fine.  I can assign alphabetical characters, numbers, punctuation, enter, escape, function keys, pretty much anything _other than capslock_.

I believe that this is still a bug

mondalaci, 2019-03-28:
Please export your configuration with Agent and attach here. I'll look into it.

kareltucek, 2019-03-28:
I think that this problem was encountered before. If i remember correctly, the bottom line was that mac requires Caps Lock to be held for certain time in order to prevent accidental activations. I believe that eltang was the one to claimed it. However, keys with secondary actions are activated only for very brief moment.

pilotclass, 2019-03-28:
I've attached my config [UHK-PRW.json.zip](https://github.com/UltimateHackingKeyboard/firmware/files/3015861/UHK-PRW.json.zip).  I'm using keymap PRW.

That said, I think than @kareltucek has it right - I ran a visual keyboard to watch keystrokes and keys with secondary actions activated for incredibly short times.  If I hit my capslock key (on the Mac itself), I can see this same behaviour - capslock isn't registered.

I guess this is now a feature request?

mondalaci, 2019-03-28:
[This](https://apple.stackexchange.com/questions/81234/how-to-remove-caps-lock-delay-on-apple-macbook-pro-aluminum-keyboard-when-booted) is the cause of your issue. The UHK firmware is flawless in this respect. Please change your system settings.

pilotclass, 2019-04-01:
Yes, thank you.  Is there somewhere we can put a screenshot for other users?
[image]

pilotclass, 2019-04-01:
That's what it needs to look like.  Nothing faster than `Short` accepts the capslock from UHK.

mondalaci, 2019-04-01:
Thanks for the screenshot! We'll simply redirect affected users to this issue.