UltimateHackingKeyboard/firmware #218: need to release modifier when releasing one key

axaluss, 2019-04-17:
i mapped [mod+u] to [cmd+leftArrow] and [mod+l] to [rightArrow].
When pressing  [mod+u] and [mod+l] then [mod+l] without releasing [mod] it will map to  [cmd+rightArrow]. [cmd] is basically not released until releasing mod key. how can i configure this behavior to release [cmd] when releasing any of these keys: [mod+u].

In short: modifiers in mappings should be released when any of the keys in the mapping is released. Or: there should be the option to make this happen.

mondalaci, 2019-04-17:
This behavior is intentional because this is the only way the Mod+D -> Alt+Tab mapping works. You can work it around by creating macros instead of shortcuts.

closed 2019-04-17 by mondalaci

kareltucek, 2019-04-18:
If I remember correctly, sticky modifiers should still be released when different action is pressed, which is not the reported behaviour.

@axaluss my guesss is that you are either on an old firmware (up-to-date is 8.5.4.) or that you are running some software which messes the keys up.

mondalaci, 2019-04-18:
@axaluss I misunderstood your original description, and @kareltucek is totally right. The cause should be what he described.

axaluss, 2019-04-18:
You guys were right. I uninstalled Karabiner, but it was still running in the background and that was fucking up things. After i quit it everything was fine.