UltimateHackingKeyboard/firmware #219: Unable to wake up OS X

mrekucci, 2019-04-25:
I'm unable to wake up OS X from sleep after the time set in `Turn display off after` from `System Preferences -> Energy Saver` elapses (see the screenshot below). I'm able to do it with another keyboard.

Tested with the following HW:
macOS High Sierra Version 10.13.6
MacBook Pro (15-inch, 2017); 2,9 GHz Intel Core i7; 16 GB 2133 MHz LPDDR3

macOS Mojave Version 10.14.2
Mac mini (Late 2018); Intel Core i7-8700B; 8 GB 2666 MHz SODIM DDR4
 
UHK Firmware: 8.5.4
UHK Agent: 1.2.12

Tested with different values of `Turn display off after`.

[image]

mondalaci, 2019-05-02:
I can't reproduce this issue. I'm running OSX 10.9.5 on a 2.4 Ghz Intel Core 2 Duo MacBook Pro.

@eltang @Lauszus Any luck with reproducing this?

jcostlow, 2019-05-04:
I can duplicate this.

MacOS 10.14.4, Macbook Pro
UHK 8.5.3

The UHK is connected to a USB-C hub which also has another keyboard and an HDMI monitor plugged in. If just the screensaver goes off, the UHK operates normally and I have to enter the password. However if the machine falls asleep, UHK key presses do not wake the computer. Pressing any keys on the other keyboard does wake it.
I have not tried disconnecting the other keyboard yet.

mondalaci, 2019-05-04:
@mrekucci Do you use dongles / hubs for connecting the UHK to your computers? If so, which types?

mrekucci, 2019-05-04:
With the Mac Mini I don't. With the MacBook Pro I use [Apple USB-C Digital AV Multiport Adapter](https://www.apple.com/shop/product/MJ1K2AM/A/usb-c-digital-av-multiport-adapter?fnode=8b).

forficate, 2019-05-05:
Same issue.

Agent: 1.2.12
Firmware: 8.5.3
OSX: 10.14.4

Imac 27" 2017.

Keyboard is connected to USB port directly on the Mac. To wake up I need to move/click a mouse button, keyboard will not wake up the machine.

mondalaci, 2019-05-05:
Thanks for the reports! We'll look into this.

jcostlow, 2019-05-07:
Curiously enough, unplugging the other keyboard seems to have solved this problem.
The other keyboard is a 20 year old Microsoft Natural Keyboard Pro that is combo USB 1.1/PS/2, so it might be slightly misbehaving by todays standards. Not sure how/why this would interact with the UHK.

mondalaci, 2019-05-07:
@jcostlow Very interesting finding. Thanks for sharing!

@mrekucci @forficate In light of @jcostlow's finding, please unplug every other USB device and retry.

mrekucci, 2019-05-07:
On my Mac mini, I don't have any other devices connected except the UHK (connected directly to the Mac mini's USB) and monitor. The problem still persists.

jcostlow, 2019-05-07:
Correction: The keyboard still sometimes fails to wake the mac without the external keyboard connected. The UHK is still connected through the hub.

If the mac is asleep and the 'QWM' light on the UHK is on, then the mac will not wake with a keypress.
If the mac is asleep and the 'QWM' light on the UHK is off, then the mac will wake with a keypress.

ssbb, 2019-06-05:
Have the same issue. iMac Pro 2017, macOS 10.14.5

Lauszus, 2019-07-01:
Sorry for not looking into this sooner. 

Can somebody that can reproduce the issue try to change the value ```8``` to something else at this line: <https://github.com/UltimateHackingKeyboard/KSDK_2.0_MK22FN512xxx12/blob/9e7707006f38245b228c32823eca087cbd0a6ef3/middleware/usb_1.0.0/device/usb_device_khci.c#L1203>?

Try to both lower and increase the value. Does that make any difference?

mondalaci, 2019-07-02:
@Lauszus Thank you so much!

@mrekucci @jcostlow @forficate @ssbb I've rebuild the firmware with various (1, 2, 4, 8, 16, 32, 64, 128, 256) values. Please [download them](https://github.com/UltimateHackingKeyboard/firmware/files/3352374/firmwares.zip), and give every variant a try by flashing them via Agent. I'm excited to hear about your findings!

mrekucci, 2019-07-22:
@mondalaci I tested every firmware for a couple of days. But nothing has changed, the problem still persists.

mondalaci, 2019-07-22:
Thanks for your feedback, @mrekucci!

Any other ideas, @Lauszus?

jwr, 2020-01-17:
Just as a data point, I can't wake up my iMac 5K 2019 with the UHK either. The keyboard is plugged directly into a USB port in the back of the iMac. I have no other keyboards plugged in.

mrekucci, 2020-01-17:
Is the UHK team working on this issue?

mondalaci, 2020-01-18:
@mrekucci We aren't working on this issue right now. I'm trying to get a hold of @Lauszus to get it fixed.

Lauszus, 2020-01-21:
@mondalaci the problem is that I do not have a Mac anymore, but I can try to see if I can replicate it. Is the bug present in any other operating system?

mondalaci, 2020-01-22:
@Lauszus The bug is only present on recent Macs.

wjlroe, 2020-01-24:
I've just tried to wake up 2 Macs from sleep using the UHK (using firmware 8.7.0). I can wake up both an older Mac (MacBook Air from 2013) and a new MacBook Pro (15-inch, 2019) (that last one with a USB-A to USB-C adapter) and they wake up right away after being put to sleep when I type anything on the UHK. This is with nothing else connected. I've also just tried the newer Mac with a USB 3.0 hub and a USB-C dongle, both nothing prevents it from being woken from sleep from the UHK. This was a bit surprising to me as I'm pretty sure I've seen problems waking this Mac from sleep before. Possibly this is more an issue when the Mac goes into some deeper form of sleep (like hibernation), but I can't reproduce this bug on any Mac I have access to even with silly amounts of USB dongles and hubs. I'll see what happens with longer sleep cycles.

mondalaci, 2020-01-24:
@wjlroe Thanks so much! Your feedback is very much appreciated. Please list all the Macs you can test, describe their exact type and year, and also the brand and type of the USB hubs. Please try every combination if possible. Waiting for a while after putting Macs to sleep might help reproduce this issue. The more information we get, the easier it is to fix this issue.

I'm wondering whether Macs have USB 3.x and non-USB 3.x ports as they may behave differently in this respect.

mrekucci, 2020-01-24:
What @wjlroe wrote `... and they wake up right away after being put to sleep when I type anything on the UHK.` doesn't necessarily mean it's working. As I mentioned in the issue description: `I'm unable to wake up OS X from sleep after the time set in Turn display off after...`. I'm also able to wake up my OS X right away before the timer elapses. @wjlroe did you try it after the timer has elapsed?

mondalaci, 2020-10-17:
Everyone affected by this issue, please check out @xxxajk's firmware, which hopefully fixes this issue. Please let us know about the results.

[uhk-firmware-8.10.2-xxxajk.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/5396353/uhk-firmware-8.10.2-xxxajk.tar.gz)

mondalaci, 2020-10-25:
@mrekucci @jcostlow @forficate @ssbb @jwr Guys, I've been chasing @xxxajk for months to fix this bug. He finally found some time to implement a potential fix, which I attached in my previous comment. You just have to flash the firmware via Agent. So pretty please, test it, and let us know the result! I'd love to get this resolved and hate to see @xxxajk's efforts go to waste.

kour1er, 2020-10-25:
Just flashed it here and was unable to wake from sleep. This is on an iMac Pro running 10.15.7 with my UHK directly connected. Can I provide any further helpful information?

mondalaci, 2020-10-25:
@kour1er Thanks for your feedback and tip regarding firmware downloading!

@xxxajk Please feel free to join the conversation and ask for specifics. You can make the LED display of the UHK show useful debugging information to gather more specific feedback. @kareltucek can show you how.

mrekucci, 2020-11-09:
I've tested the firmware for a week, the problem still persists. Hardware [the same as before](https://github.com/UltimateHackingKeyboard/firmware/issues/219#issue-437193273), OS: macOS Catalina 10.15.5.

mondalaci, 2021-05-23:
Everyone affected, please test the just-released 8.10.10 firmware version as it contains numerous USB-related fixes and might resolve this issue.

kour1er, 2021-05-23:
@mondalaci well that's promising - just flashed to the 8.10.10 firmware and successfully woke from sleep. I will run more tests and an over night one and update with more results. Fingers crossed and congrats on making progress!

On a probably unrelated side note, after flashing the firmware, my bottom row left half of the keyboard 1.75u key (mapped to space) stopped working. I reloaded my layout map from Agent and it started working again. Don't know if this was a random thing but just thought I'd mention it.

mondalaci, 2021-05-23:
@kour1er Thanks for the quick feedback! Sounds very promising!

As for the unrelated side note, you probably hit #361.

kour1er, 2021-05-24:
Woke from sleep from overnight perfectly :) Also, as another piece of good news, the display on the left half of the keyboard also went to sleep (i.e. totally blank). I think you've solved the issue!

jwr, 2021-05-24:
I upgraded to 8.10.10 today and had a similar issue as @kour1er — some keys in the bottom row of the left half were remapped. Specifically, the 2nd and 4th keys from the left, as well as the frame key.

I had to reset the configuration and import it from backup, after doing this everything started working correctly again.

mondalaci, 2021-05-24:
Looks like this issue has been resolved, so I'm closing it. All hail @abcminiuser!

I'll look into the other issue mentioned.

closed 2021-05-24 by mondalaci