UltimateHackingKeyboard/firmware #220: USB Issue

timnash, 2019-04-25:
Hello,

I've had an issue happen a few times where the keyboard seems to disconnect from USB. The keyboard still seems to be working (I can change modes for example), but I'm not getting any input on screen. When I open Agent it gives me a "Can not write to HID Device" error message. Unplugging the UHK and plugging it back in seems to solve this issue. Right now it's happening for me roughly once every two weeks.

The UHK is plugged into a port on the machine, which is a Dell Latitude E7440 Laptop running Windows 10 1809. When the issue started occurring I moved the UHK to a different port, but have still had the same issue.

I am running the latest version of both Agent and the UHK Firmware.

This may be similar to #189.

mondalaci, 2019-04-25:
Hey there!

This might be a power supply related issue.

1. Have you tried to reproduce this with any other computer?
2. Do you always use / not use a charger when it happens?
3. Are you located at the same place when it happens?
4. What's your order id?

timnash, 2019-04-25:
1. I haven't tried with any other computer, but I could take my UHK home with me and see if it happens at home as well.
2. It's always connected to the charger.
3. Always in the same place (my UHK is attached to my desk chair at work).
4. I was Crowd Supply 44297, but this is my second UHK after the first one started sending a bunch of keys all at once.

mondalaci, 2019-04-28:
Thanks for your answers!

You're saying that your UHK is connected to your desk chair. I assume you're using an alternative bridge cable, not the one we provided. How long is this bridge cable? Is this a spiral cable? Would you make a picture of your setup?

timnash, 2019-04-30:
Yes, it's a straight cable I made myself. I'm not sure of the exact length. I believe it's around 6 feet. But when it stops working, neither half seems to send input to the PC. Would the cable affect both halves like that?
[image]

mondalaci, 2019-05-01:
If you can change keymaps then your bridge cable is probably fine.

I think this is a rare hardware related issue. Let me follow up in email very soon. That way I'll be able to help you quicker.

closed 2019-05-01 by mondalaci