UltimateHackingKeyboard/firmware #221: Include extra USB reports for shortcuts for maximum compatibility

mondalaci, 2019-04-30:
Currently, UHK shortcuts are sent via USB using the smallest number of USB reports possible. For example, in the case of Ctrl+C, both the Ctrl modifier and the scancode of C gets sent in the same report at once. This is correct, but it messes up some applications like Karabiner and Hyper-V. By holding and releasing modifiers in extra reports, this might be solved:

* Report 1: Hold Ctrl
* Report 2: Hold Ctrl + send the scancode of C
* Report 3: Hold Ctrl

Hopefully, this will put an end of the nonstop reports mostly regarding Karabiner and sometimes regarding Hyper-V. We shall see.

kareltucek, 2019-12-28:
I've just realized two small problems with the PR. One is fixed. Second one is caused by interference of secondary roles - I am afraid I am not able to fix it in a simple manner with current implementation of secondary roles, so I am leaving these as single-reports for the time being. (Should not be a problem once secondary roles are refactored.)

Apart from that:
- The PR implementation makes the logic send the reports just as described, yet it still ignores situations when two keys are physically pressed in the same cycle. I assume that this is known and expected.
- The report 3 cannot be always included due to nature of sticky keys (sometimes the sticky keys cease to be active before the key is released).  I am including it where possible, but I think that the requirement on 3rd report could be safely dropped.

mondalaci, 2019-12-29:
The mentioned corner cases don't have to be dealt with as they're very rare.

Were you able to check with Karabiner whether your fix works?

Depending on how prevalent the secondary role related issue is, it may not have to be fixed. If it has, it should be noted somewhere to not be forgotten, possibly in another issue.

kareltucek, 2019-12-29:
> Were you able to check with Karabiner whether your fix works?

No, I do not have access to any mac machine. But I've got some positive feedback on similar implementation with RDP sessions, so I am quite sure that this is a step in good direction. 

Can you or someone give it a test with Karabiner and report back please? :-)

Also, if we want to focus on dealing with Karabiner specifically, it might be a good idea to figure out the cause of the problems - macro engine's power should suffice for such testing.

> Depending on how prevalent the secondary role related issue is, it may not have to be fixed. If it has, it should be noted somewhere to not be forgotten, possibly in another issue.

I will deal with it at proper time, but a new ticket cannot hurt -> #254.

mondalaci, 2020-01-03:
I've been looking for related Karabiner issues and found #170 and #218. I mapped Cmd+Left to Mod+M. I could reproduce this kind of issue on OSX by holding Mod, tapping M, then tapping L. Strangely, when I interspresed "I" between M and L then Cmd only carried over to I and not to L. In other words, Cmd only carried over to the next immediate action.

I've also noticed that b3744436516511161512dfca0ba99d6762748b18 did fix the above issue, so I'm wondering whether this issue solves anything. Can you clearly reproduce any issues without applying the recent PR, but not when applying it?

kareltucek, 2020-01-03:
> In other words, Cmd only carried over to the next immediate action.

That is quite expected, because at that time Cmd has been released for quite a long time. 

So, do the extra USB report help or does it not? (You can enable stickies again for testing purposes on the fork by making shouldStick return always true.)

> I've also noticed that b374443 did fix the above issue, 

Well, obviously by disabling sticky modifiers the problematic area has shrinked, but still any action after alt+tab and cmd+tab and any other sticky-mod-allowed ones will suffer by the problem.

> so I'm wondering whether this issue solves anything.

Not directly, since my testing capabilities are somewhat limited... But lets first break the problem down a bit, since we are apparently dealing with at least two types of scenarios:

- Activation of modifiers - if added in the same report, modifiers might not been considered while interpretting the scancode. This seems to happen with RDP sessions, and might possibly exhibit with other software too. This is not mitigated by disabling sticky keys, unless there have been multiple big misunderstandings of symptoms (and therefore all reported cases are actuall of type two).
- Deactivation of modifiers - as you report, some software might carry the modifier over to next keypress if the modifier has been released with the modifier. This apparently happens in case of Karabiner, and is mitigated by disabling sticky keys.

Activation scenario related problems:
- There has ben UltimateHackingKeyboard/agent#871, which is quite clearly solved by this. Not entirelly sure if there has not been a big misunderstanding of symptoms, but I assume that 
- UltimateHackingKeyboard/agent#631 could be solved by this

Deactivation scenario related problems:
- All sticky-mod-enabled shortcuts on Karabiner I guess?
- Have you tested the text action? I suppose that the carrying over might have exhibited there (prior to the PR) on words beginning by capital letter.

kareltucek, 2020-01-04:
(In case you actually decide to test the text action, it might be good to focus on seven and five letter long strings which contain no repeating scancodes and are repeated at least seven times over... Like `AbcdeAbcdeAbcdeAbcdeAbcdeAbcdeAbcde AbcdefgAbcdefgAbcdefgAbcdefgAbcdefgAbcdefgAbcdefg`.)

kareltucek, 2020-01-04:
Anyway, is there some built-in support for testing things like this (e.g., the USB reports in general)?

mondalaci, 2020-01-04:
Thanks for elaborating above! As it turns out, you were right, and your extra reports PR actually improves things!

Before the sticky fix, the second action was always affected by the modifier of the first action. After the sticky fix but before the extra reports fix, the second action was only affected by the modifier of the first action if the key of the first action was not released before hitting the key of the second action. But after the extra report fix, the second action is never affected by the first action.

Well done!

PR #253 (2020-01-04): Extra usb reports
Closes #221
closed 2020-01-04 by mondalaci