UltimateHackingKeyboard/firmware #222: Misbehaving layer toggling via activated layer

mondalaci, 2019-05-19:
1. On the Base layer, let's bind activate mouse layer to key 1 and toggle mod layer to key 2.
2. Hold key 1.
3. Press key 2.
4. At this point, instead of triggering the action of key 2 on the mouse layer, the base layer action gets triggered, hence the mod layer gets toggled.

See https://github.com/UltimateHackingKeyboard/agent/issues/949

sribst, 2019-12-19:
Hi, 
I quick update on this issue, with the latest firmware the behavior has changed a bit. 
Now :

1. On the Base layer, let's bind "activate mouse layer" to key 1 and "toggle mod layer" to key 2. 
2. Hold key 1.
3. Press key 2.
4. At this point, instead of triggering only the action of key 2 on the mouse layer both base layer and mouse layer actions gets triggered

Thank you again !

kareltucek, 2019-12-21:
Interesting... I assume that this is related to one of my PRs. Will look into it soon.

closed 2020-04-29 by mondalaci