UltimateHackingKeyboard/firmware #227: repeating macros

bistabil, 2019-06-25:
I don't know if this feature already exists and it does than it's hard to find.
Is it possible to create macros that will repeat:
1. indefinitely
2. until some key is pressed
3. set amount time without having to copy macro actions multiple times (counter)
In the page featuring the keyboard I've found following text:
"A macro action is composed of a sequence of key actions, mouse actions and delays. **It will also have a loop flag which will loop the macro until the associated key is pressed.**"

mondalaci, 2019-06-25:
This is not implemented yet, and I doubt its usefulness. What's your use case?

richrd, 2019-06-25:
I guess point 2. might be useful for some special cases like gaming? I don't have a personal usecase right now though.

bistabil, 2019-06-25:
@richrd I have multiple usecases and yes, they are for automating gaming tasks.

I've thought about this a bit more and now I see potential issue if macro would repeat while at the same there is no way of manually stopping it.
Let's say in middle of macro I am interrupted and have to switch focus to different program.
Repeating (or actually any long macro) could create a lot of mess.

@mondalaci is it currently possible to forcefully cancel running macro?

As for loops I can use unofficial firmware.

mondalaci, 2019-06-25:
It's not possible to forcefully cancel a running macro. We were talking about implementing a macro cancellation action eventually.

Case 2 might be implemented in the future. The other cases seem rather specific. Chances are good that [Karel's UHK firmware](https://github.com/kareltucek/firmware) supports these cases.

mondalaci, 2020-04-29:
I'm sorry, but it's too much of a niche feature, and I rather wouldn't get it included into the main firmware. Feel free to implement it @kareltucek in your fork if you see fit.

closed 2020-04-29 by mondalaci

kareltucek, 2020-04-30:
All cases are already more-or-less supported. The last release also contains a command to stop running macros, which I have totally forgotten about and didnt mention in the changelog :-).