UltimateHackingKeyboard/firmware #228: Can't use two secondary roles at the same time

emanuelefirmani, 2019-06-26:
I configured my UHK with secondary roles on F (LCtrl) and D (Shift).
I'm not able to select words by clicking Shift + LCtrl + RightArrow.
UHK Agent 1.2.12 on Windows

kareltucek, 2019-06-26:
Secondary roles are not resolved recursively, although might I agree that they probably should be. The solution proposed in #230 still applies.

mondalaci, 2019-06-28:
@kareltucek It'd be useful if secondary role modifiers composed. Recursively resolving secondary role layers would be messy, though.

kareltucek, 2019-06-28:
I don't see any specific problem with layers... should I? 

(I assume the suggested inversion of secondary layers and a reasonable stateful implementation of layer switching.)

mondalaci, 2019-06-28:
@kareltucek Assuming that layer switcher actions can only be present on the base layer and their own layer (which is currently enforced by Agent), there should be no problems, indeed. Otherwise, one would be able to switch to a non-base layer from another non-base layer which I personally find quite confusing.

kareltucek, 2019-06-28:
Thanks for clarifying. I assumed that this ticket was (effectively) about modifiers only (i.e., with current layer switching restrictions in place).

emanuelefirmani, 2019-06-29:
This is what I built on my ergodox: 
https://github.com/emanuelefirmani/qmk_firmware/blob/master/keyboards/ergodox/keymaps/dvorak_programmer/keymap.c

As you mentioned, the ticket is about multiple second roles and not specifically to use them on other layers. Anyway, I thought secondary roles were available on any layer! 

As you can see from my qmk layout, I have ctrl below my left finger on all layers

mondalaci, 2019-06-29:
Secondary roles are available on every UHK layer. They just can't be composed for the time being. We should make them composable in the future.

emanuelefirmani, 2019-06-29:
Would be amazing if you could implement something like the qmk. 
Thanks for your timely reply

p4elkin, 2019-07-15:
@emanuelefirmani had a similar issue/requirement and ended up creating my own firmware fork: https://github.com/p4elkin/firmware, feel free to try it out (you can also find a link to it in the bottom of the official firmware's readme).

greneholt, 2019-12-30:
I also ran into this limitation. I have my semicolon key setup to activate the mod layer as a secondary role, and the caps lock key set to ctrl as a secondary role. I cannot hold down caps-lock, then depress semicolon, then use the page up/down keys. I had to set caps-lock to be purely ctrl on the mod layer, which allows me to depress semicolon first, then press caps-lock, then use other mod layer keys. This works, but it will take some time to retrain my muscle memory.

emanuelefirmani, 2019-12-30:
I sold the uhk and going to buy a keyboard that supports qmk. My ergodox does and works flawlessly

PR #265 (2020-04-11): Rewrite of secondary roles
So, rough list of changes:
- added postponer
- renamed keystate->next to keystate->hardwareSwitchState and added keystate->debouncedSwitchState
- integrated postponer on transition between the keystate->debouncedSwitchState and keystate->current (search for "PostponerCore" in usb_report_updater.c)
- added secondary role driver
- moved keystroke code from applyAction to applyKeystroke and added one indirection to decide between primary and secondary "keystroke" scenario. (Integrated secondaryRoleDriver here.)
- removed old secondary role hooks (updateActiveUsbReports  [... cut]
closed 2020-04-11 by mondalaci