UltimateHackingKeyboard/firmware #230: Secondary role doesn't work together with mouse

emanuelefirmani, 2019-06-26:
I modified the left control so that when tapped it acts like an Esc, LCtrl when kept hold (secondary role).
When I want to change the zoom on applications, LCtrl + Mouse wheel does not work at all. Even LCtrl + Mouse click on excel (select multiple cells) is not working.
UHK Agent 1.2.12 on Windows

kareltucek, 2019-06-26:
long story short (for @emanuelefirmani ): that's not how it works. The secondary role is not activated until the primary action is initiated (even though it could be). I.e., the "secondary role key" waits for as long as it is held for another UHK's action. If such action comes, it triggers the secondary role and then triggers the other key. Otherwise, it produces primary action tap on release. But the UHK cannot know that you are doing something with your mouse, and therefore cannot act on such action.

short story long (for @mondalaci ):  as a followup of #222, let me assert that I think that it would be more fortunate to implement secondary role the other way around - to always trigger secondary role, and once the key is released, then either execute primary action or 

bottom line (for @emanuelefirmani ): if you wish, you can use mine firmware fork to implement the secondary role style - in that case search for `ifInterrupted` examples in example section ( http://github.com/kareltucek/firmware ). However, be warned - writing correct macros using that firmware might be difficult unless you understand the keyboard-related abstractions (like difference between key, scancode and produced character, or the difference between UHK layers and modifiers).

mondalaci, 2019-06-28:
@kareltucek I'm leaning towards your suggested secondary role behavior, but afraid of the unintentional activation of secondary roles. I'd much rather have a global, optional secondary role timeout.

mondalaci, 2019-06-28:
@kareltucek On second though, maybe it'd be worth to implement your suggestion and if some people dislike it then introduce a global boolean setting for it. (An optional secondary role timeout could be useful regardless.)

kareltucek, 2019-06-28:
Actually... my suggestion would trigger an unintentional primary action in this scenario. I didn't realize it since I use the other style for layer switching, but I am not using secondary role modifiers at all (i.e., I am not using secondary roles together with mouse). Still, it would solve the other issue.

Regarding this scenario, combining secondary role with a hardware mouse is quite a niche usecase - I would probably be in favor of declaring it "illegal"/"exotic feature"... Timeout could solve it, but I feel it to be quite messy.

mondalaci, 2019-06-28:
@kareltucek How would your suggestion trigger an unintentional primary action in this scenario?

kareltucek, 2019-06-28:
Well, we are talking about a hardware mouse, aren't we? UHK knows nothing about such mouse and therefore, my suggestion (free of any timeout logic) would interpret single tap of ctrl (from UHK's point of view) as a primary action.

mondalaci, 2019-06-28:
I assumed that you suggested to hold the secondary role modifier until the primary role kicks in (if it does) at which point release the secondary role modifier and trigger the primary role.

Thinking about this implementation, it would surely have unintended side effects. Imagine Super as the secondary role modifier. When the primary action kicks in, Super would be released which would  invoke the Start menu on Windows unintentionally.

kareltucek, 2019-06-28:
Yeah, I did, but if you use a usb mouse to Ctrl+click, the primary action at the end of ctrl is no longer desired...

Yep, true. Nevermind...

mondalaci, 2019-06-28:
Gotcha, thanks for clarifying!

mondalaci, 2019-07-02:
Given the way secondary roles work, I consider this a non-issue. I'm closing this, but feel free to follow up.

closed 2019-07-02 by mondalaci