UltimateHackingKeyboard/firmware #231: Alt - left arrow not working

snaggen, 2019-06-28:
Using the attached configuration I have remapped left arrow to Mod+h as part som my vim mode. And that works fine. I have shuffled around the Alt key, not sure if it is standard or not, but it is on the second from the left on the left half. It works fine. Using different key capturing (tested agent and also gnome settings keyboard shortcuts) I can capture Alt+up arrow, doing Alt+Mod+k, all find and expected. However, I can't capture Alt+Left using Alt+Mod+h. I can do Alt+h and I can do left arrow Mod+h, but the combo Alt+Mod+h is not working. 

Using Firmware 8.5.4 and Agent version: 1.2.12

[UhkConfiguration8.zip](https://github.com/UltimateHackingKeyboard/firmware/files/3339357/UhkConfiguration8.zip)

mondalaci, 2019-06-28:
I can trigger Alt+Left via Alt+Mod+h by using your configuration. It works flawlessly. Maybe your window manager steals this shortcut. Whatever happens, I think it's specific to the configuration of your computer. Feel free to give this a try on another computer.

snaggen, 2019-06-30:
Yes, it seems to be in my OS or something. Sorry for bothering you.

closed 2019-06-30 by snaggen