UltimateHackingKeyboard/firmware #233: Need firmware update LINIVISION HDMI KVM04

shanechak, 2019-07-02:
Hello,

I am New user of UHK and i need firmware update for LINIVISION HDMI KVM04 to support it.
Specs Page:
https://www.amazon.com/LINOVISION-monitor-Keyboard-control-Players/dp/B07NZZ6RQW/ref=sr_1_1?keywords=linovision+hdmi+kvm04&qid=1562073062&s=gateway&sr=8-1

Thanks

mondalaci, 2019-07-02:
Hi there!

I firmware update may address this issue, but not the firmware update of
your UHK. The firmware of your KVM should be updated. Unfortunately, many
KVM switches don't handle the sophisticated USB interface of the UHK.
Please see our KVM switch compatibility
<https://ultimatehackingkeyboard.com/kvm-switch-compatibility> page for
more information.

Any more questions, feel free to ask me.

shanechak, 2019-07-02:
Are they new kvm models they support from the companies mentioned in your
website?

Best regards
Kumar S S

mondalaci, 2019-07-02:
I don't know for sure. The new KVM models should be tested.

mondalaci, 2019-07-02:
I'm closing this issue, but any more questions, feel free to follow up.

closed 2019-07-02 by mondalaci