UltimateHackingKeyboard/firmware #235: Gnome3 alt tab behaviour question/request

pbartyik, 2019-07-17:
The default alt tab bind is MOD + D
In Gnome3 when you alt tab, you can switch between open windows with arrows as well. In case a program (say terminal) has multiple windows open, you can hit the 'down arrow' and then chose which window you want to switch to.

With UHK when you hit MOD + D you can cycle between the active windows, but when you want to invoke a down arrow with MOD + K it drops you to the window you stand on at the moment (e.g. first active of a program).

The same behaviour is not present when I hit ALT + TAB and then MOD + K.

Can this behaviour be changed?

P.S. Just got it today, maybe I missing something?

mondalaci, 2019-07-17:
I'm sorry, but this is quite an edge case which is not planned to be supported. Feel free to check out [Karel's UHK firmware fork](https://github.com/kareltucek/firmware) for more advanced configuration possibilities.

closed 2019-07-17 by mondalaci

kareltucek, 2019-07-17:
@pbartyik I think that you might be missing that the gnome's behaviour is that Alt has to remain pressed while you press the arrow keys and that MOD is not Alt. If you remap the keys on the mod layer to alt+<arrow key>, it should start working. Of course, it means that you loose ability to press the arrows without alt.

The fork is able to help, however the usecase is quite an edge case even for that firmware, so the corresponding macro is not going to be simple - if you decide to try that way, feel free to create a new ticket (e.g., containing just a link to this one) at the linked repository so that we don't spam here with a possibly long discussion.

Practically, this could be also solved by remapping your window manager bindings or by learning to press arrows by your right hand and keeping the bindings as actual (hardware) alt+tab in your left hand.