UltimateHackingKeyboard/firmware #237: Getting the CLA signed

mondalaci, 2019-08-03:
Like many other open source projects, we recently added a Contributor License Agreement (CLA) to our GitHub repos. I got the CLA signed by almost every major contributor so far.

@algernon Would you please sign the CLA according to [the instructions](/UltimateHackingKeyboard/firmware/blob/master/CONTRIBUTING.md)? I've been trying to reach you via email about this, but haven't received a reply so far. Thank you very much in advance!

algernon, 2019-08-03:
I'm sorry, but I'm not comfortable with signing the CLA. I only ever signed two CLAs  for F/LOSS projects in my life, one for the FSF, another for SFLC, and I do not plan to change that anytime soon.

It would also be... somewhat challenging to sign it, as at the time of my past contributions, I was working for a rather large company with lots of bureaucracy, and while I had permission to work on anything open source outside of company-time, signing a CLA is something I'd have to consult their legal department about, since it covers a time I was at the company. I have neither time nor desire to do that.

mondalaci, 2019-08-03:
I understand. Thanks for explaining!

closed 2019-08-03 by mondalaci