UltimateHackingKeyboard/firmware #241: Unable to flash firmware 8.6.0 / intermittent connectivity when connected via docker

pbartyik, 2019-09-17:
Running fully updated Arch linux at the time of writing this.
Originally wanted to upgrade the firmware as the keyboard is misbehaving when connected via USB-C docker (ASUS SimPro Dock connected to Asus UX501WV, tried as well with another docker, same issue), keep getting this message in dmesg:

> usb 3-1.1.4: reset full-speed USB device number 13 using xhci_hcd

Using agent from [AUR](https://aur.archlinux.org/packages/uhk-agent-appimage/)
Attempted to upgrade firmware to 8.6.0, got the following log as error when only the UHK was connected to the laptop:

> Start flashing firmware
[18:32:10.556] Agent version: 1.2.12
[18:32:10.566] [UhkHidDevice] Available devices unchanged
[18:32:10.621] [UhkHidDevice] Used device: {"vendorId":7504,"productId":24866,"path":"/dev/hidraw1","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":0}
[18:32:10.623] [UhkHidDevice] USB[W]: 09 
[18:32:10.633] [UhkHidDevice] USB[R]: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[18:32:10.634] [DeviceOperation] USB[T]: Read left module version information
[18:32:10.635] [UhkHidDevice] USB[W]: 0e 01 00 
[18:32:10.643] [UhkHidDevice] USB[R]: 00 01 04 00 00 00 00 00 08 00 05 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[18:32:10.645] [DeviceOperation] USB[T]: Read right module version information
[18:32:10.647] [UhkHidDevice] USB[W]: 00 01 
[18:32:10.655] [UhkHidDevice] USB[R]: 00 08 00 05 00 04 00 04 00 05 00 00 00 04 00 00 00 00 00 04 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[18:32:10.659] Device right firmware version: 8.5.4
[18:32:10.660] Device left firmware version: 8.5.4
[18:32:10.852] New firmware version: 8.6.0
[18:32:10.854] [UhkOperations] Operating system: Linux 5.2.14-arch2-1-ARCH x64
[18:32:10.854] [UhkOperations] Start flashing right firmware
[18:32:10.855] [UhkHidDevice] Start reenumeration, mode: Bootloader
[18:32:10.965] [UhkHidDevice] USB[T]: Enumerate device. Mode: Bootloader
[18:32:10.966] [UhkHidDevice] USB[W]: 01 00 88 13 00 00 
[18:32:11.634] [UhkHidDevice] reenumeration devices up
[18:32:11.634] [UhkHidDevice] Device communication closing.
[18:32:11.635] [UhkHidDevice] Device communication closed.
[18:32:11.639] [blhost] RUN: /tmp/.mount_UHK.AgZjXH8O/resources/packages/blhost/linux/x86_64/blhost --usb 0x1d50,0x6120 flash-security-disable 0403020108070605
[18:32:11.764] [blhost] STDOUT: Error: UsbHidPeripheral() cannot open USB HID device (vid=0x1d50, pid=0x6120, sn=).
[18:32:11.770] [blhost] EXIT_CODE: 1
[18:32:11.771] [blhost] FINISHED: 1
[18:32:12.101] [DeviceService] updateFirmware error { message: 'blhost error code:1',
  stack: 'Error: blhost error code:1\n    at finish (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/src/uhk-blhost.ts:53:35)\n    at ChildProcess.childProcess.on.code (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/src/uhk-blhost.ts:36:17)\n    at emitTwo (events.js:125:13)\n    at ChildProcess.emit (events.js:213:7)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)' }
[18:32:12.103] [UhkHidDevice] Available devices:
[18:32:12.103] {"vendorId":7504,"productId":24864,"path":"/dev/hidraw1","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK Bootloader","release":2,"interface":0,"usagePage":16360,"usage":48420}
[18:32:12.104] [DeviceService] Read hardware modules information failed Error: [UhkHidDevice] Device is not connected
    at Promise (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:107:1)
    at Promise (<anonymous>)
    at UhkHidDevice.<anonymous> (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:104:1)
    at Generator.next (<anonymous>)
    at /tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:7:1
    at Promise (<anonymous>)
    at __awaiter (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:3:1)
    at UhkHidDevice.write (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:103:1)
    at UhkHidDevice.<anonymous> (/tmp/.mount_UHK.AgZjXH8O/resources/app.asar/webpack:/uhk-usb/dist/src/uhk-hid-device.js:155:1)
    at Generator.next (<anonymous>)
[18:32:12.105] [blhost] CLOSE_CODE: 1
blhost error code:1

mondalaci, 2019-09-17:
Does the bootloader USB device show up during the firmware update process according to `dmesg`?

If so, this must be a permission issue which should be fixed when installing the [udev rule file](https://github.com/UltimateHackingKeyboard/agent/blob/e33a391938ba7bdd015489243d5777a959915236/rules/50-uhk60_v2.rules).

If no, I'm not sure why it doesn't work, but it should upon implementing #1042.

pbartyik, 2019-09-17:
`dmesg` during the firmware upgrade:

> [  541.686713] usb 1-1: USB disconnect, device number 2
[  542.215325] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  542.356302] usb 1-1: New USB device found, idVendor=1d50, idProduct=6120, bcdDevice= 0.02
[  542.356308] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  542.356312] usb 1-1: Product: UHK Bootloader
[  542.356315] usb 1-1: Manufacturer: Ultimate Gadget Laboratories
[  542.358443] hid-generic 0003:1D50:6120.000A: hiddev0,hidraw0: USB HID v1.00 Device [Ultimate Gadget Laboratories UHK Bootloader] on usb-0000:00:14.0-1/input0
[  546.688699] usb 1-1: USB disconnect, device number 5
[  546.985235] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  547.126403] usb 1-1: New USB device found, idVendor=1d50, idProduct=6122, bcdDevice= 1.01
[  547.126406] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  547.126408] usb 1-1: Product: Ultimate Hacking Keyboard
[  547.126409] usb 1-1: Manufacturer: Ultimate Gadget Laboratories
[  547.128839] hid-generic 0003:1D50:6122.000B: hiddev0,hidraw0: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:14.0-1/input0
[  547.129324] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1D50:6122.000C/input/input29
[  547.185489] hid-generic 0003:1D50:6122.000C: input,hidraw1: USB HID v1.10 Keyboard [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:14.0-1/input1
[  547.185947] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1D50:6122.000D/input/input30
[  547.245341] hid-generic 0003:1D50:6122.000D: input,hidraw2: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:14.0-1/input2
[  547.245716] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:1D50:6122.000E/input/input31
[  547.302069] hid-generic 0003:1D50:6122.000E: input,hidraw3: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:14.0-1/input3
[  547.303002] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.4/0003:1D50:6122.000F/input/input32
[  547.303139] hid-generic 0003:1D50:6122.000F: input,hidraw4: USB HID v1.10 Mouse [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:14.0-1/input4

`udev` rule is there:

> cat 50-uhk60.rules 
# Ultimate Hacking Keyboard rules
# These are the udev rules for accessing the USB interfaces of the UHK as non-root users.
# Copy this file to /etc/udev/rules.d and physically reconnect the UHK afterwards.
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", MODE:="0666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", MODE="0666", GROUP="plugdev"

Any idea what can cause the original issue with the device resets?

mondalaci, 2019-09-17:
What happens in general and regarding the firmware update when your UHK is directly connected to your laptop without dockers?

pbartyik, 2019-09-18:
same, the agent log I pasted into the opening was when only the UHK was connected to the laptop without anything in between

mondalaci, 2019-09-18:
The device resets are unlikely to be caused by the UHK if they don't happen when directly connecting your UHK to your laptop. Can you update the firmware of your docker? Or perhaps if you can power vs not power your docker, does it make any difference?

Although not advisable by the security conscious, please run Agent as root, and try to update the firmware this way. If it works, then your USB permissions are not set up correctly in which case please ask help from the Arch community because Arch USB settings may differ from most distro.

pbartyik, 2019-09-18:
I tried to upgrade to agent 1.2.13, didn't work - did not try running the agent as root.
I ended up upgrading the firmware via another laptop running windows.
With firmware 5.6.0 I do not get the same `device disconnect` in `dmesg`, I'll keep monitoring this maybe the issue is just hiding again.

Firmware upgrade on the docking station will have to wait as I don't have a windows laptop with USB-C at hand right now to be able to do it.

mondalaci, 2019-09-18:
Glad you could update your firmware via another laptop. Chances are very high you'd have been able to update it via your laptop when running Agent as root. You're welcome to try it the next time. Of course, it should work as a normal user, but Arch in general or your installation in specific probably has some permission issues.

Also glad to hear you don't get `reset` in `dmesg` anymore, although I don't understand why. I don't think anything has changed in the firmware which affects this.

Your dock may be unable to handle sophisticated USB devices with multiple USB interfaces, such as the UHK. In this case, the sympomps are easy to spot, as everything works without the dock, and, well, not everything works via the dock.

I'm closing this issue, but feel free to follow up regardless.

closed 2019-09-18 by mondalaci