UltimateHackingKeyboard/firmware #242: Shift should release on keydown when part of a remapped key on a mod layer

johnlindquist, 2019-10-05:
[image]

I have some key remappings that look like the above where a single key is mapped to "shift+9" on a mod layer. 

Current behavior: "Shift" sticks as long as I'm pressing the mod layer key.
Desired behavior: "Shift" releases on key down. It should not be part of the next keystroke.

mondalaci, 2019-10-05:
@johnlindquist Yeah, well, the sticky behavior you experience on Mac is intentional according to my previous comment to make Alt+Tab and Cmd+Tab work as expected.

The sticky behavior may not be preferred in other cases though. I usually don't like to treat special cases differently than others, but in this case, it might make sense to only make Alt+Tab and Cmd+Tab stick, and not the other shortcuts.

What do you think?

johnlindquist, 2019-10-05:
I'm trying to think of other "sticky" scenarios. "Tab" seems to be the only
key requiring sticky modifiers, but maybe Return, Delete, and others have
some I can't think of. Maybe backtick/tilde and arrows?

If possible, I'd make all Alphanumeric and Symbol keys non-sticky and leave
others as sticky.

mondalaci, 2019-10-05:
I'm pretty sure that non-sticky behavior is preferred by users, so I'd make key actions sticky only if justified.

The firmware can match not only the scancodes, but modifiers too, so {Alt,Cmd}+Tab can be matched.

I can't see why making Return, Delete, Backtick, and Arrow keys sticky would be useful. Can you explain this?

kareltucek, 2019-10-05:
I can provide such example - same mechanics as alt tab, but under different window managers. E.g., if I remember correctly, ubuntu's unity or gnome shell use super+arrows to do window and workspace switching in similar manner...

I think that "guessing" the user's needs is a wrong approach to this. I.e., stickiness should either be enabled on demand by a check box in the key popover, or the current state of affairs should be preserved. Having to make macros as workaround may be an unintuitive pain, but it works and is conceptually coherent.

johnlindquist, 2019-10-05:
I agree that a sticky checkbox would be preferred 👍

I know cmd+~ is used in some "switch to different instance of same app"
scenarios, but I'm just guessing at the other keys.

Thanks for your time with this 🙏

johnlindquist, 2019-10-05:
For what it's worth, I'm pretty happy to keep using Karabiner for my remappings on osx

mondalaci, 2019-10-06:
In principle, I agree with you guys that the firmware should exhibit a consistent behavior without any exceptions, but I think this case is different. The macro workaround is very cumbersome, and few would understand or care about the sticky option. I think making a select few shortcut sticky could work surprisingly well. I plan to implement this eventually.

kareltucek, 2019-12-26:
Well, to open this again. This is not basically about choice of sticky shortcuts - the original report claims that shift sometimes carries over to next (UHK) keystroke even when that next keystroke does not contain shift. This is definitely not intentional and is a problem. However, I cannot reproduce it on Linux. 

@johnlindquist can you confirm again that this happens even with disabled Karabiner *and* that there are no other accessibility/whatever options which might carry the shift over? The cause has to be somewhere, and it is probably within Mac.

Then, can you please test with the attached firmware? (Contains extra modifier-free report before next keystroke.)

[uhk_firmware-8.7.0.zip](https://github.com/UltimateHackingKeyboard/firmware/files/4002079/uhk_firmware-8.7.0.zip)

kareltucek, 2020-04-24:
Should be solved:
- firmware 8.7.1. added extra usb reports as default feature, which should solve any unwanted carrying of modifiers
- conditional application of sticky keys has also been implemented.

Therefore I am closing this, that is unless anyone objects.

closed 2020-04-24 by kareltucek