UltimateHackingKeyboard/firmware #243: Keyboard becoming unresponsive and crashing

johnlindquist, 2019-10-11:
I wish I had a log to share, but I'll just have to describe the symptoms:

1. Keyboard slowly becomes unresponsive. Repeating keys that I didn't press. Then stops working
2. Lights on the upper left hand shut off (no mod, fn, mouse, code lights)

It's currently plugged into a Windows machine. Firmware 8.6. Using a customized Colemak layer.

I can easily fix it by unplugging my keyboard and plugging it back it. I don't even know where to begin looking for the issue though. Sorry. Help please? Is there logging software I could run?

mondalaci, 2019-10-11:
Can you reproduce this:
1. with other computers
2. when your UHK is merged and the bridge cable is not connected

johnlindquist, 2019-10-11:
It's difficult to reproduce at all. It's happened a couple of times randomly. I was hoping I could leave a logging tool open, cause I don't know when it's going to happen.

mondalaci, 2019-10-11:
Is your UHK from the pilot run?

johnlindquist, 2019-10-11:
It was one of the first units, yes.

mondalaci, 2019-10-11:
Do you use the spiral cable we provided? If you use an alternative cable, let me know if it's a spiral cable and its compressed or expanded length.

johnlindquist, 2019-10-11:
Yes, it's the provided cable. Pic below:

[image]

mondalaci, 2019-10-11:
There's only one thing I can think of then. The 50 pilot run UHKs were sent out with 4.7K pull-up resistors. Some of those UHKs exhibited similar issues as yours, so I switched to 1K pull-up resistors after the pilot run to resolve this issue.

[image]

These are ordinary 0805 ceramic resistors. Can you possibly replace them with 1K? If not, we can replace your UHK.

johnlindquist, 2019-10-11:
I've never done anything with resistors before, but I'm willing to try... Is there a youtube video of someone replacing resistors like these could follow along with? I don't know what tools I would need.

mondalaci, 2019-10-12:
You'd need a soldering iron, solder, and two 1K ohm 0805 resistors. You can desolder the 4.7K resistors with the iron by quickly alternating between their ends per resistor. Putting some solder to their ends first probably helps. I can remember of a relevant YouTube video, but can't find it. Let me know how it goes, or if you want your UHK to be replaced.

mikhey, 2019-11-20:
Interesting, I've recently noticed my UHK acting up since the latest firmware update.  I'm now getting the unresponsive random keys and repeatiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiing random keys again.  It's becoming excruciatingly frussssssssssssssssstratinnnnnnnnnnnnnnnnng.

mondalaci, 2019-11-20:
@mikhey What's your order id? Can you reproduce this issue when the bridge cable is unconnected and the halves are merged?

mikhey, 2019-11-20:
I'll have lookup my order number, it happens with the keyboard combined or separate.  But only on Windows, I don't have this problem on my MacBook Pro.  I do take my keyboard from home to the office and back each day and I do worry about the mini-USB connector wear and tear, but I'm careful and there would be other issue(s).  I do want to order another to just keep at the office, but haven't gotten around to it quite yet.  Would it be a possibility to order another USB cable?

mondalaci, 2019-11-20:
@mikhey We can provide another USB cable, but such cables are readily available from many webshops, and I don't think this issue is caused by the USB cable. Based on your order number, your UHK is from the pilot run, and as such, it's probably affected by the [I2C resistor issue](https://github.com/UltimateHackingKeyboard/firmware/issues/243#issuecomment-541227985). Wondering whether you can implement the suggested fix.

mikhey, 2019-11-20:
While I definitely want to get more into hardware and IoT etc, I'm not comfortably versed in soldering.  It could be some update that may have been pushed to my Dell laptop at the office.  There's a pre-boot screen that comes up and I don't have any issues with it.  Since, I've rebooted a few times it seems to be stable at the moment.  I'll keep my an eye on it to see if behavior continues - the difficulty I have is determining if it's the keyboard vs the OS/Computer.

mondalaci, 2019-11-21:
@mikhey Noisy power probably makes this issue more prevalent. It's worth checking if you can reproduce it when the charger is connected / unconnected. Cheap knockoff chargers may cause problems.

mikhey, 2019-11-22:
@mondalaci My guess is it was a combination of several things, corporate pushed Windows updates, USB registry problems (after a BSOD), the dock and/or power acting up - so many variables and not a reliable way to isolate and test them.  However, since I downgraded the firmware, it seems to be stable.  Given that I didn't have any problems on the MacBook Pro the whole time, I would chalk it up to something Windows specific.  I ❤️ my UHK!  Can't wait to have a second one to leave at the office!

mondalaci, 2019-11-25:
@mikhey So glad this seems to be resolved on your side! Also, thank you for planning to support us further!

I'm closing this issue because it seems that these issues have been resolved, and their root cause may very well differ. Feel free to shoot an email to support at ultimatehackingkeyboard dot com if you encounter any freeze related issues, guys.

closed 2019-11-25 by mondalaci

juneidysoo, 2020-01-21:
Hi guys, is anyone experiencing issues with latest firmware (8.7.1)?

I just updated to latest firmware about 2 days ago and only today I started to experience issues where occasionally the keyboard became unresponsive. (happened about 3 times within 5 hours today)

Just before the issue happened, the QWR LED died, then shortly after, it's on again. But after that, normal keyboard input does not work at all. Only mouse layer movement is working.

The only way to fix the issue is unplug then plug the usb back.

I am not using alternative cable:
[image]

juneidysoo, 2020-01-21:
Whoops, I should have used the search properly: https://github.com/UltimateHackingKeyboard/firmware/issues/261

mcscript, 2022-12-29:
I have experienced the same as @johnlindquist for a while: left keyboard half went "dead" from time to time.
The issue got worse (i.e. more frequent) when I installed the trackpoint module on the right side. I have tried everything from changing USB cables to using an externally powered hub (as I was suspecting some power supply issues).

I finally did a similar fix as described by @mondalaci above. Although, instead of de-soldering the existing 4.7k resistors, I just soldered 1.5k resistors on top of the existing ones, resulting in a resistance of (4.7k || 1.5k) ≈ 1.1k.

After doing this, I have not experienced the issues so far.