UltimateHackingKeyboard/firmware #244: Keyboard unresponsive after reboot

james64, 2019-10-29:
When I turn off & turn on my nb (with hardware power down in between) everything is ok. Issue manifests when I command my OS to reboot so that hardware is not powered down in a process. On reaching login screen I see that:
* Uhk's display showing layout is lit
* Other indicators are not lit and they do not respond to pushing mouse/mod keys
* Trying to type produces no input

When this happens I need to reach out and mechanically plug out & plug in my uhk. After this it works normally.

I use manjaro-i3 linux. Maybe this can be due to misconfiguration on os side?

I am running firmware 8.6.0 but saw this happening on all versions I tried do far.

mondalaci, 2019-10-29:
Can you reproduce this issue both when the charger of your notebook is connected / unconnected?

When your UHK becomes unresponsive and you press Fn+{1,2,...}, can you see the updated keymap abbreviation on the LED display?

james64, 2019-10-29:
Whole situation is unaffected by whether charger is connected or not.

LED display does not react on Fn+{1,2,...} presses. Once I mechanically reconnect this starts to work together with all other input as described.

Am I correct that you are not able to reproduce this with your uhk?

laxu, 2019-11-03:
I've had similar things happen sometimes when coming back from sleep on Windows 10. This seems to have only started after I flashed the 8.6.0 firmware. The keyboard lights up with the right keymap but it won't send anything to the computer.

Haven't had it happen to me on restart yet.

mondalaci, 2019-11-03:
@laxu Does this consistently occur with firmware 8.6.0, but not before?

james64, 2019-11-04:
@mondalaci I tried mine on mac and different linux. It worked in both cases. So this is some mis-configuration on my side. Before I am fine with closing this can you advice me on where should I look to fix this? What subsystem or something. No detailed steps needed I can research this myself. But I have absolutely no idea what to put into google to start with :)

laxu, 2019-11-05:
> @laxu Does this consistently occur with firmware 8.6.0, but not before?

It seems very random, I have had it happen only a couple of times. Replugging USB fixes it. Before I was running 8.5.2 or something like that without issues.

mondalaci, 2019-11-05:
@james64 Is your UHK directly connected to your computer? If so, please try it via a USB hub. If not, please try it without a USB hub.

@laxu Please downgrade to 8.5.2 and test it for a while. I don't think that any change since 8.5.2 should affect this issue.

mondalaci, 2019-12-02:
Closing this issue due to the lack of replies.

closed 2019-12-02 by mondalaci

james64, 2019-12-03:
Sorry, I somehow missed this. I had this issue for a while. And somehow it was solved after I updated my system like a week after I reported this. Miracle! Thanks for your help.