UltimateHackingKeyboard/firmware #245: Key states on the left half aren't synchronised enough with debouncer logic

p4elkin, 2019-11-13:
The usb report updater logic is totally in sync with the right half state: the loop in`main.c` will first scan the right key matrix state and the updater will read it before running yet another iteration of report update. 

However, the left half state is synchronised in `UhkModuleSlaveDriver_Update` function via async I2C communication (`I2C_MasterTransferNonBlocking` API), which [to my limited knowledge, received mostly from reading the API docs and tutorials like [this](https://www.allaboutcircuits.com/technical-articles/using-interrupts-on-arduino/)] may do the following:

> ...an Interrupt  interrupts whatever the processor is doing, and executes some code designed to react to whatever ... is being fed to the Arduino. Once that code has wrapped up, the processor goes back to whatever it was originally doing as if nothing happened!

In case of debouncing this could e.g. mean that the debouncer ensures the key state is set to non-active, but the rx message may update it to active (bounce!) and override bebouncer's efforts for the rest of the USB report updaters iteration logic. 

The current state of the firmware does not seem to obviously suffer from the issue due to the way the algorithm designed (the key states are rarely checked, each key is synched with report right after its bebouncing is handled).

However, if we synthetically make the algo a bit more complex (though still legit) by increasing the debounce timeout and by first handling the bouncing state for all the keys before propagating them to the reports (see this branch: https://github.com/p4elkin/firmware/commits/simulate-sync-issues), we may observe that while the right half will stably produce one key stroke per 5 seconds, the left may allow a series of identical chars over the same period.

One workaround would be to not write the RX update directly into the `KeyStates` matrix, but rather to some intermediate structure and then update the left half's `KeyStates` portion in the USB report updater iteration (pretty much the same way the right half is updated). This will ensure that during the report update phase the key state matrix is completely intact regardless of the algo complexity and the amount of times the current state is tested. (see this commit for reference https://github.com/p4elkin/firmware/commit/573a2ef1dcbd0a9b60fa8e9531005c6a1988fd72).

I used this workaroung for my fork and stopped experiencing chattiness _completely_!

Kudos to @kareltucek for coining in the insights for the investigation (see the discussion in another ticket: https://github.com/UltimateHackingKeyboard/firmware/issues/207#issuecomment-546341421)

mondalaci, 2019-11-16:
Thank you so much for elaborating about this issue! Your analysis is spot on, and so is your solution, but it should be generalized to work with the modules, too. I'll fix this when I get there.

kareltucek, 2019-11-16:
The cleanest solution would probably be to extend key_state_t by `next` field... That would work out of box with modules, and does not introduce more complexity into the code.

mondalaci, 2019-11-16:
@kareltucek Would you elaborate on the type and purpose of the suggested `next` field?

kareltucek, 2019-11-16:
Well, it would in principle exactly the same field as the existing `prev` and `current` fields, except it would the store "future" value. I.e., bools which indicate whether a key was/is(/will be) active or not

Basically it means the following:
- All new updates of states would be written into the `next` instead of `current` (as it is now). 
- Then, debouncing algorithm (which currently takes `current` and saves results to `current`) would take `next` and save result to `current`.
- On top of it, `prev = current` (which is currently orphaned at the end of the update loop) could be moved to the place of as well. That would even simplify code, since all state processing (except suppressions maybe) would be done at one place.

-----------------------

Or to put it another way - current problem is that the `KeyStatex->current` field is used for both distribution of the last debounced value and for storing the raw not-yet-debounced "future" value. So the refactor would simply split these two colliding uses of the field so that we would end up with clean `prev`, `current`, and `next` states.

kareltucek, 2019-11-16:
I can probably do it on Monday - should take no more than few minutes...

mondalaci, 2019-11-16:
@kareltucek Thanks for explaining! Your solution sounds like the way to go, and we'd termendously appreciate a PR.

kareltucek, 2019-11-16:
Ok

p4elkin, 2019-11-17:
I've some doubts about the simplicity of introduction of a third state of the key:
* developers are still encouraged to modify the `KeyStates` matrix in a non-deterministic way. It's just now it is not `current` property that may change unpredictably during the update, but `next`.
* `next` has imo confusing semantics. `Current` and `previous` states are clear - _is the key pressed now_ and _was it pressed on the previous update cycle_. What is `next`? _They key will be pressed in the future_? If we know that already, why `current` isn't updated? (i understand the synchronisation quirks, but from the common sense point of view, I am confused).
* re: debouncer, I don't understand this: `which currently takes current and saves results to current`.  What does *take* mean? As far as I can tell in deboucer it is the `previous` value that matters most and based on it, the current is updated (if debouncing timeout hasn't elapsed).

Maybe taking a look at the PR will make it clear for me too. For the moment I am pretty convinced that merely synchronising the I2C update with the USB report updates is much simpler:
* gurantee that KeyStates is intact during report update cycle
* only two states of the key: `current` denotes whether the key should be reported on this cycle,  `previous` - whether it was reported on the previous.
* of course, the way I do it now is rather crude and memory inefficient (by copying the whole KeyStates matrix), but using smth more lightweight like `key_matrix_t` might do a better trick.

kareltucek, 2019-11-17:
> * developers are still encouraged to modify the `KeyStates` matrix in a non-deterministic way. It's just now it is not `current` property that may change unpredictably during the update, but `next`.

Well, yes and no. Namely "no" because neither reading nor writing `next` discards its validity. The `next` simply contains the actual "up-to-date" information which is yet to be processed, and it does not matter *when* it changes. It is simply always up-to-date and always relevant (valid), but is not debounced. `current` and `previous` on the contrary contain information which has already been preprocessed ("commited for processing") and can be (and is being) acted upon. 

On the contrary I feel that this is the proper way, because all fields are used for exactly one purpose and are always valid with respect to that purpose. The simple duplication solution fixes nondeterminism (w.r.t. debouncer), but retains the property that `current`'s semantics changes in time (contains debounced value at some times and non-debounced value at others) and does so totally unpredictably from perspective of an external oberver. 

> * `next` has imo confusing semantics. `Current` and `previous` states are clear - _is the key pressed now_ and _was it pressed on the previous update cycle_. What is `next`? _They key will be [...]

My intuition is that the value is "not yet processed".

I can see that the name is a bit misleading (if that is what you are pointing at), but it is the most contextually consistent I can think of at the moment. 

I guess that semantics will become clear if the new field is marked as `volatile`, which explicitly tells the reader that the value can change at any time "on its own".

> * re: debouncer, I don't understand this: `which currently takes current and saves results to current`.  What does _take_ mean? As far as I can tell in deboucer it is the `previous` value that matters most and based on it, the current is updated (if debouncing timeout hasn't elapsed).

From my point of view, the debouncer's purpose is to properly detect and handle state changes rather than to prevent bounces. The information about state changes currently comes from `current`, not from `previous`.

> [...] For the moment I am pretty convinced that merely synchronising the I2C update with the USB report updates is much simpler:

My belief is that the two solutions are almost the same, I am just proposing to:
- Allocate the new memory within the existing structures in such way that both memory and actual code have better locality. I.e., so that:
  - it is easy to understand the code, and (if needed) document it.
  - it is easy to optimize memory requirements in future if needed. 
- Fix the semantical problem discussed above.

mondalaci, 2019-11-17:
I'm with @kareltucek, and I'm thrilled to see that you guys care so much!

p4elkin, 2019-11-17:
I now can see @kareltucek's point, it's just I tried to breakdown an algorithm into steps and debouncing is a pre-requisite to any further operations, so the othewise valid statement that bi-state approach `... retains the property that current's semantics changes in time` is just not so important for me: post-debounce it never changes during update cycle.

I will, however, test the approach once there's a PR to the official firmware, since I anyway planned to start catching up with it in the light of the add-ons progress.

closed 2019-12-02 by mondalaci