UltimateHackingKeyboard/firmware #246: Secondary role timeframe skips.

kareltucek, 2019-11-18:
If I read the code for current implementation of secondary roles correctly (never did that before), then the second key of a shortcut (i.e., first key is a key with secondary layer role, second key is an action key on the referenced layer) disables the key for current debouncer timeframe on transition from `SecondaryRoleState_Pressed` to `SecondaryRoleState_Triggered` - i.e., lets the debouncer lock on `(previous, current) == (0, 0)`. The key is then reactivated again after debouncer runs out.

That means that if the second key is pressed for less than 50 ms, or if the two keys have less than 50ms overlap, then the second keystroke will be ignored.  Also, the key activation is delayed by 50ms, which gamers could be pretty angry about.

As far as I my statistics are correct, the press time of a keystroke can usually fit within 90-250ms, but still - can we rule out that a professional writer or a user with a special type of switches cannot occasionally fit under 50 ms? Could anyone mind an unwanted 50ms delay? And even if we don't mind - Is the requirement of 50ms time overlap a bug or a feature?

Btw., this could simply be fixed by disabling the `KeyState->debouncing` flag at the transition...

kareltucek, 2019-11-18:
Also, assume that a secondary role key is pressed and then after some delay two different action keys are pressed in the same update cycle. When second one is processed, the state is already `Triggered` so the action is activated immediately, even though the layer has not been activated yet. I.e., it will produce action from wrong layer.

This second realization could be fixed by adding another state between Pressed and Triggered... (Or by rewriting it from scratch.)

mondalaci, 2019-11-18:
Can you reproduce this issue with a higher than 50 ms timeout value? If so, this should be fixed.

The 50 ms debouncing interval is overkill. We've received reports about chatter, and this is how I tried to combat the issue in a bruteforce manner. As it turned out, the root cause of such reports was hardware related which could be fixed by spraying contact cleaner spray into the misbehaving key switches. Cherry recommends 5 ms timeout, although I'm worried that it might be too low.

This should not cause a delay however. Key presses should immediately register. It's only subsequent key presses of the pressed keys which are delayed.

kareltucek, 2019-11-19:
Yes, if I increase the release debouncer timeout to 200 ms I can clearly reproduce the first issue.

kareltucek, 2019-11-19:
Anyway, there is a quick hackfix of both problems.

mondalaci, 2019-12-01:
@kareltucek I've been thinking about your PR for a while. Yours is the algorithmically correct implementation based on which I should accept it. But I'm concerned about other factors, too, namely maintainability and practical advantages.

When it comes to maintainability, the secondary role related code already melts my brain whenever I revisit it, and your fix makes it more complex by introducing yet another state transition.

When it comes to practical advantages, they seem to be almost negligible. I plan to reduce the debouncing timeout quite a bit, possibly to 5 ms which will make delayed activation practically a non-issue, even for gamers who seldom use dual-role keys for fast paced games. As for pressing multiple keys when triggering the secondary role, it's possible, but extremely rare, so it doesn't seem to make a practical difference.

Is there anything I'm missing? If not, even though I greatly appreciate your effort, I'd rather close your PR.

kareltucek, 2019-12-01:
Yes sir - the fact that the two problems are actually quite independent and that it is only the "two simultaneous keys race" which requires new state...

So the question is - what about just adding `keyState->debouncing = false;` into the clause? 

                if (secondaryRoleState == SecondaryRoleState_Pressed) {
                    // Trigger secondary role.
                    secondaryRoleState = SecondaryRoleState_Triggered;
                    keyState->current = false;
                    keyState->debouncing = false;
                }

It will solve the delay *and* will make the transition more explicit in the code - imho expecting the reader of the code to realize that the debouncer will kick in again on its own at that place is pretty sadistic.

kareltucek, 2019-12-01:
(...and unnecessary.)

kareltucek, 2019-12-01:
I mean, the impact is not big, but the delay feels wrong on principle and costs nothing...

mondalaci, 2019-12-01:
I love your updated fix. I can make this change or you can make a PR (or possibly modify your existing PR). In the latter case, the glory will be yours according to the commit history much deservedly.

kareltucek, 2019-12-01:
Go on :-).

commit b6511bb8 (2019-12-02): When a dual-role key is held, and the secondary role gets triggered by another key, don't debounce the latter key. Fixes #246 by @kareltucek.
closed 2019-12-02 by mondalaci