UltimateHackingKeyboard/firmware #256: Diagonal mouse movement speed

greneholt, 2019-12-31:
This is more of a feature request than a bug, but one of the things I am struggling with when using mouse mode is that the cursor moves faster diagonally than straight. I believe this is because the UHK is moving the cursor in both the X and Y axes at N pixels/second, which is truly a speed of `sqrt(2)*N` pixels/second, or 41% faster than when it is moving only in the X or Y axis at N pixels/second. This causes me to miss targets far more often when moving the cursor diagonally, as my brain is confused by the sudden acceleration.

I believe the usability of mouse mode would be greatly enhanced by decreasing the X and Y velocity to `N/sqrt(2)` when moving diagonally, so the cursor has the same absolute speed regardless of direction. This would better match the way humans typically use a real mouse, since we do not think about separate velocities in the X and Y axis, but only about the speed with which we are moving our hand.

kareltucek, 2019-12-31:
I've got no idea what Laci's opinion on this is going to be, but nevertheless, feel free to try this firmware [uhk-firmware-8.7.0.diagonal_slow_movement.zip](https://github.com/UltimateHackingKeyboard/firmware/files/4013529/uhk-firmware-8.7.0.diagonal_slow_movement.zip).

(You will need to unzip the file first - github does not allow uploading tar.bz2 archives.)

kareltucek, 2019-12-31:
(It is vanilla 8.7.0. with just the speed normalized - compiled from https://github.com/kareltucek/firmware/tree/diagonal_slow_movement )

greneholt, 2019-12-31:
Thank you for the quick response! I won't be able to try it until later, but I'm confused by the code change. It seems like the distance variable is modified after it has already been used to modify xSum and ySum, so I don't see how the change would have any effect...

kareltucek, 2019-12-31:
True, that happens when one writes faster than thinks...

kareltucek, 2019-12-31:
Should be better now :-). [uhk-firmware-8.7.0.diagonal_movement_slow.zip](https://github.com/UltimateHackingKeyboard/firmware/files/4013591/uhk-firmware-8.7.0.diagonal_movement_slow.zip)

greneholt, 2019-12-31:
Thank you! That works perfectly, and feels so much better in my opinion.

pcooke9, 2019-12-31:
I'm not a coder, so I don't know if this is related or not. It's just something that caught my attention a while back. I noticed that when holding the mouse left, and then hitting right while still holding left, the cursor doesn't change direction. Doing the opposite (right to left) does change cursor direction. The same goes for up & down. It occasionally effects my accuracy if I don't lift my fingers enough before changing direction.

I have Agent v1.3.0 and firmware v8.7.0 installed.

greneholt, 2019-12-31:
That is because the current implementation doesn't pay attention to which key was pressed most recently; left and up simply take precedence in case of a conflict.

kareltucek, 2020-01-01:
> Thank you! That works perfectly, and feels so much better in my opinion.

Glad you like it :-). 

>  I noticed that when holding the mouse left, and then hitting right while still holding left, the cursor doesn't change direction. Doing the opposite (right to left) does change cursor direction. 

Yes, that's right too... Curiously, no one seems to have either noticed or minded either of the two behaviours for as long as I remember.

Unless @mondalaci objects, I will add it to the list of tasks to solve...

mondalaci, 2020-01-01:
@greneholt Thanks for the sharp observation! Apparently, I hadn't thought of correcting diagonal movement speed when implementing this feature. It may very well be that Doom in general and the id Tech 1 engine in specific spoiled me which exhibits the same issue. :)

@kareltucek The corrected diagonal speed does feel more natural, although now I feel that the default mouse speed is a bit too slow, so I'll probably increase the defaults a bit. This doesn't affect your commit, just mentioning.

What's your opinion of the corrected diagonal speed, @kareltucek and @pcooke9?

I'm very well aware that the that left-right and up-down directions don't neutralize each other, but I wouldn't change the current behavior because this way inertia is preserved when swiftly changing directions.

kareltucek, 2020-01-01:
> What's your opinion of the corrected diagonal speed, @kareltucek and @pcooke9?

I do not perceive much difference - both feel natural to me. I've probably too been spoiled by videogames too. I've even messed up testing this twice in a row.

If this was up to me, I would probably let people configure this by a global option. 

> I'm very well aware that the that left-right and up-down directions don't neutralize each other, but I wouldn't change the current behavior because this way inertia is preserved when swiftly changing directions.

I'm not proposing neutralizing each other. The problem is that the behaviour is not left-right and up-down symmetric, because left/up movements always take precedence over right/down. Behaviour should rather be that last-pressed direction should take precedence over conflicting previously-pressed-and-held one.

mondalaci, 2020-01-01:
> > What's your opinion of the corrected diagonal speed, @kareltucek and @pcooke9?
> 
> I do not perceive much difference - both feel natural to me. I've probably too been spoiled by videogames. I've even messed up testing this twice in a row.
> 
> If this was up to me, I would probably let people configure this by a global option.

Let's wait for the feedback of more folks as I'm not super confident about this yet. As usual, I'd much rather provide the best behavior out of the box than providing an option. Feel free to provide an option in your firmware fork though.

> I'm not proposing neutralizing each other. The problem is that the behaviour is not left-right and up-down symmetric, because left/up movements always take precedence over right/down. Behaviour [...]

I love your proposal. Please implement it in a separate PR. Please make sure to follow the coding standards in light of the [recently spotted inconsistencies](https://github.com/UltimateHackingKeyboard/firmware/pull/251).

pcooke9, 2020-01-02:
> What's your opinion of the corrected diagonal speed, @kareltucek and @pcooke9?

To be honest, I didn't notice the speed change until I read this issue. Now that I compare the two, the correction does seem a little smoother.

> Behaviour should rather be that last-pressed direction should take precedence over conflicting previously-pressed-and-held one.

I'm disabled and have a bit of peripheral neuropathy in my hands, so this would definitely be appreciated.

I've had the UHK for a little over a year and it's helped me tremendously. I have a lot less discomfort while typing now. Thank you all for working so hard to make such a great keyboard. I can't wait for the release of the modules and backlight kit.  :)

mondalaci, 2020-01-02:
@pcooke9 Thanks for your quick feedback and nice words!

The last-pressed direction fix is already implemented and will be rolled out in the next firmware release. We'll give the diagonal fix a bit more time to gather feedback.

Super glad your UHK proved to be so helpful! Looking forward to release the backlight upgrade kit and modules. :)

mondalaci, 2020-01-07:
I've been testing this firmware for a while, and based on my experience, it's a step back usability-wise.

From a logical standpoint, one would think that making diagonal movements the same speed as non-diagonal movements is the Right Thing. However, the problem is that when only doing x or y movements, and then doing both, suddenly diagonal kicks in, and it artifically slows down the x and y axis movements. In practice, I find this quite confusing and I think it results in a worse key mousing experience.

closed 2020-01-07 by mondalaci

greneholt, 2020-01-07:
Could this please be added as an option then? As a brand new user to keyboard driven mouse movement, I find this to be a huge improvement over the original behavior. It could be that people who have already adapted to keyboard mousing with a faster diagonal speed find this change confusing, but I think for new users it makes the learning curve lower.

Also, there is an argument to be made that it's a generational change. Newer FPS games don't allow faster movement on the diagonal like Doom and Quake did (you don't see speed runners moving diagonally anymore), so people (like myself) that grew up playing Halo rather than Doom find this behavior more confusing.

mondalaci, 2020-01-07:
I'd rather not include this feature into the main UHK firmware, but would you be interested to include it into [your UHK firmware fork](https://github.com/kareltucek/firmware), @kareltucek?

kareltucek, 2020-01-07:
I will include a configuration option for that after merging the official-firmware-related refactors, but I am not convinced about principal correctness of such solution.

kareltucek, 2020-04-26:
This feature is now available in https://github.com/kareltucek/firmware (search for `setCompensateDiagonalSpeed`; or directly create a "write text" macro containing `$setCompensateDiagonalSpeed 1` and use it on keyboard startup).

Beware - current release is considered an experimental pre-release, although I think that basic features should be quite stable.