UltimateHackingKeyboard/firmware #259: Refactor usb_report_updater.c

mondalaci, 2020-01-04:
The codebase is in a pretty good shape with the notable exception of `usb_report_updater.c` which is already too big and  keeps growing.

It'd be worth to extract the mouse code to its own file. Further ideas are welcome.

kareltucek, 2020-01-04:
Noted. Furthermore, I plan to extract secondary role logic and probably layer switching logic into separate files, and also to make some changes in control flow (namely connected to the secondary role refactor)...

Anyway, I think I can share now the idea of secondary role refactor...

kareltucek, 2020-05-02:
Current situation:
- Old secondary roles were removed and replaced by a new mechanism which extracted the logic into new files.
- Layer switching refactored/rewritten into `layer_switcher.c`. Usb report updater still contains a branching related to action structure.
- Mouse code extracted to `mouse_controller.c`.

Possible next steps:
- Usb_report_updater still contains action branching logic. It might make sense to extract these too. I.e., to cut the tree at `applyLayerHolds` and `applyKeyAction` and move the code into new file(s).
- Sticky shortcut related code might also be extracted into a separate module.

I would prefer to leave this open for the time being and reconsider further steps later.

mondalaci, 2020-05-03:
Excellent job so far! Feel free to experiment with the suggested next steps. It's quite hard to foresee whether the suggested changes will actually simplify the codebase, so I understand it takes time to figure it out.

Let's not insist to chopping usb_report_updater.c to more pieces if the codebase gets more complex, though. I'm sure we're on the same page, just wanted to be explicit.

mondalaci, 2022-09-04:
There are an increasing number of overly large files. Currently, it's not a priority to refactor them, but eventually, I'll push refactoring and possibly participate in it. In the meantime, I don't want to have random related issues, such as this one, so I'm closing this issue.

closed 2022-09-04 by mondalaci