UltimateHackingKeyboard/firmware #260: Secondary role refactor.

kareltucek, 2020-01-04:
Finally, here is the proposal on replacing the secondary role implementation. Originally, I wanted to do this together with layer handling refactor, but I think it makes sense to at least discuss the two issues separately (this text is long enough as it is). Due to integration reasons, I will need to look into layers first though. (At the moment, I have a clear idea on what to do with secondary roles, yet I am still unclear on layer handling, so I will continue there with research in a few days time.)

So long story short - I would like to extend the keystate preprocessing by one more phase using a new module called Postponer. Its purpose would be to allow any keyboard action initiate "postponing mode", in which key activations are queued rather than activated straight away. The postponer would then be used to implement secondary roles via a simple "secondary role driver". 

The integration would be pretty minimalistic - just a few hooks in usb_report_updater, and the rest of the logic would be placed in a separate files, hopefully not adding much overall complexity and slightly simplifying the usb_report_updater.

In more detail, there are the following new modules (i.e., new files):

Postponer
---------
General purpose: to allow postponing new key activations until (e.g.) some condition is met. 

Specific usecase: this will allow secondary roles to intercept keydown events of other keys. If desired, this can allow implementation of more advanced secondary role resolution strategies, and of other things too. 

Principle: 
- The new module would contain an event queue (circular buffer), where postponed events would be stored. Postponer would be always either in an active or an inactive state. When inactive, `preprocessKeyState` would function just as now - when `keyState->next` changes, it is written into `keyState->current`. When active, `preprocessKeyState` would pass the state changes to the postponer instead. The postponer would save them into the queue, and later replay them into `keyState->next`. 
- Postponer would be active whenever there is an unprocessed event in its queue, or if someone asked it to be active in previous update cycle. 

Needed changes:
- Keystate_t would have to be extended by one more state - lets rename `next` to `hardwareSwitchState` and lets call the new state `debouncedSwitchState` - since `previous` and `current` will become detached from the hardware state of the switch. (Postponer will be inserted in the transition between `keyState->debouncedSwitchState` and `keyState->current`.)
- A few hooks would be placed at various places in report updater.

Pros & Cons:
- (+) It is a general solution which allows (simple) implementation of more advanced features, such as the resolution strategy for alphabetic keys.
- (+) Implementation already exists on my fork, so this is basically about refactoring it, simplifying it (stripping of not-needed functionality) and integrating with official firmware. (As a result, I plan to throw out a lot of experimental stuff out from my fork, and to steer the fork back towards the official firmware.)
- (-) It is more general than necessary.
- (+/-) It is transparent w.r.t. key action implementation. The good thing is low complexity. But it is less robust than p4elkin's solution, in the sense that it is not designed to keep and provide information about keystroke's lifetime to the actions/macros.

Secondary role driver
---------------------
Purpose: to deal with logic of secondary roles. It would provide some simple API to the outside world - something like `result secondaryRoleIsActive(keystate_t* keystate)` where `result = {PRIMARY, SECONDARY, DONT_KNOW_YET}` (or something along those lines). The keys with secondary role would simply query the module every time when they are processed. 

Principle: internally, the driver would be simple. It would work based on activating the postponer and then checking its queue until something happens. The driver needs to keep state per just one key (whose resolution is currently active). If different key queries for secondary role, result is always `IS_SECONDARY`.  All other state information are actually "hidden" in the serial nature of the postponer. This automatically makes it compatible with nested activation of the roles, and automatically solves corner cases and race conditions (e.g., time needed for the extra USB report).

Efficiency: I can write the query code so that regular check is done via inlined function based on a simple comparison, so actual function call would be executed only on state changes.

Pros & Cons:
- (+) I imagine it as a simple and elegant solution. 
- (-) Requires the Postponer.

Alternatives:
- This can probably be written directly without the Postponer, which however means retaining or creating new means of intercepting activations of other keys, and probably special handling of key deactivations too. As a result, a lot of current complexity would probably have to remain where it is. 
- Implement a state-pool (or somehting) so that every key can have it's own information-rich state (i.e., so that it can save a few integers for its lifespan). Then, using just the postponer and the state, secondary role could be implemented directly as an action of a key. This would allow us to leave out the possibly-tricky automaton-style implementation aspects of the driver. This alternative is actually quite interesting.

Discussion
---------

- So, does this sound reasonable at all?

- I've looked into p4elkin's implementation too. He has done quite an interesting piece of work there. Not sure how much you actually looked into it... His solution has some pros and cons (more robust, yet more complex), and probably should be considered too before plunging into my proposal (which is more minimalistic, less coupled complexity-wise, but also a bit work-aroundish). 

- What is your actual attitude towards the alphabetic-key-friendly resolution strategies? (#207 etc) Do you want such strategy to be implemented at all? Should I steer towards it, or not? (In any case such strategy should be configurable since it is bound to have some tradeoffs, but configuration is assumed anyway by UltimateHackingKeyboard/agent#792 if I am not mistaken.) 

- Regarding the advanced resolution strategies, I should probably note that although I can implement such strategy quite easily at the moment, it is still not as sophisticated as p4elkin's. 

- Taking note of UltimateHackingKeyboard/agent#792 and assuming that it is still desired.

- This solves #228, #198, #254 and relates to #207 and UltimateHackingKeyboard/agent#792.

mondalaci, 2020-01-06:
According to the above, the purpose of the Postponer seems to be strongly tied to secondary roles. Is this a sufficiently general mechanism to enable all the other features that belong to you? If not, it's not a big deal; I just want to see the big picture.

I really appreciate your above, super detailed description, but I still can't understand the details. If you already have this implemented and it's a matter of refactoring to merge into the main firmware, I'd say do it, and it'd be much easier for me to give feedback based on the PR.

Regarding the alphabetic-key-friendly resolution, I'd like to see it implemented. I'd rather not involve it for the time being because this issue is already quite complex as it is, but if you've already implemented it, then feel free to take care of it in the same PR.

kareltucek, 2020-01-07:
> According to the above, the purpose of the Postponer seems to be strongly tied to secondary roles. 

Well, the secondary roles are strongly dependent on the Postponer, but there is no coupling in the other direction. I.e., the Postponer provides some general APIs, but it does not reference secondary role mechanisms. But more generally yes, on the fork, the Postponer was implemented specifically in order to allow the alphabetic-friendly resolution strategy.

> Is this a sufficiently general mechanism to enable all the other features that belong to you?

All that are "peek-into-the-future" based (as long as the API is sufficiently broad). Two other "my" features depend on this are "bind action to arbitrary chord" and "bind action to gesture".

Of course there are other features which are not related to the posponer. From bigger "modules" for instance runtime macro recorder (or maybe better "scancode recorder") and stack-based layer resolution mechanism... And then plenty of little commands, conditions, sync mechanisms, parsing machinery, control flow.
 
> I'd say do it, and it'd be much easier for me to give feedback based on the PR.

Alright.

 > Regarding the alphabetic-key-friendly resolution, I'd like to see it implemented. I'd rather not involve it for the time being because this issue is already quite complex as it is, but if you've already implemented it, then feel free to take care of it in the same PR.

Ok, I am not yet sure which I choose. In the end, I would probably like to see the advanced resolution implemented in a more robust variant, which will require some further development of the Postponer, so unless somehow I manage to do the improvements just by the way, it probably makes sense to do it separately.

mondalaci, 2020-01-07:
Thanks for your explanations, they make sense. The firmware is progressively getting more complicated in the above respects, and I'm trying my best to keep up. Show me the code, and I'll comment on it. ;)

kareltucek, 2020-01-07:
Parse error: Not sure if you are expressing your enthusiasm towards future PR, or requesting pointers into the existing code on the fork. I assume the former. If latter was meant, please protest :-)

mondalaci, 2020-01-07:
The former :)

kareltucek, 2020-04-11:
Merged in #265.

closed 2020-04-11 by kareltucek