UltimateHackingKeyboard/firmware #261: 8.7.1 periodically ceases to send keystrokes

josephvusich, 2020-01-10:
Firmware: 8.7.1
Agent: 1.3.0

First incident happened within an hour after upgrading two UHKs from 8.5.3 to 8.7.1. Both keyboards had previously been running 8.5.3 for over a year with no issue.

### Occurrences
Total of 4 incidents on two different UHKs, over a period of 48 hours. In all cases it happened while typing, not while sitting idle.

UHK 1:
* Upgraded via Agent 1.3.0 on Windows 10 desktop with November 2019 Update.
* Froze twice, both times connected to the same Windows 10 desktop.

UHK 2:
* Upgraded via Agent 1.3.0 on Mac OS X 10.15.2.
* Froze twice, once while connected to Mac OS X 10.15.2, and once while connected to a laptop running Windows 10 1709.

### Description of event
The UHK abruptly stops sending keystrokes.

Keyboard seems to be responsive (layer switching, keymap switching, etc. still work). Unplugging and replugging the USB cable resolves the issue immediately.

josephvusich, 2020-01-10:
I have downgraded to 8.7.0 for now, to determine if the issue persists.

mondalaci, 2020-01-10:
Thanks so much for reporting this! I've experienced similar issues with 8.7.1 lately. I thought that my USB hub is the culprit, but given your report, chances are it's actually the firmware. Just followed suit and downgraded to 8.7.0. Let's keep each other updated!

kareltucek, 2020-01-11:
This might be unrelated, but I have been experiencing similar issue for quite some time on my fork. 

Approximately once every week, some key becomes "stuck" (in pressed state) and UHK does not react to subsequent presses. Not sure if all the details are the same (layer switching LEDs), will check next time it happens.

So far, I have been blaming my poor soldering skills and related possibility of either having damaged something by heat or having shortcirquited something by rosin residue.

pcooke9, 2020-01-12:
> I thought that my USB hub is the culprit, but given your report, chances are it's actually the firmware. Just followed suit and downgraded to 8.7.0. Let's keep each other updated!

I've had the issue with no keystroke input once with v8.7.1. I've also had stuck keys three times now. I could be wrong, but I'm pretty sure the first time I had stuck keys was with the normalized diagonal speed version of 8.7.0 that @kareltucek linked to in #256. It also happens with the new v8.7.1, but it definitely happened before I updated as well. I'll revert back to official v8.7.0 and see if it continues.

> So far, I have been blaming my poor soldering skills and related possibility of either having damaged something by heat or having shortcirquited something by rosin residue.

My soldering skills are terrible too ;-) but I don't remember anything happening that could have shorted my UHK and I haven't even opened it's case, so I doubt that's the cause. I don't use a USB hub.

fent, 2020-01-13:
I've had this happened 3 times now after upgrading to 8.7.1. I don't remember what I was doing specifically the last two times it happened. but this latest time, I was using mouse mode extensively.

I could still use mouse mode keys, but not any other keys or modes. I also checked if I was locked into mouse mode, but according to the keyboard led, I wasn't.

kareltucek, 2020-01-13:
> I'm pretty sure the first time I had stuck keys was with the normalized diagonal speed version of 8.7.0 that @kareltucek 

That was based on **b9af5e9cdf835261459e665e713eab46f16c4dcc** - approximately halfway between 8.7.0 and 8.7.1, i.e., before merging quite lot of Laci's code.

laxu, 2020-01-13:
Can report something similar on mine. I can be typing and suddenly a key gets stuck and keeps repeating and the only way to stop that seems to be to disconnect the keyboard and reconnect. Will drop back to 8.7.0.

bbolgradov, 2020-01-14:
I can also confirm. Sometimes it freezes while not being used, other times it freezes while typing and keeps repeating the key until reconnect. I'll stay on 8.7.1 as it doesn't happen often enough to annoy me (for now :)). Is there any debug log that I can attach when it happens that might help you?

mondalaci, 2020-01-15:
Firmware 8.7.1 is clearly affected. Firmware 8.6.0 should be stable because we've been flashing it to new UHKs in the factory for a while, and haven't received bug reports like this. We have to test whether 8.7.0 is affected.

Once we find the latest stable firmware version, we'll `git bisect` the commits between it and the subsequent (unstable) firmware version to find the commit that introduced this issue. Iterating the bisect will be challenging because this bug cannot be reproduced instantly, so we'll build releases per bisect iteration, attach them here for everyone to test them, and then we'll iterate the bisect further based on the results step by step. Would you proceed as suggested unless you have a better idea, @kareltucek?

I should also mention that there's a hidden feature that might help reproduce this issue. Hit Ctrl+Super+Alt+Shift+\ in Agent. This will unlock the USB Test Mode. Then hit Fn+` in any application to trigger/untrigger the USB Test Mode. This will exercise the USB interfaces by typing 'a', backspace forever, volume up, volume down forever, and moving the mouse leftward, rightward forever. When I say forever, I mean until hitting Fn+` again. Power-cycle your UHK to recover the functionality of Fn+`.

We're waiting for your feedback regarding firmware 8.7.0, guys.

kareltucek, 2020-01-16:
No better idea here. I tried to stress test it in hope to make the problem more reproducible, but without any result. 

Do you have any suspictions regarding possible cause? 

Regarding my code, I was wondering if the `volatile` flag of the new `keystate->next` could have increased the time of the interrupt callback enough to cause problems, but that was merged in 8.7.0 and that seems ok so far...

mondalaci, 2020-01-16:
I'm also suspicious of overly long interrupt routine execution, and I don't have better ideas. Bisecting is the best way I can think of to figure this out.

RaitoBezarius, 2020-01-16:
I'm flashing the 8.7.0 firmware and I'll try to conduct normal usage and give a try to the USB test mode. (@mondalaci we discussed this over emails few minutes ago)

epilgrim, 2020-01-16:
I've had multiple freezes in 8.7.1 in a single morning. Reverted to 8.7.0 2 days ago, and no problems so far

josephvusich, 2020-01-16:
No further freezes since flashing 8.7.0 a week ago.

RaitoBezarius, 2020-01-16:
Since flashing to 8.7.0, no freeze during ~ 3 hours of intense use.

mondalaci, 2020-01-17:
Thanks for your feedback, guys! Let the bisect begin.

```
$ git checkout v8.7.1
HEAD is now at 381d88e Bump firmware version to 8.7.1 in package.json
$ git bisect start
$ git bisect bad
$ git bisect good v8.7.0
Bisecting: 35 revisions left to test after this (roughly 5 steps)
[f608926a41d768f3c11673a66aaff7392117a1bc] Greatly simplify LedDisplay_SetText()
```

Please test [uhk-firmware-f608926a41d768f3c11673a66aaff7392117a1bc.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4074370/uhk-firmware-f608926a41d768f3c11673a66aaff7392117a1bc.tar.gz), folks.

You'll have to use the latest Agent release as only it supports the new .tar.gz firmware release format.

RaitoBezarius, 2020-01-17:
Okay, flashed it, will let you know how it goes during normal operations.

kareltucek, 2020-01-17:
> Would you proceed as suggested unless you have a better idea, @kareltucek?

Very sorry, did not read this properly until now. Promising not to deal with github at 1AM again.  

Yes, I will take care of further iterations.

RaitoBezarius, 2020-01-17:
@mondalaci It has been ~ 1/2 hours I have been using the firmware you sent, no freeze so far.

mondalaci, 2020-01-17:
@RaitoBezarius Thanks for your feedback! Let's test firmware builds for at least a day, and it may be worth to wait a couple of days for consistent results.

josephvusich, 2020-01-17:
Just experienced freeze with https://github.com/UltimateHackingKeyboard/firmware/commit/f608926a41d768f3c11673a66aaff7392117a1bc during an Alt+Tab. Keyboard was repeatedly sending Alt+Tab until unplugged.

Current layer at time of freeze had Alt and Tab mapped natively (i.e., not involving any layer switching or sticky shortcuts.)

RaitoBezarius, 2020-01-17:
Alright, for me, it has been more than 4 hours w/o freeze, I don't use Alt+Tab though, but I have used extensively the Mod/Mouse layer.

kareltucek, 2020-01-18:
Ok, thanks guys! Since josephvusich has experienced a freeze, I think we can consider f608926 as affected. This corresponds with the report about diagonal-movement-compensated version (b9af5e9) which is just a few commits ahead.

(Regarding my testing, I experienced one freeze yesterday evening with v8.7.1 (I wanted to make sure that my unit is affected too before diving into other versions). On the other hand, I have let the usb testmode running overnight with v8.7.1 and did not reproduce the issue. )

    git bisect bad
    Bisecting: 17 revisions left to test after this (roughly 4 steps)
    [fa0f29c5d45449721bd1038ef4e346a1415284c3] Make SPI communication more robust by toggling NCS before SPI transactions.

So next iteration is 
[uhk-firmware-8.7.0.fa0f29c5d45449721bd1038ef4e346a1415284c3.zip](https://github.com/UltimateHackingKeyboard/firmware/files/4079978/uhk-firmware-8.7.0.fa0f29c5d45449721bd1038ef4e346a1415284c3.zip) 

This is a version where `make-release.js` automatically creates a .tar.bz, so I had to zip it first and you will need to unzip it before flashing. (Or, @mondalaci, is there a way to pack old releases using the new format?)

So folks, please whoever can, flash the above firmware and ideally report that you are testing the release.

mondalaci, 2020-01-18:
@kareltucek Please repackage the .tar.bz2 firmware release as .tar.gz manually for the current and further releases for easier usability.

kareltucek, 2020-01-18:
Alright, should have occured to me that there is no magic involved :-).
 
[uhk-firmware-8.7.0.fa0f29c5d45449721bd1038ef4e346a1415284c3.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4080947/uhk-firmware-8.7.0.fa0f29c5d45449721bd1038ef4e346a1415284c3.tar.gz)

mondalaci, 2020-01-18:
The latest build is super unstable for me. How about you, guys?

kareltucek, 2020-01-18:
I have used it just for a few minutes today - no problem there. The rest of the time it ran on usb testing mode - no problem there either. It seems like the problem is triggered only by physical interaction.

But it suffices one relevant report to prove the firmware bad, so if you are sure about your report (I assume you are), I guess we can continue to 
[uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4081766/uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz)

    $ git bisect bad
    Bisecting: 8 revisions left to test after this (roughly 3 steps)
    [909e30433fb07b4f4146a4b4889ce3d5377ce6f4] Unset the shutdown pin, this time for real.

mondalaci, 2020-01-18:
Thanks! I'd much rather wait for somebody else to confirm, though. Let's make it a policy to have at least two people reproduce the bug. I'll test the latest bisect revision this time regardless, but you folks please test uhk-firmware-8.7.0.fa0f29c5d45449721bd1038ef4e346a1415284c3.tar.gz first until some of you reproduce the bug.

RaitoBezarius, 2020-01-18:
I'm testing the previous revision you said was super unstable, will give you in some hours, classical workload (development, Mod/Mouse layer).

So far, I see no problem, I flashed it ~ 30 mn ago.

kareltucek, 2020-01-18:
@mondalaci, on second thought, I agree that I would like someone to confirm.  If you were right, then it would narrow the problem down to this diff:

https://github.com/UltimateHackingKeyboard/firmware/compare/v8.7.0...fa0f29c5d45449721bd1038ef4e346a1415284c3

Otherwise, the range is https://github.com/UltimateHackingKeyboard/firmware/compare/fa0f29...f608926 , which somehow looks a bit more acceptable.

RaitoBezarius, 2020-01-18:
> v8.7.0...fa0f29c (https://github.com/UltimateHackingKeyboard/firmware/compare/v8.7.0...fa0f29c5d45449721bd1038ef4e346a1415284c3)
> Otherwise, the range is fa0f29...f608926 (https://github.com/UltimateHackingKeyboard/firmware/compare/fa0f29...f608926) , which somehow looks a bit more acceptable.

I confirm a freeze few minutes ago on the previous revision @mondalaci said was super unstable.

We can move on to the next revision.

RaitoBezarius, 2020-01-18:
I just flashed the latest firmware for test, will let you know how it goes as always.

RaitoBezarius, 2020-01-19:
I confirm no freeze during ~ 3 hours of usage.

kareltucek, 2020-01-19:
Thanks for detailed reporting! Knowing approximate usage times is very valuable in my opinion. 

Lets gather reports for one or two more days - we want to be sure when assuming "good", and we want to be very sure at these last iterations. 

(I have nothing to report - did not use my keyboard much since last revision.)

RaitoBezarius, 2020-01-19:
I confirm a freeze on the revision (the last one) I reported that nothing happened during ~3 hours of usage, in ~ 45 mn of usage, after a suspend/unsuspend of my laptop.
The key space was hold and it spammed spaces.

kareltucek, 2020-01-19:
@RaitoBezarius This is weird. Please reflash once more to the 909 firmware and try to reproduce the problem once again. Make sure you are really flashing the 909 version and not (e.g.) an alphabetically-last file. (If you do reproduce it, next step will be to test official 8.7.0 again.)

The thing is that 909 is already identical to v8.7.0. (All changes between 8.7.0 and 909 are in trackball module, which however is not packed into release files at the moment so I can literally run diff on unpacked tar.bz archives and get "everything is identical" results.) My local binaries for 8.7.0 differ from official ones though.

kareltucek, 2020-01-19:
My today testing - 909(etc) firmware, no occurence of the problem during approximately four hours (two hours of heavy writing load, rest lighter).

RaitoBezarius, 2020-01-19:
Karel, I reflashed 909 (I am pretty sure I flashed 909, but stupid mistakes happen and you're right.)

So far, for >2/3 hours of normal usage (light writing load, mostly mouse layer, scrolling, etc), everything was fine.
I guess we can try it for another day or two and can move on afterwards.

laxu, 2020-01-19:
Haven't had any issues on mine after flashing back to 8.7.0.

mondalaci, 2020-01-19:
@laxu Thanks! Please test uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz now and let us know the results.

pcooke9, 2020-01-20:
No issues for about a week with v8.7.0 and no issues with 909 for about 12 hours of light use.

juneidysoo, 2020-01-21:
I wonder if the UHK account has information on user's email address, but may I suggest sending email to people to not upgrade to 8.7.1 and explaining what symptomp that might occur with latest firmware?

mondalaci, 2020-01-21:
@juneidysoo We won't send emails, but I've added a relevant note to the firmware 8.7.1 release notes, and we'll bundle firmware 8.7.0 for further Agent releases until this bug gets fixed.

mondalaci, 2020-01-21:
Anyone who's joining this issue, please flash [uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4081766/uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz) with Agent 1.3.1, and report back whether it's stable. Test it for at least a day until you report, and possibly more depending on how often this bug occurs with firmware 8.7.1 on your UHK.

RaitoBezarius, 2020-01-21:
After ~1.5 days of testing with an intensive day of programming, I didn't
have any freeze with 909.

mondalaci, 2020-01-21:
909e30433fb07b4f4146a4b4889ce3d5377ce6f4 seems pretty stable so far for me, too. Let's test it for some more days, possibly until the end of the week to be sure it's stable.

kareltucek, 2020-01-23:
Update: Although I have been checking correctness of 909 package multiple times, it seems that I have probably messed up packing fa0f29 (and maybe packed wrong version).  This does not mean that we need to repeat the whole procedure, but the one iteration should be disregarded in further decision making. Also, since most changes on master are trackball-related, I think it makes sense to pick further bisect commits by hand rather than by strict binary halving. 

Disregarding fa0f29 means that the observation (made in [PR#265](https://github.com/UltimateHackingKeyboard/firmware/pull/265) ) about increased update cycle times in keystate refactor (b9af5e9) are still the possible cause (either direct or indirect). Unfortunately, my testing yields poor results. I managed to reproduce this problem five to ten times in the first ten minutes of using an artificially-slown-down firmware, but then it became gradually more and more difficult to reproduce. I think that something along the way is adapting to the changing parameters of the communication.

So I think that next bisect iteration should be to test the keystate refactor commit (b9af5e9). (It is approximately where next bisect should go anyways.) And if it proves bad, then to test the directly preceding commit.

Summary of the state (indented lines denote range between non-indented  commits):

    v8.7.1 (bad)
        (didn't check content of this range)
    f608926 (bad)
        LedDisplay refactors
    b9af5e9 Keystate refactor (probably bad)
    078be0b Scroll timer fix (probably good)
        just trackball-related changes
    fa0f29c (not sure about this)
        just trackball-related changes
    909e304 (testing now - assuming good)
        no changes that would compile into release
    v8.7.0 (good)

mondalaci, 2020-01-23:
@kareltucek Please check which revision the above 909e30433fb07b4f4146a4b4889ce3d5377ce6f4 build contains. I'd like to know for sure.

RaitoBezarius, 2020-01-23:
At least, for those past days, I never had any freeze with 909 and used it intensively for development with some updates to my keymap, etc.

bbolgradov, 2020-01-23:
8.7.1 was freezing a few times per day. I flashed 909 2 days ago. Coding during the day, some gaming during the night. No freezes so far.

epilgrim, 2020-01-23:
same here, no freezes during a couple of days on 909

kareltucek, 2020-01-24:
> Please check which revision the above uhk-firmware-8.7.0.909e30433fb07b4f4146a4b4889ce3d5377ce6f4.tar.gz build contains

I have made a few scripts for dealing with the packages today, so I am quite confident now. All three packages are correct (zipped bz fa0f29c, repacked fa0f29c and the 909), i.e., contain what they should contain. I must have mistaken folders when doing comparisons. I am very sorry for false alarm. 

Lets test 47e184b now, it should confirm the state of the affairs. 

[uhk-fw-47e184b.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4109509/uhk-fw-47e184b.tar.gz)

Thanks for all the reports on 909! (@RaitoBezarius  @bbolgradov, @epilgrim, @pcooke9, @mondalaci)  It would be very hard to test reliably without you!

dbankmann, 2020-01-24:
I can also confirm, no more freezes.

josephvusich, 2020-01-24:
No freezes after a day and a half on https://github.com/UltimateHackingKeyboard/firmware/commit/909e30433fb07b4f4146a4b4889ce3d5377ce6f4.

Just installed https://github.com/UltimateHackingKeyboard/firmware/commit/47e184b9b3c891dba7db0178d3aa551aca1df2ec and had a freeze within 5 minutes.

mondalaci, 2020-01-24:
47e184b froze 3 times so far for me, so we can safely say it's buggy, and time to bisect further.

kareltucek, 2020-01-24:
alright guys, now please give a shot to [uhk-fw-acf7c25.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4110181/uhk-fw-acf7c25.tar.gz)

kareltucek, 2020-01-25:
How is acf7c25 going so far?

Btw, @mondalaci when testing, do you have connected module prototypes?

bbolgradov, 2020-01-25:
It's been on for ~16 hours today, no freezes, but I haven't used it almost at all. I'll report on Monday when I get to work.

mondalaci, 2020-01-25:
It's been stable so far, and I haven't used the modules recently. Let's wait until early next week before bisecting further.

kareltucek, 2020-01-25:
This is the last iteration. acf7c25 directly precedes 47e184b . So it looks like 47e184b is the "culprit".

It is definitely worth to gather some more feedback to be sure (maybe to have another confirmation on 47e184b too), but I think in the meantime we can start thinking how the two lines in 47e184b in 'right/src/usb_report_updater.c' could be related to the problem:

    for (uint8_t moduleId=0; moduleId<UHK_MODULE_MAX_COUNT; moduleId++) {
        uhk_module_state_t *moduleState = UhkModuleStates + moduleId;
        if (moduleState->pointerCount) {
    -       ActiveUsbMouseReport->wheelX += moduleState->pointerDelta.x;
    -       ActiveUsbMouseReport->wheelY -= moduleState->pointerDelta.y;
    +       ActiveUsbMouseReport->x += moduleState->pointerDelta.x;
    +       ActiveUsbMouseReport->y -= moduleState->pointerDelta.y;
            moduleState->pointerDelta.x = 0;
            moduleState->pointerDelta.y = 0;
        }

I am pretty much out of ideas.

As far as I understand the code correctly, the clause should unreachable without connected modules. I have some doubts whether moduleState is always properly initialized, but even if it was not, adding garbage value into the report should not cause any problem. 

All I can think of is the possibility that the change has made the compiler compile the rest of the file differently (which is not unlikely) which has uncovered another problem *somewhere*. 

Even if acf7c25 was bad, then acf7c25 is the culprit since preceding a688ac5 is qeual to the good 909e304. But it contains only a change in signedness of the pointerDelta.

[45 more replies]

closed 2020-02-16 by mondalaci