UltimateHackingKeyboard/firmware #262: Random macro activation

scottStevenson8, 2020-01-13:
Macros activate randomly and then the mod, fn, and mouse indicator lights turn on permanently, and the keyboard becomes unresponsive until disconnecting the USB.

The macros are assigned to a different layer and it seems to only trigger the macros assigned to the left hand half of the keyboard (1-6 keys are reverse shell one liners).

Might be due to a new 2m coiled bridge cable or wear on the USB cable, but I only recently assigned the macros so it might be a coincidence. Is anyone else having this issue?

kareltucek, 2020-01-13:
Which firmware version?

Is it a new keyboard or is one from the pilot run?

How often does it happen?

Does it happen only when writing, or while idle too?

Is there any coincidence of macro key mappings and layer switchers? (E.g., having bound layer activation on one layer, while having bound macro on another.)

Do you use secondary roles on the keyboard?

Can you reproduce the issue with normal keystroke actions? 

And finally can you reproduce it without the 2m bridge cable? Since this happens on left half only, I some interference in the communication is quite likely.

mondalaci, 2020-01-13:
@scottStevenson8 If you use a spiral cable and its collapsed length is 2 meters then it's probably too long and communication errors occur. Please try to reproduce the issue with the stock bridge cable.

mondalaci, 2020-01-19:
Closing this issue due to the lack of response, and I assume that the suggested solution worked.

closed 2020-01-19 by mondalaci