UltimateHackingKeyboard/firmware #263: Ubuntu application switch works with 8.6.0 but doesn't work with 8.7.1 anymore

peterquiel, 2020-01-15:
Just received my UHK today and played around.

I discovered the  arrow keys with lalt modifier on the mod layer:
[image]

I can use the tab+lalt key to invoke the linux application switcher. Navigating to other application works pretty good with the arrow key beside the tab key. 
That works with firmware 8.6.0.

After updating to 8.7.1 that didn't work anymore. After hitting an arrow key the application switcher is closed immediately after releasing the key.

mondalaci, 2020-01-15:
Please state the window manager you're using, and elaborate on the exact key sequence you're trying to make work.

peterquiel, 2020-01-15:
I am on Ubuntu 19.10, gnome version is : Version: 3.34.1+git20191024-1ubuntu1~19.10.1

I can press lalt+tab to call the application switcher. I hold lalt all the time. 
Then I can use the arrow keys to navigate around using the mod layer. That just works.

Just by playing around, I discovered that using the <kbd>lalt</kbd>+<kbd>tab</kbd> and <kbd>lalt</kbd>+ &#8592; or <kbd>lalt</kbd>+ &#8594;  works as well. When releasing the mod key, the plication switcher switches to the selected application. 
Example: 
<kbd>lalt</kbd>+(<kbd>tab</kbd> <kbd>&#8494</kdb> <kbd>&#8494</kdb>)

This would invoke the application switcher, switch to another application and after releasing the <kdb>lalt</kbd> key the selected application will be activated. 

That works with the <kbd>lalt</kbd>+<kbd>tab</kbd> from the mod layer pretty good with 8.6.0 but in in 8.7.1 the application switcher immediatly switches to the application when pressing one of the lalt+arrow keys.

mondalaci, 2020-01-15:
@peterquiel Thanks for explaining! The reason what you're after doesn't work anymore is because we've made the sticky behavior of shortcuts stricter, and they only stick when Alt+Tab or Cmd+Tab is used. The new behavior is mostly beneficial except in this case.

@kareltucek I think we should keep the modifiers of the above shortcuts sticky for subsequent key presses until the layer switcher key gets released. That'd make this case work.

kareltucek, 2020-01-15:
Yep. {Alt|Gui}+arrows or just Alt+Arrows? Should I add them or will you? (Personally i am quite undecided on how it should be written for best readability & etc...)

mondalaci, 2020-01-15:
I'm not suggesting to extend the set of the current sticky keys, but making the modifiers of the activated shortcut stick for subsequent key presses until the layer switcher key gets released.

kareltucek, 2020-01-15:
Wait, what? Sounds as a very bad idea - what if the layer contains also other shortcuts which should not be affected.

And why not sticky? If i understand than your proposal, then it is a strict superset of making them sticky and for no reason. (You want to apply that rule to tab and arrows right?.)

kareltucek, 2020-01-15:
I can see point if you want to do this without the arrow keys having also assigned alt, but that was not the point, was it? 

In any case I think that such rule is unpredictable and can break a lot of things.

mondalaci, 2020-01-16:
I didn't think this thorough, but let me suggest a slightly different proposal. We'd keep the current set of sticky shortcuts. Then for every sticky shortcut, we'd specify a secondary set of shortcuts which would stick after triggering the primary shortcut.

kareltucek, 2020-01-16:
I don't see any major problems, but I am not convinced that it is a good thing to do.

I think that it will eventually turn out that we want Gui+arrows and maybe alt+arrows as primary shortcuts anyway (since as far as I remember, various linux window managers use different variations of the above functionality), and in that case these proposals will need to be relaxed anyway in the future.

Is there any specific usecase which makes extension of sticky key set by gui/alt+arrows a problem?

mondalaci, 2020-01-16:
I'm not aware of any specific usecase which makes the extension of sticky key set by gui/alt+arrows a problem, so let's simply add them as you suggest.

peterquiel, 2020-01-16:
@mondalaci Thanks for the explanation. This feature isn't that crucial for me. I just discovered a feature that didn't worked after firmeware update anymore. 

So, If this sticky stuff will cause any problems for other usecase, don't use my usecase as a reason to create any risks or bad side effects. 

Thanks a lot for this smooth keyboard remapping experience. It works like a charm.

mondalaci, 2020-01-16:
@peterquiel Appreciate your report, and we'll adjust the sticky keys to address your use case. I don't think it'll ever interfere with other use cases, but if it will, we might revert. Thanks for your nice words and support!

PR #266 (2020-01-19): Make arrows stick their modifiers.
Closes #263.
closed 2020-01-19 by mondalaci